esp8266/Arduino #4957: NodeMCU v1.0 high power consumption in deepsleep

fmarzocca, 2018-07-25:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12E]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

-----------------------
I am using a NodeMCU v.1.0 ESP-12 for a small project. It is powered by 3xAA batteries (4.5v) in Vin, but to reduce power consumption I need to put in in deepsleep every now and then.

Running this small sketch:

```
void setup() {
  Serial.begin(115200);
  Serial.setTimeout(2000);

  // Wait for serial to initialize.
  while(!Serial) { }

  Serial.println("I'm awake.");

  Serial.println("Going into deep sleep for 20 seconds");
  ESP.deepSleep(20e6); // 20e6 is 20 seconds
}

void loop() {
}
```
I have measured a consumption of 8mA when it goes to deepsleep, and this is not acceptable for my purpose. In order to wake up from sleep I have connected pin GPIO16 (D0) to RST, and it works. I knew that NodeMCU should have only few µA of consumption in deepsleep. How can I fix this?

Eszartek, 2018-07-25:
If the power consumption is unacceptable, then complain to the person who purchased it without reading the boards specifications. There is more than just an ESP8266 chip on that board. The 3.3v voltage regulator draws at least 5ma and the USB to serial chip draws the rest. Discconnect those two devices and you can "fix this".

Eszartek, 2018-07-25:
Unfortunately, that page you reference shows photos/pinouts of a NodeMCU board, but then only shows a power consumprion table for the bare ESP8266, and never mentions the other components on the NodeMCU board. Theres even a photo of a multimeter in the header graphic, but there were no real world measurements taken/reported in the article. That's very misleading and I'm sure very frustrating for the readers who would think that the author was being complete.

Eszartek, 2018-07-25:
Check out page 6 in https://www.espressif.com/sites/default/files/9b-esp8266-low_power_solutions_en_0.pdf and you'll notice that table 1-1 on that NodeMCU web page was taken directly from this document.

Pablo2048, 2018-07-25:
@fmarzocca You clearly read wrong pages (why You didn't ask original author? I can only see Your question about GPIO<->RST connection?). Try to read this:
https://tinker.yeoman.com.au/2016/05/29/running-nodemcu-on-a-battery-esp8266-low-power-consumption-revisited/
or this:
https://electronics.stackexchange.com/questions/224607/nodemcu-sleep-mode-works-but-still-consumes-too-much-power
BTW: this has nothing to do with Core IMHO...

fmarzocca, 2018-07-25:
Ok, thank you all for your time. I didn't wrote to that author as he never answered to my first question so I think he is no more following the thread.
I will have a look at your links, but I think I have found my solution: using an external low poer timer to switch off the board (I need it to work only few seconds each hour). TPL5110.
Thanks again

closed 2018-07-25 by fmarzocca

themelisx, 2019-07-31:
I know it's late answer but usefull info:
The embedded regulator has a lot of quiescent current (~ 6-7mA). Thats why you must NOT use 5V at Vin.
The solution is to give directly 3.3v to the board, using a LDO regulator like HT7333 that has quiescent current 4μA