esp8266/Arduino #4963: Failed to ping in softAP mode

pramitbiswas, 2018-07-26:
#### Platform

- Hardware:				[ESP-12e]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Size: 		        [4MB/1MB]
- lwip Variant:                  [v2 Lower Memory]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:              [115200]

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

	/* Headers */
	#include <ESP8266WiFi.h>      // ESP8266WiFi library
	#include <WiFiClient.h>
	#include <ESP8266mDNS.h>        // Include the mDNS library

	void setup() {
	  // put your setup code here, to run once:
	  Serial.begin(115200);
	  delay(1000);

	  WiFi.softAP(ssid,Pass);
	  
	  if (!MDNS.begin("esp8266")) {             // Start the mDNS responder for esp8266.local
		Serial.println("Error setting up MDNS responder!");
	  }
	  Serial.println("mDNS responder started");

	  delay(5000);
	}

	void loop() {
	  // put your main code here, to run repeatedly:
	}

```

### Debug Messages
In serial console it shows
```
mDNS responder started
```
However, I'm not able to `ping esp8266.local` from cmd.

Juppit, 2018-07-27:
I can confirm that on Windows 10 it is not (or not more) possible to access esp8266.local. It does not work anymore if bonjour is installed by Apple.
You can test with an iPhone to make sure your sketch works correct.

DonKracho, 2018-08-22:
besides MDNS do you see that you AP does emmit a signal?
You may check with an Android App like "WiFi Analyzer"

I already invested several hours in debugging but I can't get WiFi.softAP() to work stable.  The WiFi signal is already instable (broken every few seconds). Tested with 2.4.0(rc2) and already tried eraseConfig() followed by ESP.reset(). I'm going to prepare a MVCE sketch.

DonKracho, 2018-08-22:
Great with 2.4.2 all issues I had with WiFi in AP mode are gone! Just note that the password has to be at least 8 characters long, otherwise the ESP device will start on WiFi channel 1 with it's default name ESP_<last 3 bytes MAC>. When uploading set Erase Flash to "All Flash Contents" to get rid of may be weird WiFi config. This is the sketch that worked for me (without using MDNS, which has been the topic):

```
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

#define SSID "ESP8266WebServer"   // define your WiFi access point ssid here
#define PWD "#secret#"  		      // define your WiFi password here. it has bobe al least 8 characters! 
#define CHANNEL 9                 // WiFi channel

ESP8266WebServer server(80);

String GetPage()
{
  String page = "<!DOCTYPE html>\n<html>\n<head>\n<title>Dummy Table</title>\n";
  page += "<style>body{color:#eee;background:#346;align:center}table,th,td{font:0.9em Arial;white-space:nowrap;border:1px solid black;border-collapse:collapse;color:#222;background:#eee;padding:3px 10px}";
  page += "table{width:80%}th{padding:6px;font-weight:bold;color:#eee;background:linear-gradient(to bottom,#888,#000);}</style>\n</head>\n";
  page += "<body>\n<table>\n<tr>\n<th colspan='2'>Table Dancing</th>\n</tr>\n";
  for (int i = 2; i <= 30; i++) {
    page += String("<tr>\n<td>Row:  ") + i + "</td><td>The quick brown fox jumps over the lazy dog.</td>\n</tr>\n";
  }
  page += "</table>\n<p>Web page complete.</p></body>\n</html>\n";
  return page;
}

void setup() {
  Serial.begin(115200);
  delay(500);

  IPAddress apIP(10, 0, 0, 1);      // server IP
  IPAddress mask(255,255,255,0);    // subnet mask
    
  WiFi.disconnect();
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(apIP, apIP, mask);  
  WiFi.softAP(SSID, PWD, CHANNEL);
  Serial.printf("Started AccessPoint \"%s\" with IP: ", SSID);
  Serial.println(WiFi.softAPIP());

  server.on("/", [](){
    server.send(200, "text/html", GetPage());
  });

  server.onNotFound( [](){
    server.send(404, "text/plain", "");
  });

  server.begin();    
}

void loop() {
  server.handleClient();
}
```

devyte, 2019-11-10:
MDNS has been rewritten from scratch since this was reported. In addition, the original sketch was missing MDNS.update() in the loop, which is required to assure correct behavior of the responder.
Closing.

closed 2019-11-10 by devyte