esp8266/Arduino #4966: [Arduino OTA] Python consuming 25% of an I5 processor - Showing error in IDE after upload

rtek1000, 2018-07-27:
Hello,

I'm trying the Arduino OTA (built-in - Arduino IDE 1.8.5), and at some point I can note that after the upload is done, python gets stuck. The esp8266 program works normally, but the IDE returns error after a while.

When the Arduino IDE informs that the upload was done correctly, in the task manager I can see that python was consuming little processing, about 1% (of the Intel I5 processor). And after the IDE shows the upload completion message, python exits.

But when the IDE does not show the completion message as soon as the upload is done, then by looking at the task manager, I can see python consuming about 25% processing.

And after a long time the IDE shows an error, then the processor is released and I can notice that until the ventilation returns to the normal state.

When python is consuming 25% of processing, if I terminate the python process in the task manager, everything returns to normal, but the IDE shows error.

Is this problem caused in python or in IDE?

How does this interaction between IDE and Python work?

Thank you.

rtek1000, 2018-07-27:
Note, the python version I installed was the 2.7.10 (x64) I saw now that has version 2.7.15, I'll try: https://www.python.org/downloads/

SWR-DMaster, 2018-08-10:
Hi rteck,

I, too, am having this issue; though at 15% cpu per instance.
I found this post with a user having 100%, same issue. [](https://github.com/esp8266/Arduino/issues/4746)

rtek1000, 2018-08-10:
Hi,

Now I'm using Linux instead of Windows, it's faster for builds and I did not have that problem with phyton until now.

Using Lubuntu 18.04 running in a VirtualBox.

I'm also using the Arduino Pro Mini instead of ESP8266, I'm using ESP8266 with "esp-link" just to reprogram the Arduino flash memory.

[https://github.com/jeelabs/esp-link](url)

Operating in similar way to Arduino UNO Wifi:

[https://github.com/jeelabs/esp-link/issues/394](url)

Sometimes the IDE, the PING command, and the browser do not quickly find the ESP8266. But I can see in the settings of my router the IP address of the ESP8266. Maybe mDNS or another service is not working very well yet (This happens in Windows and Linux, however it seems that in Linux it solves a little faster).

Thanks.

d-a-v, 2018-09-13:
Can you try with #5135 ?

gordoste, 2018-09-22:
I am experiencing a very similar issue. I am running Windows 7 with PlatformIO and NodeMCU ESP8266 (ESP-12E).
#4746 seems similar but a little different as that user's espota.py NEVER came back, and it was intermittent. My issue occurs ALL the time.

The OTA update is completes successfully, but espota.py simply times out while waiting for the result. I ran WireShark on my PC and can see that no TCP packet with "OK" message from the OTA client gets to my PC. I can see all of the earlier ones saying how many bytes were received.

I tried adding #5135 in WifiClient.cpp but it didn't help. (In fact, I already tried doing flush() between client.print("OK") and client.stop() in ArduinoOTA.cpp before looking here for help!)

gordoste, 2018-09-22:
I have just found that putting delay(1000) between client.print("OK") and client.stop() in ArduinoOTA.cpp makes it work. I guess the Wifi module was not getting a chance to get the data out.
Obviously, this is a very dirty kludge, but hopefully it helps you understand what is going on.

d-a-v, 2018-09-22:
Could you make another tcpdump so we can check on the real delay ?
Current version of flush() only wait for 10ms at most. It is probably too short and should be increased and configurable.

d-a-v, 2018-09-23:
@gordoste any chance you can try #5089 ?
There is now a fair timeout for TCP flushing before stop among other changes (of course without your delay).

gordoste, 2018-09-23:
This fixes the issue, however the OTA update is much slower. Perhaps the timeout is being triggered before stop()?

d-a-v, 2018-09-23:
I could reproduce and this is fixed by `client.setNoDelay(true)` in OTA code (because now Nagle is working, and in the OTA protocol the ESP sends lots of sparse small packets).
You may please retry it, again without your delay.
I hope you can dedicate your i5 busy core back to something more productive.

d-a-v, 2018-09-24:
Should we understand that you tried again #5089 and got success ?
In that case, thanks for the feedback !
The OTA fix will probably be integrated since we have your confirmation

washcroft, 2018-12-08:
@d-a-v see https://github.com/esp8266/Arduino/issues/4283#issuecomment-445447030

earlephilhower, 2019-01-16:
This was fixed with the referenced PR #5089 .

closed 2019-01-16 by earlephilhower