esp8266/Arduino #4967: SPI::transferBytes pending changes

devyte, 2018-07-27:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			        all
- Core Version:			latest git
- Development Env:		all
- Operating System:	        all

### Settings in IDE

- Module:  			all
- Flash Mode:			all
- Flash Size: 		all
- lwip Variant:   all
- Reset Method:		all
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz|160MHz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

This issue is meant to track pending changes to SPI::transferBytes, specifically a missing assert and some api optimizations.
Ref: #4925 #2677 .

earlephilhower, 2019-01-22:
I would suggest supporting a simple alignment check and fix for transferBytes() because I believe it used to support unaligned transfers and libs like SDFat were using it that way.  The simple fixup should decompse to an "AND" of the address and a jump-on-zero (2 or 3 insns total) addition to transferBytes() and make things easier for existing older libs.

earlephilhower, 2019-01-31:
Specifically, transferBytes needs to handle misaligned pointers and non-by-4 lengths.
https://github.com/greiman/SdFat/pull/125

PR #5709 (2019-02-03): Allow unaligned input/output to SPI::transferBytes
Fixes #4967

Support any alignment of input and output pointers and transfer lengths
in SPI::transferBytes.  Use 32-bit transfers and FIFO as much as
possible.
commit 8c01516f (2019-02-03): Allow unaligned input/output to SPI::transferBytes (#5709)

* Allow unaligned input/output to SPI::transferBytes

Fixes #4967

Support any alignment of input and output pointers and transfer lengths
in SPI::transferBytes.  Use 32-bit transfers and FIFO as much as
possible.

* Refactor misaligned transfer, avoid RMW to FIFO

The SPI FIFO can't properly do RMW (i.e. bytewise updates) because when
you read the FIFO you are actually reading the SPI read data, not what
was written into the write FIFO.

Refactor the transferBytes to take account of this.  For aligned input
and outputs, perform as be [... cut]
closed 2019-02-03 by devyte