esp8266/Arduino #4970: MDNS does not respond to "QM" question

probonopd, 2018-07-28:
Slic3r Prusa Edition 1.4.0 tries to discover devices on the network like this according to Wireshark:

```
0000   01 00 5e 00 00 fb f0 4d a2 32 c2 5d 08 00 45 00  ..^....M.2.]..E.
0010   00 43 bd 82 40 00 01 11 1a 76 c0 a8 00 0e e0 00  .C..@....v......
0020   00 fb d5 88 14 e9 00 2f a1 f2 e7 42 00 00 00 01  ......./...B....
0030   00 00 00 00 00 00 0a 5f 6f 63 74 6f 70 72 69 6e  ......._octoprin
0040   74 04 5f 74 63 70 05 6c 6f 63 61 6c 00 00 0c 00  t._tcp.local....
0050   ff                                               .
```

[image]

Real OctoPrint instance on a Raspberry Pi answers immedately with

```
0000   f0 4d a2 32 c2 5d b8 27 eb 10 e9 71 08 00 45 00  .M.2.].'...q..E.
0010   00 df 0c 09 40 00 ff 11 ed 90 c0 a8 00 15 c0 a8  ....@...........
0020   00 0e 14 e9 d5 88 00 cb f6 f3 e7 42 84 00 00 01  ...........B....
0030   00 05 00 00 00 00 0a 5f 6f 63 74 6f 70 72 69 6e  ......._octoprin
0040   74 04 5f 74 63 70 05 6c 6f 63 61 6c 00 00 0c 00  t._tcp.local....
0050   ff c0 0c 00 0c 00 01 00 00 00 0a 00 1f 1c 4f 63  ..............Oc
0060   74 6f 50 72 69 6e 74 20 69 6e 73 74 61 6e 63 65  toPrint instance
0070   20 6f 6e 20 6f 63 74 6f 70 69 c0 0c c0 33 00 10   on octopi...3..
0080   00 01 00 00 00 0a 00 1e 06 70 61 74 68 3d 2f 07  .........path=/.
0090   61 70 69 3d 30 2e 31 0e 76 65 72 73 69 6f 6e 3d  api=0.1.version=
00a0   31 2e 32 2e 31 30 c0 33 00 21 00 01 00 00 00 0a  1.2.10.3.!......
00b0   00 0f 00 00 00 00 00 50 06 6f 63 74 6f 70 69 c0  .......P.octopi.
00c0   1c c0 8e 00 1c 00 01 00 00 00 0a 00 10 2a 02 09  .............*..
00d0   08 18 71 84 81 c9 e3 de 13 ab d9 f1 85 c0 8e 00  ..q.............
00e0   01 00 01 00 00 00 0a 00 04 c0 a8 00 15           .............
```

The ESP using Arduino 1.8.5 with esp8266 2.4.1 and this sketch does __not__ answer this. __WHY?__

```
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiClient.h>

const char* ssid     = "...";
const char* password = "...";

// TCP server at port 80 will respond to HTTP requests
WiFiServer server(80);

void setup(void)
{  
  Serial.begin(115200);
  
  // Connect to WiFi network
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");  
  
  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (!MDNS.begin("esp8266")) {
    Serial.println("Error setting up MDNS responder!");
    while(1) { 
      delay(1000);
    }
  }
  Serial.println("mDNS responder started");
  
  // Start TCP (HTTP) server
  server.begin();
  Serial.println("TCP server started");
  
  // Add service to MDNS-SD
  MDNS.addService("octoprint", "tcp", 80);
  MDNS.setInstanceName("OctoPrint instance on ESP8266");
  MDNS.addServiceTxt("octoprint", "tcp", "api", "0.1");
  MDNS.addServiceTxt("octoprint", "tcp", "path", "/");
  MDNS.addServiceTxt("octoprint", "tcp", "version", "1.2.10");
  
}

void loop(void)
{
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  Serial.println("");
  Serial.println("New client");

  // Wait for data from client to become available
  while(client.connected() && !client.available()){
    delay(1);
  }
  
  // Read the first line of HTTP request
  String req = client.readStringUntil('\r');
  
  // First line of HTTP request looks like "GET /path HTTP/1.1"
  // Retrieve the "/path" part by finding the spaces
  int addr_start = req.indexOf(' ');
  int addr_end = req.indexOf(' ', addr_start + 1);
  if (addr_start == -1 || addr_end == -1) {
    Serial.print("Invalid request: ");
    Serial.println(req);
    return;
  }
  req = req.substring(addr_start + 1, addr_end);
  Serial.print("Request: ");
  Serial.println(req);
  client.flush();
  
  String s;
  if (req == "/")
  {
    IPAddress ip = WiFi.localIP();
    String ipStr = String(ip[0]) + '.' + String(ip[1]) + '.' + String(ip[2]) + '.' + String(ip[3]);
    s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html>Hello from ESP8266 at ";
    s += ipStr;
    s += "</html>\r\n\r\n";
    Serial.println("Sending 200");
  }
  else
  {
    s = "HTTP/1.1 404 Not Found\r\n\r\n";
    Serial.println("Sending 404");
  }
  client.print(s);
  
  Serial.println("Done with client");
}

```

Does not show up in Slic3r. (Sometimes, does not even show up in _avahi-discover_ either.) Makes me wonder whether the whole Zeroconf thing is broken on esp8266...

cc @devyte
Reference: https://github.com/probonopd/WirelessPrinting/issues/32

devyte, 2018-07-28:
Yes it is, yes it's being rewritten, no there's no ETA because it will take a long time.
Even if the currently known bugs were fixed, the current mdns implementation does not pass Apple's conformance tests.
If you want to propose fixes for mdns for the meantime, they will be reviewed.
Closing as known issue.

closed 2018-07-28 by devyte

probonopd, 2018-07-28:
@devyte do you think the situation has worsened compared to earlier releases? Because right now (Arudino 1.8.5 with ESP 2.4.1) I can't seem to get it to work at all anymore...

devyte, 2018-07-28:
The mdns code hasn't changed much. What has changed (I.e. corrected) is lwip, and how packets are handled. What has also apparently changed in some cases is the increase in features in some devices, how strict some devices are wrt mdns compliance, and what happens when faced with devices that are not so compliant, hence the whole discussion about passing Apple's conformance tests.
A trivial/partial mdns implementation seems to no longer be enough.