esp8266/Arduino #4972: ESP-01 Initialization and Serial message

dimitre, 2018-07-28:
I'm building a prototype sending midi out from ESP and I've noticed every time I boot it sends an undesirable message from Serial TX. (translates as program change message on MIDI)
Even uploading an empty sketch with empty setup and loop seems to send this message.
And it blinks the blue LED (is it tied to Serial data pin?)

It would be great to be able to supress this Serial message (or LED blink, in the case both use the same pin)
Thank you

Pablo2048, 2018-07-28:
Do some simple search for this before opening an issue next time please :-( ...
https://bbs.espressif.com/viewtopic.php?t=2212
https://forum.micropython.org/viewtopic.php?t=3476
and many, many more...

devyte, 2018-07-28:
The bytes come from Espressif code during boot and can't be suppressed. There's supposed to be a trick possible to divert the bytes to a different pin, per Espressif docs, but you would have to figure that out yourself.

closed 2018-07-28 by devyte