esp8266/Arduino #4974: Include SSDP.SetUDN

laercionit, 2018-07-28:
Hello, I would like to know if it is possible to include the UDN for the SSDP as a parameter.
My code generates a specific UUID, so I analyzed the Lib it is using CHIP to mount the UUID.

laercionit, 2018-07-29:
Hello, I made an example, basically check if the UUID was set or not.
If it is informed I assume the value informed, if not informed I maintain the use of the function that generates the UUID based on CHIPID.

I hope you like it and can be incorporated from the master.

[SSDP.zip](https://github.com/esp8266/Arduino/files/2238549/SSDP.zip)

devyte, 2018-07-29:
@laercionit Could you please make a pull request with this for consideration?

closed 2018-08-01 by devyte