esp8266/Arduino #4986: Turn off lights doesnt work correctly

dimitris17563, 2018-07-30:
Hi,
First i want to thank you for the very nice project you share with us. I wanted to make a wi-fi E27 bulb controling it from mobile so your project is ideal for it. I managed everything to work fine althought i had some problem with not stable operation of leds (some flickering) which was because of bad connection with signal and also i reduced the resistor to 100ohms and now works fine. I have only one problem. When i click the OFF button in app the leds turn off but after 1-2 seconds turn on again. How can i fix it. I have not install the button because i will not have a contact with bulb, i want everything to be remotely.
Thank you very much for your help.

Best Regards
Dimitris

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12r]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[|Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Detailed problem description goes here.

pipi61, 2018-07-30:
where the program?

dimitris17563, 2018-07-30:
I use the Mclighting code https://github.com/toblum/McLighting
Sorry i thought i was writing the problem in the McLighting section. Can you help me or to ask there?
Thank you.

Pablo2048, 2018-07-30:
Open an issue here https://github.com/toblum/McLighting/issues ...

pipi61, 2018-07-30:
Unfortunately I can not help you
This issue is only arduino development.
Try the open issue on toblum/McLighting page.

dimitris17563, 2018-07-30:
Yes ok. Thank you.

minida28, 2018-07-30:
And please close this issue :-)

closed 2018-07-30 by devyte