esp8266/Arduino #4987: wifi_set_user_ie not working?

ranr, 2018-07-30:
#### Platform

- Hardware:				ESP-WROOM-02
- Core Version:			Version 4.2.1
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Problem Description

I am trying to set a 802.11 user information element (IE) in the beacon, unsuccessfully. I'm calling this function:
 wifi_set_user_ie(bool enable, uint8 *m_oui, uint8 type, uint8 *user_ie, uint8 len);

Wireshark doesn't show anything - I get the Espressif OUI and that's it. No sign of my vendor OUI.

Is there any example of how to use it?

d-a-v, 2018-07-30:
We are not controlling this part of the firmware which is closed source.
You may ask your question in dedicated forum. If you are not able to solve it,
you need to setup an example using pure nonos-sdk environment and submit your issue to espressif.
Reading [#4853](https://github.com/esp8266/Arduino/pull/4853#issuecomment-403471774) may help for this task, or better, espressif's nonos-sdk documentation.

devyte, 2018-07-30:
The wifi_set_user_ie() function is not included in the NONOS docs. It seems to be an RTOS api only. It's possible that under NONOS it doesn't work, it's bugged, or if it does work and it's just undocumented, you're somehow using it incorrectly.

Whatever else, it's an Espressif api, and not part of this core.
Ref: Espressif RTOS SDK api reference v1.4.0 section 4.14.3.2.
Ref: Espressif bbs [4955](https://bbs.espressif.com/viewtopic.php?t=4955)
Closing.

closed 2018-07-30 by devyte