esp8266/Arduino #4990: Noise from buzzer after waveform generated by calling tone() ends

fedor1210, 2018-07-31:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue (found #3759, but @earlephilhower suggested to open a new one).
- [x] I have filled out all fields below (well... partly).

#### Platform

- Hardware:				ESP8266(EX) and ESP8285
- Core Version:			29 July 2018 (git, replaced files in PlatformIO)
- Development Env:		PlatformIO
- Operating System:		macOS

### Settings in IDE

- Module:  			everything
- Flash Mode:			dio
- Flash Size: 		n/a
- lwip Variant:   ?
- Reset Method:		n/a
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		both OTA and Serial
- Upload Speed:   115200

### Problem Description

After `tone()` some strange noise (annoying clicking sounds, it seems the network activity causes them) is generated by buzzer. Noise isn't produced before `tone()` is called. I have pretty sensitive and loud buzzer, so I hear those clicks even in another room. If you don't hear them, just try another buzzer (I use electromagnetic ones, not piezo) or hold it near ear. `digitalWrite(buzzerPin, LOW)` doesn't help even after delay.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>

void setup() {
   WiFi.mode(WIFI_STA);
   WiFi.begin("ssid", "password");
   pinMode(buzzerPin, OUTPUT);
   tone(buzzerPin, 3072, 50);
}

void loop() {

}

```

P.S. sorry for my poor English

earlephilhower, 2018-07-31:
Your English is fine, but I have a few questions,

A buzzer is something that makes 1 kind of noise when you connect VCC and GND.  So you could hook it to a battery and it would make noise until the battery died.

A speaker is something that needs to be driven by something else (like in your TV or headphones) to make sound.  Hooking it up to a battery won't make any noise other than a single click/pop.

If you are using `tone()` with a buzzer, that won't work since it doesn't want a frequency, it only wants VCC or GND connection.

Also, if you are driving either a speaker or buzzer directly from an ESP8266 pin, that won't work., either  The ESP8266 cannot source or sink enough current to safely drive a speaker or buzzer.  You need use something like a transistor or MOSFET with base connected to the pin to actually drive the pin up or down.  See the writeup in my audio project for the simple [schematic](https://github.com/earlephilhower/ESP8266Audio)

The fact that you're hearing clicks after a `digitalWrite()` which completely disables the waveform on that pin means probably that the ESP8266 is going crazy due to V drop or overcurrent on an IO pin.  Or you could have a ground/VCC supply issue, even if the ESP8266 is still running, where ripple gets injected.

I know of at least one other project using WiFi and the waveform generator (fixed a bug for them and they verified their weather station was now running fine), so I don't think there is a SW problem here.  I will give it a try and look at the waveform with my logic analyzer to verify, though.

earlephilhower, 2018-08-01:
Using your sketch and substituting my own AP info, here is the upload and tone which lasts for 50ms:

[image]

I didn't see any movement on the Tone() pin after that for a period of over 300 seconds.

Give that, I think you have an analog HW problem unrelated to the core SW.  See my above comments.

earlephilhower, 2018-08-18:
Closing due to lack of feedback.

closed 2018-08-18 by earlephilhower