esp8266/Arduino #4991: Unstable system (probably due to mem corruption)

Raphyyy, 2018-07-31:
Hello

I am getting back to you today with the same issue I opened months ago : https://github.com/esp8266/Arduino/issues/4193

After many talks with @devyte who I am grateful to, I still haven't found the origin of those troubles.

It's obviously some memory corruption problem, but I can't achieve to find where could possibly be the problem. 

**The entiere problem and all of what I tried are well described in the issue I linked above**.

If some of you are interested to helping me to spot those memory problem, I can send you an archive of my code on gitter : https://gitter.im/Raphyyy

I am really sorry to flood here with the same issue, but this concern a professionnal embedded project and I really need to find a solution after months of trouble. 

Best regards

devyte, 2018-07-31:
@Raphyyy you ignored the issue template again. Please edit the post and add in all of the required info.

Raphyyy, 2018-07-31:
Hello devyte
My bad. Template filled.

Raphyyy, 2018-08-10:
@devyte I have lot more mem corruption with > 2.4 so currently the 2.3.0 core run in prod.
I use latest Git, so nothing change from 2.4.1 to 2.4.2.

Some news : I updated all of my code to use stack the less possible. No more char array buffer.
Tested with 2.4.2, 2.4.1, 2.3.0, still the same problems...

devyte, 2018-12-14:
@Raphyyy what is your status with 2.5.0-beta1? At least two fixes related to mem corruption are included, although I don't know if they apply to you.

earlephilhower, 2019-01-16:
Closing as no MCVE and no response.

closed 2019-01-16 by earlephilhower