esp8266/Arduino #4994: mDNS announces services and then disappear after some minutes (v 2.4.1)

daspuru, 2018-08-01:
ESP8266 Arduino Core Version: 2.4.1

Hi, I am trying to register a mDNS service in the network, however the service gets registered and after 10 minutes or less it disappear, and I am not able to re announce it again, only after restarting the device. I also tried with the latest AT firmware with the AT+MDNS command and it behaves exactly the same. 

My development board is a Teensy 3.2 with connected to an ESP8266-12F ( [here you can see the board](https://www.tindie.com/products/cburgess129/arduino-teensy-v44-with-wifibluetoothnrf24l01/?pt=ac_prod_search) ) running a program that transfers from usb to hardware serial port at 115,200 BPS, which seems ok good for programming the ESP8266, as I make the programs be writable to the device.

In the IDE, the board selected is Generic ESP8266, and flash size is 512K (NO SPIFFS)

Here is my code:

    #include <Arduino.h>
    #include <ESP8266WiFi.h>
    #include <ESP8266mDNS.h>

    char hostString[16] = {0};

    long previousMillis = 0;
    long interval = 20000;

    void setup()
    {
      Serial.begin(115200);
      Serial.println();

      WiFi.mode(WIFI_STA);

      sprintf(hostString, "ESP_%06X", ESP.getChipId());
      Serial.print("Hostname: ");
      Serial.println(hostString);
      WiFi.hostname(hostString);

      WiFi.begin("omitted", "omitted");

      Serial.print("Connecting");
      while (WiFi.status() != WL_CONNECTED)
      {
        delay(500);
        Serial.print(".");
      }
      Serial.println();

      Serial.print("Connected, IP address: ");
      Serial.println(WiFi.localIP());

      if (!MDNS.begin(hostString)) {
        Serial.println("Error setting up MDNS responder!");
      }
      Serial.println("mDNS responder started");
      MDNS.addService("osc", "udp", 8000); // Announce esp tcp service on port 8080
      //sendMdns();
  
      Serial.println("Sending mDNS query");
      int n = MDNS.queryService("doc", "tcp"); // Send out query for esp tcp services
      Serial.println("mDNS query done");
      if (n == 0) {
        Serial.println("no services found");
      }
      else {
        Serial.print(n);
        Serial.println(" service(s) found");
        for (int i = 0; i < n; ++i) {
          // Print details for each service found
          Serial.print(i + 1);
          Serial.print(": ");
          Serial.print(MDNS.hostname(i));
          Serial.print(" (");
          Serial.print(MDNS.IP(i));
          Serial.print(":");
          Serial.print(MDNS.port(i));
          Serial.println(")");
        }
      }
      Serial.println();
  
    }

    void loop() 
    {
      //unsigned long currentMillis = millis();

      //if (currentMillis - previousMillis > interval) {
      //  previousMillis = currentMillis;
      //  //sendMdns();
      //}
    }

    void sendMdns() {
      MDNS.addService("esp", "tcp", 8080); // Announce esp tcp service on port 8080
      int n = MDNS.queryService("esp", "tcp"); // Send out query for esp tcp services
      Serial.println("mDNS query done");
      if (n == 0) {
        Serial.println("no services found");
     } else {
        Serial.print(n);
        Serial.println(" service(s) found");
        for (int i = 0; i < n; ++i) {
          // Print details for each service found
          Serial.print(i + 1);
          Serial.print(": ");
          Serial.print(MDNS.hostname(i));
          Serial.print(" (");
          Serial.print(MDNS.IP(i));
          Serial.print(":");
          Serial.print(MDNS.port(i));
          Serial.println(")");
        }
      }
      Serial.println("MDNS SENT");
    }

fabiuz7, 2018-09-06:
Yeah for me the same! Compared to mDNS for ESP32-Arduino is a bit worse.
Moveover, observing the 2 versions simultaneously, the esp8266 one sometimes misses packets, returning 0 services.

devyte, 2018-12-05:
Closing in view of #5442 with a full rewrite of mdns.

closed 2018-12-05 by devyte