esp8266/Arduino #4997: I can only connect to open AP's

balzreber, 2018-08-02:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue. **_There are some similar issues (e.g https://github.com/nodemcu/nodemcu-firmware/issues/1630), but they are not exactly the same in my view._**
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266 ESP-01 / (ESP-12E)
- Core Version:			2.4.0
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			Generic ESP8266 Module / (NodeMCU 1.0 (ESP-12E Module))
- Flash Mode:			qio
- Flash Size: 		        1MB
- lwip Variant:                 v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:            115200

### Problem Description

I can only connect to open AP's with my ESP. As soon as I switch on any wireless security on my router my ESP no longer is able to connect.
I tried with several different ESP-01 and ESP-12E modules, but the problem persists.

I have a Linksys WTP3200ACM running DD-WRT v3.0-r35244 std. I tried to connect to both network cards in this router (Although I know ESP only supports 2.4GHz). I also tried to set the interface to B/G/N only. Of course, I also tested all the security modes WEP/WPA/WPA2 and also the EAS and TKIP WPA algorithms. But the problem persists.

If I set the security mode to disabled (open network) the ESP connects without problem to the exact same router.

Even stranger is when I assign the ESP a fix IP. In this case, even with wireless security on, the ESP acts like it has connected. But I don't see a connection on the router, and I'm not able to ping the fix IP.

I can connect to other Routers, even when security is turned on.

After a lot of hours of trying every possible combination of variables I came to the following conclusions:
Since I can use different ESP's with the exact same result, and I can connect to other AP's, the problem is probably the ESP8266WiFi library in combination with my particular router. And since I can connect when I turn the wireless security off, on the router side, it is probably something with the security settings.
But unfortunately, I'm bound to this router. I can't just use another one.

Any help is much appreciated.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

// Basically a stripped down version of the example sketch

#include <ESP8266WiFi.h>

const char* ssid     = "ssid";
const char* password = "password";

void setup() {
  Serial.begin(115200);
  delay(10);

  WiFi.mode(WIFI_STA);
  WiFi.hostname("ESP8266");

  //IPAddress ip(10, 0, 0, 102);
  //IPAddress gateway(10, 0, 0, 1);
  //IPAddress subnet(255, 255, 255, 0);
  //IPAddress dns(10, 0, 0, 1);
  //WiFi.config(ip, dns, gateway, subnet);

  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  Serial.println("Looping");
}

```

### Debug Messages

When I disable the wireless security on my router
```
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
wifi evt: 1
STA disconnect: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
.state: 3 -> 5 (10)
add 0
aid 1

connected with ssid, channel 9
dhcp client start...
cnt 
wifi evt: 0
........ip:10.0.0.5,mask:255.255.255.0,gw:10.0.0.1
wifi evt: 3
.
WiFi connected
IP address: 
10.0.0.5
Looping
Looping
Looping
```

When I enable wireless security (no matter which mode)
```
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
wifi evt: 1
STA disconnect: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
.state: 3 -> 5 (10)
add 0
aid 6
cnt 

connected with ssid, channel 1
dhcp client start...
wifi evt: 0
....................pm open,type:2 0
..................................................................
```

When I set a fix IP (with wireless security enabled). Remember, it looks like I'm connected here. But I'm not.
```
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
wifi evt: 1
STA disconnect: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
.state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with ssid, channel 1
ip:10.0.0.102,mask:255.255.255.0,gw:10.0.0.1
ip:10.0.0.102,mask:255.255.255.0,gw:10.0.0.1
wifi evt: 0
wifi evt: 3
.
WiFi connected
IP address: 
10.0.0.102
Looping
Looping
Looping
```

When I search for the mac address of the syslog of my router I see the following. Looks good, but no connection is made.
```
Aug 2 11:57:21 ssid daemon.info hostapd: ath1: STA xx:xx:xx:xx:xx:xx MLME: auth request, signal 0 (Accepted)
--
Aug 2 11:57:21 ssid daemon.info hostapd: ath1: STA xx:xx:xx:xx:xx:xx MLME: auth request, signal 0 (Accepted)
Aug 2 11:57:21 ssid daemon.info hostapd: ath1: STA xx:xx:xx:xx:xx:xx MLME: assoc request, signal 0 (Accepted)
Aug 2 11:57:21 ssid daemon.info hostapd: ath1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 1)
Aug 2 11:57:21 ssid daemon.info hostapd: ath1: STA xx:xx:xx:xx:xx:xx RADIUS: starting accounting session 3A942571B26B3F14
Aug 2 11:57:21 ssid daemon.info hostapd: ath1: STA xx:xx:xx:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Aug 2 11:57:35 ssid daemon.info hostapd: ath1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
```

Pablo2048, 2018-08-02:
> Aug 2 11:57:21 ssid daemon.info hostapd: ath1: STA xx:xx:xx:xx:xx:xx RADIUS: starting accounting session 3A942571B26B3F1

It seems like You are using Radius server (WPA Enterprise??) here. There is known problem with this kind of authentication IMHO...

balzreber, 2018-08-02:
You are right. It looks like it. Strangely I haven't activated Radius (or Free Radius as the service is called in dd-wrt). I also have WPA2-Personal activated not Enterprise (or Radius directly)

devyte, 2018-08-02:
I don't have the exact same router, but my router is running dd-wrt v3.something, and I can't reproduce this with latest git.
I suspect some issue with your router.
Also, 2.4.0 is rather old. 
Closing as can't reproduce.

closed 2018-08-02 by devyte

balzreber, 2018-08-02:
I'm sorry, I was using 2.4.1 not 2.4.0. And I just found out that with 2.0.0 the same script works on the same router without a hitch?!