esp8266/Arduino #5004: Problem Downloading ESP8266 Board in IDE

wchen3333, 2018-08-03:
A bit of a noob here, so apologies if this is a simple problem.  I am trying to download the ESP8266 board in Arduino IDE and keep getting an error.

I put this address: (http://arduino.esp8266.com/stable/package_esp8266com_index.json) into the additional boards manager under preferences, which seems to work, however I always get the error below when I go to the Boards manager to try to get the esp8266.  I have tried this on several Windows computers and the same thing happens.  I was able to get this to work on a Mac though.  Any thoughts on what is going wrong?

Error downloading https://github.com/esp8266/Arduino/releases/download/2.4.2/esp8266-2.4.2.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading https://github.com/esp8266/Arduino/releases/download/2.4.2/esp8266-2.4.2.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://github.com/esp8266/Arduino/releases/download/2.4.2/esp8266-2.4.2.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:69)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:102)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more

etc, etc. etc.

5chufti, 2018-08-05:
I just tried the link and an install (on windows) and it worked ok.
Maybe you there was a connection or server problem, did you try again?
Can you download it when you click on the links in your post?

wchen3333, 2018-08-06:
Yes, I've tried it over several days on several computers.  Those links do download outside of Aruduino IDE, is there a way to manually download the board?  Is there some configuration or other thing I need to download?

devyte, 2018-08-06:
I can't reproduce this either.
I suggeest a clean install from scratch of both IDE and board.
Closing.

closed 2018-08-06 by devyte