esp8266/Arduino #5005: boards.txt.py fails in 2.4.2 release package

ccantill, 2018-08-04:
The 2.4.2 release zip doesn't include the `package` directory or the `docs` directory. Trying to run `tools\boards.txt.py --allgen --allowWPS` results in a crash. You need to manually download those files from github to make it work.

earlephilhower, 2018-08-05:
Verified on my own.  The auto-package-generator needs to be updated:
````
$ python tools/boards.txt.py --allgen
Traceback (most recent call last):
  File "tools/boards.txt.py", line 1539, in <module>
    package()
  File "tools/boards.txt.py", line 1300, in package
    os.rename(pkgfname, pkgfname_read)
OSError: [Errno 2] No such file or directory
````

In boards.txt.py it's looking for the JSON file:
````
def package ():

    pkgfname = "package/package_esp8266com_index.template.json"
    pkgfname_read = pkgfname

    checkdir()

    if packagegen:
        pkgfname_read = pkgfname + '.orig'
        if os.path.isfile(pkgfname_read):
            os.remove(pkgfname_read)
        os.rename(pkgfname, pkgfname_read)
...
````

ccantill, 2018-08-05:
With `--allowWPS --boardsgen` it's still looking for the `package/package_esp8266com_index.template.json` file.

d-a-v, 2018-09-04:
Fixed in documentation by #5018

closed 2018-09-04 by d-a-v