esp8266/Arduino #5006: NodeMCU suddenly Exception 29

kizuna21, 2018-08-04:
I try to use nodemcu to send my data to firebase and calculate AC power that equipment use also control which equipment that active. The equipment is household equipment. First I try the control code and have no problem. Also I try the power measure with emonlib and its ok to send the data. After combine the 2 code its work for a moment and suddenly this error appear. I tried to upload each of the code and its still have the error. I dont know why but this error like effecting the other. Please help to fix this problem.

This is the error i got
[serialmonitor code error.txt](https://github.com/esp8266/Arduino/files/2259818/serialmonitor.code.error.txt)

and this is the sketch i use
`
#include <EmonLib.h>

#include <ESP8266WiFi.h>
#include <FirebaseArduino.h>
EnergyMonitor emon1;
EnergyMonitor emon2;

// Set these to run example.
#define FIREBASE_HOST "firebase project host"
#define FIREBASE_AUTH "firebase authentication"
#define WIFI_SSID "wifi id"
#define WIFI_PASSWORD "pass"

int s0 = D5;
int s1 = D4;
int s2 = D3;
int analogPin = A0;
String name;

void setup() {
  Serial.begin(9600);

  // connect to wifi.
  pinMode(D1, OUTPUT);
  pinMode(D2, OUTPUT);
  pinMode(s0, OUTPUT);
  pinMode(s1, OUTPUT);
  pinMode(s2, OUTPUT);
  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("connecting");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("connected: ");
  Serial.println(WiFi.localIP());
  
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.set("Saklar1",1);
  Firebase.set("Saklar2",1);
  emon1.voltage(A0, 234.26, 1.7);  // Voltage: input pin, calibration, phase_shift
  emon1.current(A0, 111.1); 

  emon2.voltage(A0, 234.26, 1.7);  // Voltage: input pin, calibration, phase_shift
  emon2.current(A0, 111.1); 
}

int n = 1;
int a = 1;

void loop() {
  n=Firebase.getInt("Saklar1");
  a=Firebase.getInt("Saklar2");
  // controling on off
  if (n==1 && a==1) {
      digitalWrite(D1, HIGH);
      digitalWrite(D2, HIGH);
      
  }
  else if(n==1 && a==0){
      digitalWrite(D1, HIGH);
      digitalWrite(D2, LOW);
      
      }
      
      else if(n==0 && a==1){
      digitalWrite(D1, LOW);
      digitalWrite(D2, HIGH);
      //saklar1;
      }
      else{
      digitalWrite(D1, LOW);
      digitalWrite(D2, LOW);
      //saklar1;
      
      }
}
   //saklar 1 kirim data
void saklar1(){
  digitalWrite(s0,LOW);
  digitalWrite(s1,LOW);
  digitalWrite(s2,LOW);
  if(D1,HIGH){
  emon1.calcVI(20,400);  
  float tegangan1 = emon1.Vrms; 
  float arus1 = emon1.Irms;
  float daya1 = emon1.Vrms * emon1.Irms;
          name=Firebase.pushFloat("Arus 1", arus1);
              if (Firebase.failed()) {
              Serial.print("pushing /Arus 1 failed:");
              Serial.println(Firebase.error());  
              return;
              }
          Serial.print("pushed: /Arus 1/");
           Serial.println(name);
           delay(100);

  digitalWrite(s0,HIGH);
  digitalWrite(s1,LOW);
  digitalWrite(s2,LOW);
          name=Firebase.pushFloat("Tegangan 1", tegangan1);
          if (Firebase.failed()) {
          Serial.print("pushing /Tegangan 1 failed:");
          Serial.println(Firebase.error());  
          return;
          }
          Serial.print("pushed: /Tegangan 1/");
          Serial.println(name);
          delay(100);
  
          name=Firebase.pushFloat("Daya 1", daya1);
              if (Firebase.failed()) {
              Serial.print("pushing /Daya 1 failed:");
              Serial.println(Firebase.error());  
              return;
              }
          Serial.print("pushed: /Daya 1/");
          Serial.println(name);
}
}
  //saklar 2 kirim data
 void saklar2 (){
  digitalWrite(s0,LOW);
  digitalWrite(s1,HIGH);
  digitalWrite(s2,LOW);
  
  emon2.calcVI(20,400); 
  float tegangan2 = emon2.Vrms;
  float daya2 = emon2.Vrms * emon2.Irms;
  float arus2 = emon2.Irms;
   name=Firebase.pushFloat("Arus 2", arus2);
   if (Firebase.failed()) {
      Serial.print("pushing /Arus 2 failed:");
      Serial.println(Firebase.error());  
      return;
  }
  Serial.print("pushed: /Arus 2/");
  Serial.println(name);
  delay(100);

  digitalWrite(s0,HIGH);
  digitalWrite(s1,HIGH);
  digitalWrite(s2,LOW);
  name=Firebase.pushFloat("Tegangan 2", tegangan2);
   if (Firebase.failed()) {
      Serial.print("pushing /Tegangan 2 failed:");
      Serial.println(Firebase.error());  
      return;
  }
  Serial.print("pushed: /Tegangan 2/");
  Serial.println(name);
  delay(100);
  
  name=Firebase.pushFloat("Daya 2", daya2);
   if (Firebase.failed()) {
      Serial.print("pushing /Daya 2 failed:");
      Serial.println(Firebase.error());  
      return;
  }
  Serial.print("pushed: /Daya 2/");
  Serial.println(name);
  }
`

this is the decode from the error
[error.txt](https://github.com/esp8266/Arduino/files/2259808/error.txt)

and this is the code that i use before and its work before but now not work suddenly
`#include <ESP8266WiFi.h>
#include <FirebaseArduino.h>
#define FIREBASE_HOST "smarthome-6af1d.firebaseio.com"
#define FIREBASE_AUTH "5fAOnlvMGJg33Gv26VxuYMDPL4F2HfpmxuOlvQ2y"
#define WIFI_SSID "wifi id"
#define WIFI_PASSWORD "wifi password"
void setup() {
  Serial.begin(9600);

  // connect to wifi.
  pinMode(D1, OUTPUT);
  pinMode(D2, OUTPUT);
  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("connecting");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("connected: ");
  Serial.println(WiFi.localIP());
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.set("Saklar1",1);
  Firebase.set("Saklar2",1);
int n = 1;
int a = 1;

void loop() {
  n=Firebase.getInt("Saklar1");
  a=Firebase.getInt("Saklar2");
  // kontroling on off
  if (n==1 && a==1) {
      digitalWrite(D1, HIGH);
      digitalWrite(D2, HIGH);
      
  }
  else if(n==1 && a==0){
      digitalWrite(D1, HIGH);
      digitalWrite(D2, LOW);
      
      }
      
      else if(n==0 && a==1){
      digitalWrite(D1, LOW);
      digitalWrite(D2, HIGH);
    
      }
      else{
      digitalWrite(D1, LOW);
      digitalWrite(D2, LOW);
 
      
      }
}`

kizuna21, 2018-08-04:
sorry if the code show like this. This is my first time using github so please sorry again if it look like this

5chufti, 2018-08-05:
You did see the issue template when opening the iseue???
It is not to be ignored if you realy want an answer!
--> go back to start

kizuna21, 2018-08-05:
ok i tried again later

devyte, 2018-08-05:
Several comments:
- you ignored the issue template and the instructions there
- from your log I see you're using release 2.4.1, which has several known issues that have since been fixed. I suggest 2.4.2, or latest git.
- by default, secure sockets use axtls lib, which is no longer maintained by the authors. It is planned for deprecation and then retirement in this core. You should figure out how tonuse Firebase with bearssl lib, which is recently integrated and so far has proven to be far more reliable.
- you may be running into oom for some reason. Build with oom debug enabled, which will show an error message over serial on oom.

Closing.

closed 2018-08-05 by devyte