esp8266/Arduino #5008: IDE 1.8.5 recompiles every time after update to core 2.4.2

philbowles, 2018-08-05:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:				All
- Core Version:			[2.4.2]
- Development Env:		[Arduino IDE 1.8.5]
- Operating System:		[Windows10]

### Settings in IDE

N/A, apllies to all

### Problem Description

I'm sure I have done nothing to any settings in the IDE (1.8.5) but since updating to esp8266 core 2.4.2 everything is recompiled on a build, every time even if it has not been touched.

Not a "showstopper" but it sure slows down my dev cycle. Please tell me it's nothing something daft I have done...

On the plus side, I am now able to compile a huge program that just grew to busting memory limits, so well done for clawing back lots of heap/ram etc, it saved my project, thanks!

Pablo2048, 2018-08-05:
Probably same as https://github.com/esp8266/Arduino/issues/4646 ?

5chufti, 2018-08-05:
I use 1.8.6 hourly and it works well
```

D:\arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=esp8266:esp8266_242:generic:CpuFrequency=80,VTable=flash,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=qio,FlashSize=512K0,led=2,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200 -ide-version=10806 -build-path C:\DOKUME~1\ADMINI~1\LOKALE~1\Temp\arduino_build_318160 -warnings=all -build-cache C:\DOKUME~1\ADMINI~1\LOKALE~1\Temp\arduino_cache_732330 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=D:\arduino\portable\packages\esp8266\tools\esptool\0.4.13 -prefs=runtime.tools.mkspiffs.path=D:\arduino\portable\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=D:\arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -verbose C:\DOKUME~1\ADMINI~1\LOKALE~1\Temp\untitled159252224.tmp\sketch_aug05a\sketch_aug05a.ino
D:\arduino\arduino-builder -compile -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=esp8266:esp8266_242:generic:CpuFrequency=80,VTable=flash,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=qio,FlashSize=512K0,led=2,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200 -ide-version=10806 -build-path C:\DOKUME~1\ADMINI~1\LOKALE~1\Temp\arduino_build_318160 -warnings=all -build-cache C:\DOKUME~1\ADMINI~1\LOKALE~1\Temp\arduino_cache_732330 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=D:\arduino\portable\packages\esp8266\tools\esptool\0.4.13 -prefs=runtime.tools.mkspiffs.path=D:\arduino\portable\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=D:\arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -verbose C:\DOKUME~1\ADMINI~1\LOKALE~1\Temp\untitled159252224.tmp\sketch_aug05a\sketch_aug05a.ino
Using board 'generic' from platform in folder: D:\arduino\portable\packages\esp8266\hardware\esp8266_242\2.4.2
Using core 'esp8266' from platform in folder: D:\arduino\portable\packages\esp8266\hardware\esp8266_242\2.4.2
Detecting libraries used...
"D:\\arduino\\portable\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-ID:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266_242\\2.4.2/tools/sdk/include" "-ID:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266_242\\2.4.2/tools/sdk/lwip2/include" "-ID:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266_242\\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\DOKUME~1\\ADMINI~1\\LOKALE~1\\Temp\\arduino_build_318160/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DARDUINO=10806 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266_242 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DESP8266 "-ID:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266_242\\2.4.2\\cores\\esp8266" "-ID:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266_242\\2.4.2\\variants\\generic" "C:\\DOKUME~1\\ADMINI~1\\LOKALE~1\\Temp\\arduino_build_318160\\sketch\\sketch_aug05a.ino.cpp" -o nul
Generating function prototypes...
"D:\\arduino\\portable\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-ID:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266_242\\2.4.2/tools/sdk/include" "-ID:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266_242\\2.4.2/tools/sdk/lwip2/include" "-ID:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266_242\\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\DOKUME~1\\ADMINI~1\\LOKALE~1\\Temp\\arduino_build_318160/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DARDUINO=10806 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266_242 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DESP8266 "-ID:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266_242\\2.4.2\\cores\\esp8266" "-ID:\\arduino\\portable\\packages\\esp8266\\hardware\\esp8266_242\\2.4.2\\variants\\generic" "C:\\DOKUME~1\\ADMINI~1\\LOKALE~1\\Temp\\arduino_build_318160\\sketch\\sketch_aug05a.ino.cpp" -o "C:\\DOKUME~1\\ADMINI~1\\LOKALE~1\\Temp\\arduino_build_318160\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"D:\\arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\DOKUME~1\\ADMINI~1\\LOKALE~1\\Temp\\arduino_build_318160\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
[... cut ...]
"D:\\arduino\\portable\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-size" -A "C:\\DOKUME~1\\ADMINI~1\\LOKALE~1\\Temp\\arduino_build_318160/sketch_aug05a.ino.elf"
Sketch uses 246052 bytes (49%) of program storage space. Maximum is 499696 bytes.
Global variables use 27552 bytes (33%) of dynamic memory, leaving 54368 bytes for local variables. Maximum is 81920 bytes.
```

maybe activate "agressively caching compiled core"  in preferences (if av in 1.8.5)

philbowles, 2018-08-05:
Thanks for the reference. I read it. I'mdon't think so - I have been using 1.8.5 for a long time, I would have noticed as my main poject is 415k and takes a fair while to compile. A silly error and a recomp takes a few seconds. I installed 2.4.2 yesterday and now wvery compile takes a couple of minutes

philbowles, 2018-08-05:
@shufti - if there IS such an option in 1.8.5 I don't know where it is or how to set it, so I don't think that's it either

5chufti, 2018-08-05:
if it is not there in 1.8.5 -->  good cause to switch to 1.8.6

philbowles, 2018-08-05:
```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\phil\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\phil\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\phil\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:CpuFrequency=80,VTable=flash,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=921600 -ide-version=10805 -build-path C:\Users\phil\AppData\Local\Temp\arduino_build_225985 -warnings=none -build-cache C:\Users\phil\AppData\Local\Temp\arduino_cache_125192 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.13 -verbose C:\Users\phil\Desktop\20_BlinkyMQTT\20_BlinkyMQTT.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\phil\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\phil\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\phil\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:CpuFrequency=80,VTable=flash,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=921600 -ide-version=10805 -build-path C:\Users\phil\AppData\Local\Temp\arduino_build_225985 -warnings=none -build-cache C:\Users\phil\AppData\Local\Temp\arduino_cache_125192 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.13 -verbose C:\Users\phil\Desktop\20_BlinkyMQTT\20_BlinkyMQTT.ino
Using board 'd1_mini' from platform in folder: C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2
Using core 'esp8266' from platform in folder: C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2
Build options changed, rebuilding all
Detecting libraries used...
"C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/include" "-IC:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include" "-IC:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\phil\AppData\Local\Temp\arduino_build_225985/core" -c -w  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"   -DESP8266 "-IC:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266" "-IC:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\variants\d1_mini" "C:\Users\phil\AppData\Local\Temp\arduino_build_225985\sketch\20_BlinkyMQTT.ino.cpp" -o "nul"
"C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/include" "-IC:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include" "-IC:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\phil\AppData\Local\Temp\arduino_build_225985/core" -c -w  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"   -DESP8266 "-IC:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266" "-IC:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\variants\d1_mini" "-IC:\Users\phil\Documents\Arduino\libraries\ESPArto\src" "C:\Users\phil\AppData\Local\Temp\arduino_build_225985\sketch\20_BlinkyMQTT.ino.cpp" -o "nul"
"C:\Users\phil\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\phil\AppData\Local\Arduino15\packages\esp8266\h [... cut ...] :\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\Ticker" "-IC:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266mDNS" "-IC:\Users\phil\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\Hash\src" "C:\Users\phil\Documents\Arduino\libraries\ESPAsyncWebServer\src\SPIFFSEditor.cpp" -o "C:\Users\phil\AppData\Local\Temp\arduino_build_225985\libraries\ESPAsyncWebServer\SPIFFSEditor.cpp.o"
...
...
```

devyte, 2018-08-05:
This is an IDE issue, not a core issue. It can get triggered by core code changes.
It was found that IDE 1.9b seems to have solved it.
Closing.

marcelstoer, 2018-08-12:
> This is an IDE issue, not a core issue.

That's a rather bold statement when we already know that IDE 1.8.5 and core 2.4.1 did not exhibit this behavior. But yeah, 1.8.6 hourly does make it go away as well.

mhightower83, 2018-08-12:
Unless I am somehow fooling myself - again, I narrowed it down last night to the addition of this line in platform.txt:
```
## Build the app.ld linker file
recipe.hooks.linking.prelink.1.pattern="{compiler.path}{compiler.c.cmd}" -CC -E -P {build.vtable_flags} "{runtime.platform.path}/tools/sdk/ld/eagle.app.v6.common.ld.h" -o "{runtime.platform.path}/tools/sdk/ld/eagle.app.v6.common.ld"
```
I was able to work around the problem by back dating the file it created. I added this line to platform.txt:
```
recipe.hooks.linking.prelink.2.pattern=touch -t 170001010000 "{runtime.platform.path}/tools/sdk/ld/eagle.app.v6.common.ld"
```

devyte, 2018-08-12:
@marcelstoer said:
> we already know that IDE 1.8.5 and core 2.4.1 did not exhibit this behavior

That's not really true. The same behavior was observed and was reported here with 2.4.1, and even with 2.4.0, although less often. 
As I've explained elsewhere, I don't know all the details here, but the issue seems to get triggered by something in our core, yet our core is not the cause. Based on the previous comment it would appear to be related to how the generation of some of the .ld files is done, and that makes sense given some of the link time voodoo that has been added in our last releases.

I believe it was with 2.4.1 that @d-a-v discovered that the rebuild of everything didn't happen with IDE 1.9b, and now also with the imminent 1.8.6. Once 1.8.6 is out we'll move to that as our minimum recommendation for the IDE version.

closed 2018-08-20 by igrr

cstratton, 2019-03-02:
Come on, this is still broken, don't declare it close!

cstratton, 2019-03-02:
This nonsense makes it all but _impossible_ to program boards since the loader invariable times out before the several minute needless recompilation is done!!!

5chufti, 2019-03-02:
which loader gets started before the recompilation?
the esp bootloader has no timeout and the espflasher only gets started after the recompilation is done.
If any loader times out on you during recompilation, there is something very strange going on with your setup ...
I tested with IDE 1.8.6 and core 2.4.2 and have no problem with recompiling as can be seen in attached picture. 
[image]

PR #5043 (2018-08-20): build: place generated ld script into build directory
arduino-builder 1.3.25 (shipped with Arduino 1.8.5) forces full
recompilation when any file in the core directory is modified. Avoid
full recompilation by placing generated ld script into build
directory, not source directory.

Also fix an issue where git version description would not be generated
if there were spaces in build path.

Closes #5008.