esp8266/Arduino #5010: mDNS between two ESPs

iVisionary, 2018-08-05:
Hello Dears, 

One ESP runs mDNS with host name "esp8266" and works perfectly when trying to access it from web browser on my mobile phone. 

Tried to access it from another ESP but didn't work! then I tried to use the IP address of the server instead of the host name and it worked perfectly. 

The issue is, why when using the host name it didn't work? what do I need to do? 

**Server Code:** 
`#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

const char* ssid = "myWiFi";
const char* password = "myPassword";

ESP8266WebServer server(80);

const int led = 2;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp8266!");
  digitalWrite(led, 0);
}

void turnOff(){
  digitalWrite(led, 1);
  server.send(200, "text/plain", "LED OFF");
}

void turnOn(){
  digitalWrite(led,0);
  server.send(200, "text/plain", "LED ON");
}

void toggle(){
  digitalWrite(led,!digitalRead(led));
  if(digitalRead(led))
  server.send(200, "text/plain", "LED ON");
  else 
  server.send(200, "text/plain", "LED OFF");
}

void handleNotFound(){
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void){
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }

   server.on("/", handleRoot);
   server.on("/turnON", turnOn);
   server.on("/turnOFF", turnOff);
   server.on("/toggle", toggle);

   server.onNotFound(handleNotFound);

   server.begin();
   Serial.println("HTTP server started");
 }

 void loop(void){
   server.handleClient();
 }`

**Client Code:** 

`#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266mDNS.h>
 
const char* ssid = "myWiFi";
const char* password = "myPassword";
 
void setup () {
 
  Serial.begin(115200);
  WiFi.begin(ssid, password);
 
  while (WiFi.status() != WL_CONNECTED) {
 
    delay(1000);
    Serial.print("Connecting..");
 
  }
 
}
 
void loop() {
 
  if (WiFi.status() == WL_CONNECTED) { //Check WiFi connection status
 
    HTTPClient http;  //Declare an object of class HTTPClient
    Serial.println();
    
    String hostString = "esp8266";
    uint16_t port = 80;
    String Uri = "/turnOFF";
    
    http.begin("http://esp8266.local/toggle");  //Specify request destination
    int httpCode = http.GET();  //Send the request
 
    if (httpCode > 0) { //Check the returning code
 
      String payload = http.getString();   //Get the request response payload
      Serial.println(payload);                     //Print the response payload
 
    }
 
    http.end();   //Close connection
 
  }
 
  delay(3000);    //Send a request every 3 seconds
 
}`

iVisionary, 2018-08-13:
@RudyFiero Thank you VERY much it's working now perfectly. thank you for your time and efforts.

devyte, 2018-12-05:
Closing in view of #5442 with a full rewrite of mdns.

closed 2018-12-05 by devyte