esp8266/Arduino #5011: ESP8266WebServer slow / not responding

YordanYanakiev, 2018-08-06:
INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			[qio|dio|other]
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		OTA
- Upload Speed:   115200

### Problem Description

I am droping the WebServer example on my Arduino IDE, and compile and upload to my NodeMCU v3.
When I am trying to access the web server from my desktop PC browser - the respond 1st time is about a second or two. After a while it becomes not responsive or skip requests. 
Some time it is not responding to the client for a few seconds to 1-2 minutes, then it goes back, and then again falls in this state.

the WebServer example :
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/HelloServer/HelloServer.ino

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

const char* ssid = "........";
const char* password = "........";

ESP8266WebServer server(80);

const int led = 13;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp8266!");
  digitalWrite(led, 0);
}

void handleNotFound() {
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void) {
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient();
}

### Debug Messages

Nickrader, 2018-08-12:
Line 62: server.on("/inline", []() {
*differs from expample, caused problem for my compilation.  Linked example compiles fine, ty

Assuming you are entering your SSID, Password instead of all periods:

1) Then you'll want to check your hardware first.  Does problem replicate when using 5v phone charger/power bank to provide power?
2) Can the ESP be seen in your router's Attached Devices list?  Can you Ping it?
3) "the respond 1st time is about a second or two." This tells me you get no errors compiling, get the success string "hello from esp8266!" the first time you load esp8266.local/ 
4) If after re-compile/flash still have same error? List could be dependent on troubleshooting steps taken.

WiFi uses much more power than simple operation.  It is the source of many problems. After you rule out power supply, then focus other hardware/software.

thephantomphoenix, 2019-04-04:
Hi,

I thought I would just add to this for what it is worth.

I have been working on an ESP8226 for the best part of today finalising the marriage of the BearSSL webserver, wifi-manager and BearSSL connection to AWS IOT. With the last part of wifi-manager just not being consistent and the webserver being slow. I have been coding and testing this from my USB connection via the programmer.

Coming across this post reminded me of a similar issue I had ages ago. So, I just plugged the esp into the power source it will live on, and all of the niggly dramas just went away. 

So, yeah, power power power. When in doubt check power. If things are inconsistent, check power.

Anyway, hope it helps.

earlephilhower, 2019-08-21:
Closing due to the fact that this looks like the same issue as was seen with the pre-SDK version, which was removed from the tree after discovering it would freeze for 100s of milliseconds, randomly.  Also, the bug discussion seems to have moved to power issues, kind of far from the original issue.

If GIT head has the same problem, please open another bug so we can track it.

closed 2019-08-21 by earlephilhower