esp8266/Arduino #5012: Itead Wee modules unable to connect to AP

dberenguer, 2018-08-06:
### Basic Infos

- WiFiClient sample code works well on a lot of ESP-12E modules tested in my office but never on Itead's Wee modules.
- Only Wee modules are not able to connect to the AP. I tested with both WiFi security enabled and disabled

#### Platform

- Hardware:				Itead Wee modules: https://www.itead.cc/wee-serial-wifi-module.html
- Core Version:			Package 2.4.2
- Development Env:		Arduino IDE
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			Generic ESP8266 board
- Flash Mode:			DIO
- Flash Size: 		        512K
- lwip Variant:                  v2 Lower Memory (all options tested)
- Reset Method:		ck
- Flash Frequency:	        40Mhz
- CPU Frequency:	        80Mhz
- Upload Using:		SERIAL
- Upload Speed:             115200

### Problem Description

I realized that the code I had been using on hundreds of ESP-12E modules did not work correctly on any of the Itead Wee modules that I have in my office. In order to isolate from a possible code error I then tried with the WiFiClient sample code provided with the WiFi library but again the Wee modules never connect to my AP. I tried with different AP's, security enabled and disabled. I've also tried with many different board definitions and different lwIP variants.

I initially thought that this might have to do with some incompatibility with the current WiFi stack. On the other hand, I'm reading success stories about the old ESP-01 modules being programmed with the latest Arduino cores. 

Of course, WiFiClient works like a charm on common NodeMCU and ESP-12E boards.

### WiFiClient.ino sample sketch

```cpp

#include <ESP8266WiFi.h>

const char* ssid     = "myssid";
const char* password = "mypassword";

const char* host = "data.sparkfun.com";
const char* streamId   = "....................";
const char* privateKey = "....................";

void setup() {
  Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  /* Explicitly set the ESP8266 to be a WiFi-client, otherwise, it by default,
     would try to act as both a client and an access-point and could cause
     network-issues with your other WiFi-devices on your WiFi-network. */
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

int value = 0;

void loop() {
  delay(5000);
  ++value;

  Serial.print("connecting to ");
  Serial.println(host);

  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }

  // We now create a URI for the request
  String url = "/input/";
  url += streamId;
  url += "?private_key=";
  url += privateKey;
  url += "&value=";
  url += value;

  Serial.print("Requesting URL: ");
  Serial.println(url);

  // This will send the request to the server
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "Connection: close\r\n\r\n");
  unsigned long timeout = millis();
  while (client.available() == 0) {
    if (millis() - timeout > 5000) {
      Serial.println(">>> Client Timeout !");
      client.stop();
      return;
    }
  }

  // Read all the lines of the reply from server and print them to Serial
  while (client.available()) {
    String line = client.readStringUntil('\r');
    Serial.print(line);
  }

  Serial.println();
  Serial.println("closing connection");
}

```

### Debug Messages

onnecting to Panstamp
scandone
...wifi evt: 2
..scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
wifi evt: 1
STA disconnect: 2
......scandone
state: 0 -> 2 (b0)
..state: 2 -> 0 (2)
reconnect
wifi evt: 1
STA disconnect: 2

dberenguer, 2018-08-07:
Thanks @devyte

duncan-a, 2018-08-07:
Did the suggestions fix the problem???

Closed prematurely, I think...

or mute
the thread

dberenguer, 2018-08-07:
@duncan-a the list suggested by @devyte has been carefully followed but, as he says, this seems to be a problem related to hardware. The ESP8266 firmware cores seem to have been evolving beyond the capabilities of older ESP8266 platforms. I naively purchased some dozens of this Wee board without considering this potential limitation but I now understand that moving to a better supported platform (ex: ESP-12E) is my best option.

duncan-a, 2018-08-07:
OK...

While it makes sense if you're linked to selling modules, many people
(myself included) got on board early with the ESP-01 (which is often more
than adequate for many things) and now we are being ostracized and expected
to consign the modules we have, to the bin.

I'm not aware of old (compatible) firmware being easily available anywhere
- certainly not from a link that can be added to the Arduino IDE.

Maybe I'm just old and believe in fair play for the early supporters of the
ESP8266...

Why couldn't it be possible to have 'less than 4Mb' option in the boards
manager, looking at two versions, and keep everybody happy?

igrr, 2018-08-07:
@dberenguer Could you please try erasing the flash (using esptool.py or esptool-ck) of the module, and then flashing the example again? It is possible that these modules were previously flashed with an application based on older SDK, and as such they have some WiFi configuration data written to flash in old format. IIRC the SDK does not handle updates of the configuration data between 1.5.x and 2.x.x versions very well. Erasing contents of flash usually fixes these issues.

dberenguer, 2018-08-07:
@igrr I have done that test multiple times: erasing the whole flash from esptool.py and also selecting "Erasing all flash contents" from Arduino Tools before uploading the code. I'd say this is not really related to memory size but mainly to old ESP8266 chips relying on deprecated SDK's.

igrr, 2018-08-07:
Okay, thanks for the test.

> I'd say this is not really related to memory size but mainly to old ESP8266 chips relying on deprecated SDK's.

That sounds quite odd, as every ESP8266 chip ever produced is exactly the same, there have been no changes to the chip design, or to the ROM code. So the chip is not tied to any specific SDK version.

Just to check whether this might be an Arduino thing, have you tried flashing these modules with AT firmware or NodeMCU or Micropython? If you can get them to connect to WiFi with either of these firmwares, then we can potentially troubleshoot this.

dberenguer, 2018-08-07:
> Just to check whether this might be an Arduino thing, have you tried flashing these modules with AT firmware or NodeMCU or Micropython? If you can get them to connect to WiFi with either of these firmwares, then we can potentially troubleshoot this.

I was unable to make the latest AT firmware work on these modules. They were supplied with a version of the AT firmware that worked but flashing different images for different memory sizes always resulted on weird characters being printed on the terminal. I tried with multiple serial baudrates of course. Default AT firmware supplied by iTead does connect to the WiFi AP.

igrr, 2018-08-07:
> Default AT firmware supplied by iTead does connect to the WiFi AP.

Do you have a link to the binary? Can't find it on their wiki...

dberenguer, 2018-08-07:
From their wiki page and referring to NodeMCU flasher:
"This software contains an existing firmware"
And that's all about uploading new firmware to the board.

devyte, 2018-08-07:
@dberenguer maybe I should explain better. The issue here is not that newer devices have changed hardware somehow. It's just that the newer SDK versions from Espressif have grown in size, as has our own core.
Espressif stopped supporting the 512K flash devices, because they couldn't assure that all features would work in it. Most prominently OTA updates, because OTA forces the limitation that the binary size must be at most half the size of the flash size, in that case 256K in size.
Since quite some time ago, an almost empty app built with their tool chain is already bigger than 256K, so OTA won't work, at least not safely.
The same is true of our core, which adds code: An empty sketch is already bigger than 256K.
It is the same with other frameworks, e.g.: Lua: they stopped supporting the older devices in newer releases for the same reason.
Today, ESP-01s come with 1M, and that is tight already.

Having said all of the above, it doesn't mean that using the core with older devices is impossible. As @igrr said, the HW is essentially identical, except for the flash size. It just means that the onus is on you to figure it out.

In addition to the points in my previous comment:
- If this is like the original ESP-01, you have to put it in flash mode first, then flash, then physically reset and go back to normal mode
- users used to the ESP12 tend to have difficulty playing with the original ESP01. I suggest studying the schematics for similarities/differences, especially how the GPIOs are connected or pulled.

igrr, 2018-08-07:
Schematic on Itead wiki lists w25q80bvsig flash chip, which is an 8 Mbit (1 MB) part.

I couldn't get the firmware from their download tool, since it's a windows .exe file. If anyone can upload a copy of itead default firmware (which according to @dberenguer can connect to WiFi) that would be appreciated.

duncan-a, 2018-08-07:
I follow the theory but I have many ESP-01 modules (with small Flash)
0, happily running the Xose Perez "Espurna" firmware with OTA, installed
under Arduino IDE (no idea which version) and they have been, and are, 100%
reliable.

dberenguer, 2018-08-07:
@igrr binaries seem to be located here:
https://github.com/nodemcu/nodemcu-flasher/tree/master/Resources/Binaries

@devyte OK, your argument about the lack of space makes total sense. It's simply that the whole application seems to be working (serial printouts, AP mode) and only connecting to an external AP is failing.

dberenguer, 2018-08-09:
@devyte On the other hand:

"Sketch uses 260496 bytes (25%) of program storage space. Maximum is 1023984 bytes."

WifiClient on ESP8266 with 1Mb of flash. No warning is being raised due to lack of space.

dberenguer, 2018-08-09:
Today I forgot the board running on my table and when I came back I fount it... connected!! I've been however unable to reproduce this :-(