esp8266/Arduino #5014: Arduino/libraries/ESP8266WiFi/examples/HTTPSRequestCACert/CACert.ino

ehitil, 2018-08-07:
Arduino/libraries/ESP8266WiFi/examples/HTTPSRequestCACert/CACert.ino

How can I change the ca certificate to the "CACert. Ino" file?

MarcelBooms, 2018-08-07:
How I would do it:
I assume you have the root CA stored in a file. If not, extract the root CA certificate from the keystore (for example a java cacerts file). The certificate may be encoded as binary DER or as ASCII PEM. The CER and CRT extensions are nearly synonymous.  Most common among *nix systems. 
According to the HTTPSRequestCACert example, the certificate should be stored in DER encoded format. Use commands like openssl to convert the certificate file file to a DER encoded file.

Write a small program that reads the file per byte and outputs ascii strings like the certifcate in CACert.ino.  Perhaps someone else already wrote a program for this purpose.

devyte, 2018-08-07:
Issue template ignored. BTW,  the first line of the template explicitly says that this is not the right place to ask "How do I" or similar type questions.
Closing.

closed 2018-08-07 by devyte