esp8266/Arduino #5016: sprintf does not format float/double on esp8266 (with Arduino ARM Zero/M0 and Due/M3 it works though).

dsyleixa, 2018-08-07:
----------------------------- Delete below -----------------------------
spoiler:
this form is annoying for reporting simple issues!
/spoiler
----------------------------- Delete above -----------------------------

### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12E]
- Core Version:			[latest git hash]
- Development Env:		[Arduino IDE|]
- Operating System:		[Windows|

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mh
- Upload Using:		SERIAL]
- Upload Speed:   [115200|] (serial upload only)

### Problem Description

sprintf does not format float/double on esp8266 (with Arduino ARM Zero/M0 and Due/M3 it works though).
Can that be fixed please?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include "Adafruit_MCP9808.h"

#include "SH1106.h" // alias for `#include "SH1106Wire.h"`
 SH1106 display(0x3c, D2, D1);

// Create the MCP9808 temperature sensor object
Adafruit_MCP9808 tempsensor = Adafruit_MCP9808();

void setup() {
  display.init();
  display.flipScreenVertically();
  display.setFont(ArialMT_Plain_10);
  Serial.begin(115200);
}

void loop() {
  char sbuf[100];

  float c = tempsensor.readTempC();
  float f = c * 9.0 / 5.0 + 32;
  Serial.print("Temp: "); Serial.print(c); Serial.print("*C\t"); 
  Serial.print(f); Serial.println("*F"); 

  String tLine="";
  tLine += "Temp: " + (String)c + "*C  " + (String)f + "*F";
  sprintf(sbuf, "Cels.:%f   Fahrh:%f", c,f);  // <~~~~~~~~~~~~~~ fails
  display.clear();
  display.drawString(0, 0, "MCP9808 demo");
  display.drawString(0,20, tLine);
  //display.drawString(0,30, sbuf);
  display.display();
  
  //Serial.println("Shutdown MCP9808.... ");
  //tempsensor.shutdown(); // shutdown MSP9808 - power consumption ~0.1 mikro Ampere
  
  delay(1000);
}

```

### Debug Messages

```
// sbuf ouput on display and on serial:

Cels.:%f   Fahrh:%f
```

devyte, 2018-08-07:
This:
```cpp
#include <Arduino.h>

void setup() 
{
  Serial.begin(115200);
  last = millis();
}

void loop() 
{
  int curr = millis();
  if(curr-last > 1000)
  {
    last = curr;
    char sbuf[100]; // <~~~~ same size
    float c = 20.54;
    float f = 56.28;
    sprintf(sbuf, "Cels.:%f   Fahrh:%f", c,f);  // <~~~~ same line 
    Serial.println(sbuf);
  }
}
```

Produces this output every second for me:
`Cels.:20.540001   Fahrh:56.279999`

So I'd say that float formatting is correct.

devyte, 2018-08-07:
The form may be annoying, but it is *very* necessary for us maintainers to understand what you mean, and a user opening an issue needs to take it seriously. For example, you wrote this:
> Core Version:	[latest git hash]

Is it really true that you're using latest git of the core? or are you using a release version installed via board manager? If you did in fact install latest git, did you uninstall all versions of the core installed via board manager? In the past there was no support for float/doubles in printf, but that was a long time ago.
My code above is with latest git, and should also work at least with 2.4.2 and 2.4.1 releases.

dsyleixa, 2018-08-08:
I have to admit that there was a newer core but it was not listed beyond a "version list" button, it only appeared beyond "info" where I didn't expect a version list.
I now installed the latest core, but now it doesn't compile at all any more, I repeatedly (dozens of times, even at IDE restart) get the error:

C:\Program Files (x86)\Arduino\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar: unable to rename 'C:\Users\user\AppData\Local\Temp\arduino_build_954053/arduino.ar'; reason: File exists

Pablo2048, 2018-08-08:
Why did you install Arduino IDE as portable to c:/Program Files (0x86)? It makes no sense because of Administrator rights need to modify c:/Program Files folder. Try to install everything into another path.

dsyleixa, 2018-08-08:
thank you,  you are the 1st one ever who mentioned that!

closed 2018-08-08 by dsyleixa