esp8266/Arduino #5021: website doesn't show up with core 2.4.2, but correctly with 2.3.0 though

dsyleixa, 2018-08-09:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12E]
- Core Version:			[latest git hash]
- Development Env:		[Arduino IDE|]
- Operating System:		[Windows|

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mh
- Upload Using:		SERIAL]
- Upload Speed:   [115200|] (serial upload only)

### Problem Description

the following function provides a website to login (username, password). It is called if the website url is loaded and a global variable "authorized" is still "false".
 If  login is correct, the global variable "authorized" is set to true, and the full website is displayed, 
Admittedly surely not perfect, it actually works fine with core 2.3.0, 
but now as I have changed to core 2.4.2  this website does not appear any more, 
instead an error html website appears claiming: website is unreachable.
Switching back to core 2.3.0 and recompiling, it's fine again.

What do I miss about the new core or the required new coding?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

void handleNotAuthorized() {
  String readString="";
  char   strinput[MAXLEN], strpwd[MTKLEN], strusr[MTKLEN];

  WiFiClient client = wifiserver.available();

  //---------------------------------------

  strcpy(strinput, "");
  strcpy(strpwd, "");
  strcpy(strusr, "");

  while ( client.connected() ) {
    if (authorized) return;
    
    readString.toCharArray(strinput, MAXLEN);
    
    cstringarg(strusr, strinput, "uname");  // uname
    cstringarg(strpwd, strinput, "upwd");   // upwd
    

    if ( strlen(strpwd) == strlen(website_pwd)
         && strncmp(strpwd, website_pwd, strlen(website_pwd) ) == 0)   {
      authorized = true;
      //debug
      //Serial.print("check: authorized="); Serial.println(authorized);
      readString = "";
      return;
    }

    if ( client.available() ) {
      char c = client.read();

      //read char by request
      if (readString.length() < MTKLEN) {

        //store characters to string
        readString += c;
        //Serial.print(c);
      }

      //if HTTP request has ended
      if (c == '\n') {
        client.flush();
        Serial.println(readString);

        //now output html data header

        String script = "";

        script += ("HTTP/1.1 401 Log-In Required");
        script += ("Content-Type: text/html \n");
        script += ("\n");  //  do not forget this one //????
        script += ("<!DOCTYPE html> \n");
        script += ("<html> \n");
        script += ("<head> \n");

        // utf-8 für "°" Zeichen
        script +=  "<meta http-equiv=\"Content-Type\" content=\"text/html; charset=utf-8\"> \n" ;
        script +=  "<title>" ;
        script +=  website_title ;
        script +=  "</title> \n" ;
        script +=  "</head> \n" ;
        script +=  "<body> \n" ;

        script += "<h1><p style=\"color:rgb(255,0,191);\"> " + (String)website_url ;
        script += (String)": &nbsp; <wbr> <wbr> " + "Not authorized !</p> </h1> \n" ;
        script += ("<h2><p style=\"color:rgb(255,0,191);\"> log in to proceed: </p> </h2> \n");

        script += ("<FORM ACTION='/' method=GET > \n");
        script += ("<h2>user name:  <INPUT TYPE=text NAME='uname' VALUE=''  MAXLENGTH='50'> </h2> \n");
        script += ("<h2>password :  <INPUT TYPE=PASSWORD NAME='upwd' VALUE='' MAXLENGTH='50'> </h2> \n");

        script += ("<h2><INPUT TYPE=SUBMIT></h2> \n");

        script += ("</FORM> \n");
        script += ("<BR> \n");
        script += ("</body> \n");
        script += ("</html> \n");

        client.print(script);

        //stopping client
        client.stop();
        delay(1);

        //clearing string for next read
        readString = "";
      }
    }
    delay(1);
  }
}

```

### Debug Messages

```
Debug messages go here
```

5chufti, 2018-08-09:
did you check 
`WiFiServer   wifiserver(http_port);`
has "http_port" well defined?

dsyleixa, 2018-08-09:
update:
core 2.4.1 works neither,
but nevertheless also 2.4.0 does work fine, like 2.3.0 before

devyte, 2018-08-09:
The sketch is incomplete:
-What is cstringarg?
-How are you doing the startup setup?
-how/when is this function handleNotAuthorized() called?

The requirement is for a Minimal Complete Verifiable Example [MCVE](https://stackoverflow.com/help/mcve) sketch. 

On a personal note: I don't understand why so many users insist on implementing their own webservers with WiFiServer. In this case, there is even an #include <ESP8266WebServer.h>, but then the webserver is not used.

dsyleixa, 2018-08-10:
ok, here we go....

'''
test
'''

dsyleixa, 2018-08-10:
I reported that issue just to let you know about a bug in those cores/libs.
it compiles with 2.4.0 and 2.3.0, just copy and paste.
so it must be a 2.4.1 and/or 2.4.2 bug.
Developers can strip that down far better than I am able to, because they have far more skills to focus on the really suspicious parts.

devyte, 2018-08-10:
@dsyleixa there are currently 3 active developers, and there are 455 open issues and 134 pending PRs at the time of this writing. Expecting a developer to handle your code is not realistic. 
I will remind you that the developers are volunteers who work on this core as a hobby during our free time, and that this repo is a community effort and not a service provided to users. You are therefore expected to do your homework with regard to issues that affect you, the best case of which is an issue you report that is accompanied by a PR that fixes it.

I am not convinced that the problem you see is really in the core, which is one of the reasons behind the requirement of an MCVE sketch. It is very common that a user thinks it's the core, then while reducing the sketch the issue can no longer be reproduced, which then leads to finding a bug in the user's own code or a 3rd party lib.

dsyleixa, 2018-08-11:
ok, I didn't suppose it to be so complicated for a developer to see through this issue already at the 1st glance. I'll see if I can strip it down and report then, until then I'll try to stay with the 2.4.0 core.

devyte, 2018-08-14:
Not complicated, just very very time consuming. A reduction and reproduction of a reported bug can easily take upwards of an hour. Multiply times a whole lot of issues, and there aren't enough hours in a day, let alone our free time each day, to look at all the reported issues. The issue count is now up from 455 in my previous comment to 463 now, and that's after I closed some.

I still see the original sketch as incomplete, and the second sketch as huge. As a guideline, the size of the code in your original post is manageable. It's just not complete, as in I can't copy-paste it into an IDE, flash it, and see the bug.

Scippi, 2018-08-17:
Hello,

i've had the same behaviour. 
It seem's that the client.stop() has been changed. 

With Server example from https://arduino-esp8266.readthedocs.io/en/2.4.2/esp8266wifi/server-examples.html it is reproducible:

```
#include <ESP8266WiFi.h>

const char* ssid = "********";
const char* password = "********";

WiFiServer server(80);

void setup()
{
  Serial.begin(115200);
  Serial.println();

  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" connected");

  server.begin();
  Serial.printf("Web server started, open %s in a web browser\n", WiFi.localIP().toString().c_str());
}

// prepare a web page to be send to a client (web browser)
String prepareHtmlPage()
{
  String htmlPage =
     String("HTTP/1.1 200 OK\r\n") +
            "Content-Type: text/html\r\n" +
            "Connection: close\r\n" +  // the connection will be closed after completion of the response
            "Refresh: 5\r\n" +  // refresh the page automatically every 5 sec
            "\r\n" +
            "<!DOCTYPE HTML>" +
            "<html>" +
            "Analog input:  " + String(analogRead(A0)) +
            "</html>" +
            "\r\n";
  return htmlPage;
}

void loop()
{
  WiFiClient client = server.available();
  // wait for a client (web browser) to connect
  if (client)
  {
    Serial.println("\n[Client connected]");
    while (client.connected())
    {
      // read line by line what the client (web browser) is requesting
      if (client.available())
      {
        String line = client.readStringUntil('\r');
        Serial.print(line);
        // wait for end of client's request, that is marked with an empty line
        if (line.length() == 1 && line[0] == '\n')
        {
          client.println(prepareHtmlPage());
          break;
        }
      }
    }
    delay(1); // give the web browser time to receive the data

    // close the connection:
    client.stop();
    Serial.println("[Client disonnected]");
  }
}
```
This example procudes: Failed to load resource: net::ERR_CONNECTION_RESET in Chrome
Within Edge/IExplorer : INET_E_DOWNLOAD_FAILURE

If i remove the last client.stop(), the example work as expected.

dsyleixa, 2018-08-17:
@ Scippi  : for core 2.4.2 I removed the last client.stop() in void handleNotAuthorized() but still the same error message

Fehler: Verbindung unterbrochen
Die Verbindung zum Server wurde zurückgesetzt, während die Seite geladen wurde.

still everything works like a charm with 2.4.0. though, just like before.

TheNitek, 2018-08-17:
I can confirm that client.stop(); seems to break things in 2.4.2.

dsyleixa, 2018-08-18:
again code stripped down a lot - now it's the absolute minimum code IMO.
(see above)

d-a-v, 2018-08-18:
Thanks for the efforts put in the MCVE !

dsyleixa, 2018-08-18:
a pleasure! 8-)

d-a-v, 2018-08-18:
Bug reproduced. Working with 2.4.0, not with 2.4.2.

simplified MCVE (dhcp mode, no i2c screen)

Thanks, now we can work :)

```
//----------------------------------------------------------------------------
//  board: ESP8266 NodeMCU 1.0 (ESP12-E module)
//  Arduino IDE 1.8.5
//  esp8266 core 2.4.0
//----------------------------------------------------------------------------

// Wifi + website data

const char* ssid = "SSID";             // WIFI network name
const char* password = "PSK";         // WIFI network password

// define
char  website_uname[20] = "xx" ; //  website user name log in  "MyWebsiteLoginName"
char  website_upwd[20] = "yy"; //  website user pwd log in   "MyWebsiteLoginPwd"
const char* website_title = "#5021";    //  website caption           "MySiteCaption"
const char* website_url = "http://givemeluck";      //  website url               "http:\\mysite.com"
int http_port = 80;

//----------------------------------------------------------------------------
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

WiFiServer   wifiserver(http_port);
ESP8266WebServer lanserver(8081);
bool      authorized = false;

//----------------------------------------------------------------------------
// handle root +clients
//----------------------------------------------------------------------------

void handleRoot() {
  handleClients();
}

//----------------------------------------------------------------------------

void handleClients() {
  String msgtok;
  //------------------------------------------

  //client-Werte auch bei Url-Aufruf zurückgeben

  String message = "*** ";
  // re CLIENT 0
  /*
    message += (String)"&c0t1=" + c0t1.sact + "&c0h1=" + c0h1.sact;
    message += (String)"&c0t2=" + c0t2.sact + "&c0h2=" + c0h2.sact;
    message += "&c0out1=" + (String)c0out1 + "&c0out2=" + (String)c0out2 + "&c0out3=" + (String)c0out3 ;
  */
  message += " ###";
  //Serial.println(message);
  lanserver.send(200, "text/plain", message);

}

//----------------------------------------------------------------------------
// Tools
//----------------------------------------------------------------------------

int16_t  strstrpos(char * haystack,  char * needle)   // find 1st occurance of substr in str
{
  char *p = strstr(haystack, needle);
  if (p) return p - haystack;
[... 305 lines cut ...]
  script += "<p style=\"color:rgb(0,0,0);\" > </p>  " ;
  script += ("<br> \n");
  script += "</h2>";

  client.print(script);
  script = "";

  script += ("<br> \n");

  // log out
  script += ("<h3>Log Out: ");
  script += ("<a href=\" /logout\"\"> <button style=\"height:70px;width:140px\" > Log Out </button></a> </h3> ");

  script += WiFi.localIP().toString() + " " + (String)ssid + " <br>" ;
  script += "</font> </p> \n";

  script += "</body> \n";
  script += "</html> \n";

  client.print(script);

  delay(1);

}
```

dsyleixa, 2018-08-18:
re: "Tnhkas, now we can wrok :)"
gaain, a leapsure,  gald I colud phel :P

dsyleixa, 2018-08-18:
nonetheless, there still is a minor issue,

logiing in by default pwd "yy", all is fine.
Unexpectedly, "yyyy" is accepted, too, which actually is not correct.
Any ideas?

dsyleixa, 2018-08-19:
new cstringard version, accepting also '?varname=' besides '&varname= as to 'needles' in the 'haystack':

´´´´
char * cstringarg( char* haystack, char* vname, char* sarg ) {
   int i=0, pos=-1;
   unsigned char  ch=0xff;
   char  kini[3] = "&";       // start of varname: '&': 
   char  kequ[3] = "=";       // end of varname, start of argument: '='
   char  needle[TOKLEN]="";   // complete pattern:  &varname=abc1234

   strcpy(sarg,"");
   strcpy(needle, kini);
   strcat(needle, vname);
   strcat(needle, kequ);
   pos = strstrpos(haystack, needle); 
   if(pos==-1) {
      needle[0]='?';
      pos = strstrpos(haystack, needle);
      if(pos==-1) return sarg;
   }
   pos=pos+strlen(vname)+2; // start of value = kini+vname+kequ   
   while( (ch!='&')&&(ch!='\0') ) {
      ch=haystack[pos+i];    
      if( (ch=='&')||(ch==';')||(ch==' ')||(ch=='\0') ||(ch=='\n')
        ||(i+pos>=strlen(haystack))||(i>TOKLEN-1) ) {
           sarg[i]='\0';
           return sarg;
      }       
      if( (ch!='&') ) {
          sarg[i]=ch;          
          i++;       
      }      
   } 
   return sarg;
}
´´´´

d-a-v, 2018-08-20:
Here's my thoughts:
- http protocol is not handled correctly. At the first received `\n`, all web data is sent then the connection is closed. The http client / browser realizes that its header is not fully read when the connection is closed (no ack received from the esp, because only the first line is read), hence the error. When modifying the sketch to read at least the full header (two consecutive `\n`), then things work.
- I don't know why this sketch was working with some previous version of the core. Unless forced to, I will not try to find out why. With proper proper protocol handling, things are working well.

To (quickl&dirty) fix the above sketch:
```
      static char oldc = 0;
      if (c == '\n' && oldc == '\n') {
        client.flush(); // <-- useless here, this empties the output buffer, not the input buffer
        ... // write data, flush() and stop()
      }
      oldc = c;
```

Using `Stream::readBytesUntil()` would allow to have a lighter code in the sketch.

d-a-v, 2018-08-20:
My quick&dirty fixes above are not sufficient. HTTP header ends with 13/10/13/10 (\r\n\r\n)
Also, filling `readString` is useless and must not be limited by `TOKLEN` which stops reading.
it's
```
      //if HTTP request has ended
      static char oldc = 0;
      //Serial.println("readstring=" + readString);
      if (oldc == 10 && c == 13) {
        client.read(); // consume last 10
        //client.flush(); // <-- useless here, this empties the output buffer, not the input buffer
        Serial.println("header received");
```

dsyleixa, 2018-08-21:
meanwhile changed handleNotAuthorized() checking also TOKLEN, additionally fixing an additional issue, still working fine with 2.4.0, but still not working with 2.4.2. (changes courtesy of "Moppy)":

```
void handleNotAuthorized() {
  String readString = "";
  char   strinput[MAXLEN], strupwd[TOKLEN], struname[TOKLEN] ;
  
  WiFiClient client = wifiserver.available();

  //---------------------------------------
  // debug
  // authorized=true;

  strcpy(strinput, "");
  strcpy(strupwd, "");
  strcpy(struname, "");

  while ( client.connected() ) {
    if (authorized) return;

    if ( client.available() ) {
      char c = client.read();

      //read char by request
      readString = "";
      while ( (readString.length() < TOKLEN) && (c != '\n') ) {
          readString += c;
          c = client.read();
      }

    readString.toCharArray(strinput, MAXLEN);
    // cstringarg( char* haystack, char* vname, char* sarg )
    // haystack pattern: &varname=1234abc,  delimiters &, \n, \0, SPACE, EOF
    cstringarg(strinput, "uname", struname);  // uname
    cstringarg(strinput, "upwd", strupwd);   // upwd

    // debug
    Serial.print("strupwd     >>>"); Serial.print(strupwd); Serial.println("<<<");
    Serial.print("website_upwd>>>"); Serial.print(website_upwd); Serial.println("<<<");
    Serial.print("readString>>>");Serial.println(readString);

    if ( (strlen(strupwd) == strlen(website_upwd)) && (strcmp(website_upwd, strupwd) == 0) )
    {
      authorized = true;
      //debug
      //Serial.print("check: authorized="); Serial.println(authorized);
      readString = "";
      return;
    }

      
      
      //if HTTP request has ended
      if (c == '\n') {
         client.flush();

        //now output html data header

        String script = "";

        script += ("HTTP/1.1 401 Log-In Required");
        script += ("Content-Type: text/html \n");
        script += ("\n");  //  do not forget this one //????
        script += ("<!DOCTYPE html> \n");
        script += ("<html> \n");
        script += ("<head> \n");

        // utf-8 für "°" Zeichen
        script +=  "<meta http-equiv=\"Content-Type\" content=\"text/html; charset=utf-8\"> \n" ;
        script +=  "<title>" ;
        script +=  website_title ;
        script +=  "</title> \n" ;
        script +=  "</head> \n" ;
        script +=  "<body> \n" ;

        script += "<h1><p style=\"color:rgb(255,0,191);\"> " + (String)website_url ;
        script += (String)": &nbsp; <wbr> <wbr> " + "Not authorized !</p> </h1> \n" ;
        script += ("<h2><p style=\"color:rgb(255,0,191);\"> log in to proceed: </p> </h2> \n");

        script += ("<FORM ACTION='/' method=GET > \n");
        script += ("<h2>user name:  <INPUT TYPE=text NAME='uname' VALUE=''  MAXLENGTH='50'> </h2> \n");
        script += ("<h2>password :  <INPUT TYPE=PASSWORD NAME='upwd' VALUE='' MAXLENGTH='50'> </h2> \n");

        script += ("<h2><INPUT TYPE=SUBMIT></h2> \n");

        script += ("</FORM> \n");
        script += ("<BR> \n");
        script += ("</body> \n");
        script += ("</html> \n");

        client.print(script);

        //stopping client
        client.stop();

        delay(1);

        //clearing string for next read
        //readString = "";
      }
    }
    delay(1);
  }
}
```

d-a-v, 2018-08-21:
You should restart from adapting the [WebServer example](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/HelloServer/HelloServer.ino).

FWIW, here's your modified working sketch. Apart from removing external stuff, only `handleNotAuthorized` is modified. I'm afraid I/we can't help you more here.
```
//----------------------------------------------------------------------------
//  board: ESP8266 NodeMCU 1.0 (ESP12-E module)
//  Arduino IDE 1.8.5
//  esp8266 core 2.4.0
//----------------------------------------------------------------------------

// Wifi + website data

const char* ssid = "SSID";             // WIFI network name
const char* password = "PSK";         // WIFI network password

// define
char  website_uname[20] = "xx" ; //  website user name log in  "MyWebsiteLoginName"
char  website_upwd[20] = "yy"; //  website user pwd log in   "MyWebsiteLoginPwd"
const char* website_title = "#5021";    //  website caption           "MySiteCaption"
const char* website_url = "http://givemeluck";      //  website url               "http:\\mysite.com"
int http_port = 80;

//----------------------------------------------------------------------------
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

WiFiServer   wifiserver(http_port);
ESP8266WebServer lanserver(8081);
bool      authorized = false;

//----------------------------------------------------------------------------
// handle root +clients
//----------------------------------------------------------------------------

void handleRoot() {
  handleClients();
}

//----------------------------------------------------------------------------

void handleClients() {
  String msgtok;
  //------------------------------------------

  //client-Werte auch bei Url-Aufruf zurückgeben

  String message = "*** ";
  // re CLIENT 0
  /*
    message += (String)"&c0t1=" + c0t1.sact + "&c0h1=" + c0h1.sact;
    message += (String)"&c0t2=" + c0t2.sact + "&c0h2=" + c0h2.sact;
    message += "&c0out1=" + (String)c0out1 + "&c0out2=" + (String)c0out2 + "&c0out3=" + (String)c0out3 ;
  */
  message += " ###";
  //Serial.println(message);
  lanserver.send(200, "text/plain", message);

}

//----------------------------------------------------------------------------
// Tools
//----------------------------------------------------------------------------

int16_t  strstrpos(char * haystack,  char * needle)   // find 1st occurance of substr in str
{
  char *p = strstr(haystack, needle);
  if (p) return p - haystack;
  return -1;   // Not found = -1.
}

[... 311 lines cut ...]
  script += "<p style=\"color:rgb(0,0,0);\" > </p>  " ;
  script += ("<br> \n");
  script += "</h2>";

  client.print(script);
  script = "";

  script += ("<br> \n");

  // log out
  script += ("<h3>Log Out: ");
  script += ("<a href=\" /logout\"\"> <button style=\"height:70px;width:140px\" > Log Out </button></a> </h3> ");

  script += WiFi.localIP().toString() + " " + (String)ssid + " <br>" ;
  script += "</font> </p> \n";

  script += "</body> \n";
  script += "</html> \n";

  client.print(script);

  delay(1);

}
```

dsyleixa, 2018-08-21:
thanks for your reply!
Unfortunately, the last version of yours doesn't work correctly after once having logged out on the main website (LogIn-Website does not reload always reliably), and besides "yy" also "yynnnnn" (n=arbitrary) is accepted too as a valid pwd (opposite to the last source code version I had posted above).

I have to admit that I don't have the skills to understand how to change the code and fix all those issues in a reasonable way, so I'm afraid I'll have to stay with my own last version (feat. Moppi's fix)  and the 2.4.0 core, at least it's working there as expected.
Nonetheless, thank you very much for your help and for your efforts!

d-a-v, 2018-08-21:
This why arduino libraries are here, they do part of the job for you.
Please post a new request issue asking for a webserver example with credential,
maybe someone (not excluding me) will take time to do it well for the core's examples.

dsyleixa, 2018-08-21:
thanks for your advice, I did so!

wongnam, 2018-08-24:
Web site doesn't show up if i use VPN with core 2.4.0 till 2.4.2. 
Core 2.3.0 is fine in VPN.

d-a-v, 2018-08-24:
@wongnam please open a new issue with your details.

dsyleixa, 2018-11-15:
has meanwhile anyone got an idea why my example which works for 2.4.0 does NOT work with 2.4.2 core version?

dsyleixa, 2018-12-19:
2.4.2 still does not work as shown in  https://github.com/esp8266/Arduino/issues/5021#issuecomment-411997320 .
2.4.0 instead works.
Which is the reason why? What is faulty?

5chufti, 2018-12-20:
@wongnam 
this might be caused by lwip v2 lower memory, try with v2 higher bandwidth

dsyleixa, 2018-12-20:
> 
> 
> @wongnam
> this might be caused by lwip v2 lower memory, try with v2 higher bandwidth

what do you mean? Is it a cross-post? Here is no user wongnam contributing!
please describe more detailed please!

earlephilhower, 2019-01-31:
This bug seems to have meandered around the last 6 months and ended up morphing into a much simpler one asking for a specific example, which is still open.  Don't see anything to do here, closing.

dsyleixa, 2019-01-31:
> 
> 
> This bug seems to have meandered around the last 6 months and ended up morphing into a much simpler one asking for a specific example, which is still open. Don't see anything to do here, closing.

THIS topic is not resolved because the html code leads to immrdiate website abort with 2.4.2
OTOH, it still works with 2.4.0.
So how has THIS html code to be changed to make it work with 2.4.2 too?

the other topic you mentioned  ("much simpler" ?) is not resolved yet as it does even provide any  a working html code example!

So please kindly reopen this topic as it is still unresolved!

dsyleixa, 2019-01-31:
the code is cut-down here:

https://github.com/esp8266/Arduino/issues/5021#issuecomment-414085159

bug reproduced by d-a-v

earlephilhower, 2019-01-31:
Cool.  I'm sure it would never be found otherwise.  I'll reopen this one, but can you please edit the initial post here to point to that comment and the title to be more direct?

d-a-v, 2019-01-31:
@dsyleixa There's also this https://github.com/esp8266/Arduino/issues/5021#issuecomment-414633054 just after the one you reference.
From what I understand you want to stick with a malfunctioning sketch because you have issues with adapting a working sketch to your needs.

From advices given to you, you opened a new issue, in which you received help from others, and you clearly state you don't have enough skills to achieve your goals, in https://github.com/esp8266/Arduino/issues/5067#issuecomment-449398243.

We, maintainers, spend a lot of our spare time to fix bugs and improve performances. I'm sorry to tell you we just cannot write sketches for you. There are support forum for this, and 99% of the time you can find a suitable example somebody kindly shared, by just googling. We will be delighted when you come back with a meaning and working example that you will propose in a pull-request to add in our example repository.

dsyleixa, 2019-02-01:
my request of the "other" topic was about a working code example in html for the nodeMCU core examples, to be used out of the Arduino IDE. Currently the provided examples don't help very much for common Arduino users who are no computer professionals but pupils, artists, craftsmen, and hobbyists.