esp8266/Arduino #5023: HTTP_UPDATE_FAILD Error (6): Update error: ERROR[6]: Stream Read Timeout

josep112, 2018-08-09:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			2.4.2
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Problem Description
I had hope that with core 2.4.2 my problem would be solved but I have the error in the update fimrware HTTP_UPDATE_FAILD Error (6), I already did the test on the same network and the problem only happens with servers outside, I already tested in many are all from my country, the ping is low but I can not solve any idea what I can do? I've already tried
ESPhttpUpdate.setHttpTimeout (16000); but without success, I'm using the example code

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

void setup() {

  USE_SERIAL.begin(115200);
  // USE_SERIAL.setDebugOutput(true);

  USE_SERIAL.println();
  USE_SERIAL.println();
  USE_SERIAL.println();

  for (uint8_t t = 4; t > 0; t--) {
    USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
    USE_SERIAL.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("SSID", "PASSWORD");

}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    t_httpUpdate_return ret = ESPhttpUpdate.update("http://server/file.bin");

    switch (ret) {
      case HTTP_UPDATE_FAILED:
        USE_SERIAL.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        USE_SERIAL.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        USE_SERIAL.println("HTTP_UPDATE_OK");
        break;
    }
  }
}

d-a-v, 2018-08-23:
The updater is not fully network aware and does not cope well with network lags. Try with `Higher bandwitdh` options.

[This](https://github.com/esp8266/Arduino/blob/18f643c7e2d6a0da9d26ff2b14c94e6536ab78c1/cores/esp8266/Updater.cpp#L370) is the error `ERROR[6]: Stream Read Timeout`

The updater should check whether the stream (being a WiFiClient here) is still connected.
But the Stream API will not let do that as it is.

There is an old ongoing updater PR #4054, I will try to finish that work.

CRCinAU, 2018-08-23:
In my case, the server is on the same network - so there is no latency between the ESP and the apache web server.

In fact:
```
$ ping 10.1.1.6
PING 10.1.1.6 (10.1.1.6) 56(84) bytes of data.
64 bytes from 10.1.1.6: icmp_seq=1 ttl=128 time=1.60 ms
64 bytes from 10.1.1.6: icmp_seq=2 ttl=128 time=1.47 ms
64 bytes from 10.1.1.6: icmp_seq=3 ttl=128 time=4.19 ms
^C
--- 10.1.1.6 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 1.478/2.426/4.197/1.254 ms
```

The first one I tried (pasted the log from) was using high bandwidth option. Switching to low_memory made no difference.

I even tried with both `-D PIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH` and `-D PIO_FRAMEWORK_ARDUINO_LWIP_LOW_MEMORY` with no success.

CRCinAU, 2018-08-23:
Ok, so I had an interesting discovery just now....

When I switched out my managed Cisco 3802i access point and enabled the built in wifi on my VDSL router, I was able to do a httpUpdate with no issues at all.

I tested this with *TWO* different devices (an ESP01 and a D1 mini). Both updated without an issue.

So, the question is, what is the difference in this scenario between the two different bits of wireless infrastructure?

The SSID / security mode is the same on both bits of equipment.

d-a-v, 2018-08-23:
A question of delay / latency ?
A tcpdump would tell ?

Try to raise your 256, that would have a better effect with higher bandwidth (1460 instead of lower memory(536))

CRCinAU, 2018-08-23:
I might have to try over the weekend to see if I can get a working and non-working tcpdump of the two different access points.

It didn't seem to matter what was on the ESP01 (it was probably default) - but changing the AP over to the 'dumb' one in the VDSL modem worked first time - whereas it failed for weeks using the Cisco 3802i.

CRCinAU, 2018-08-24:
tcpdump on web server filtered for the ESP01's IP address with the Cisco AP connected and online:

```
17:11:50.707784 ARP, Request who-has 10.1.1.5 tell 10.1.1.5, length 46
17:11:50.737827 ARP, Request who-has 10.1.1.254 tell 10.1.1.5, length 46
17:11:50.904116 ARP, Request who-has 10.1.1.93 tell 10.1.1.5, length 46
17:11:50.904136 ARP, Reply 10.1.1.93 is-at 00:16:3e:30:aa:a5, length 28
17:11:50.905631 IP 10.1.1.5.49153 > 10.1.1.93.http: Flags [S], seq 6509, win 2144, options [mss 536], length 0
17:11:50.905659 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [S.], seq 458235548, ack 6510, win 29200, options [mss 1460], length 0
17:11:50.907638 IP 10.1.1.5.49153 > 10.1.1.93.http: Flags [.], ack 1, win 2144, length 0
17:11:50.910507 IP 10.1.1.5.49153 > 10.1.1.93.http: Flags [P.], seq 1:388, ack 1, win 2144, length 387: HTTP: GET /arduino/update/ HTTP/1.0
17:11:50.910534 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], ack 388, win 30016, length 0
17:11:50.972808 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [P.], seq 1:283, ack 388, win 30016, length 282: HTTP: HTTP/1.0 200 OK
17:11:50.973020 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 283:819, ack 388, win 30016, length 536: HTTP
17:11:50.973034 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 819:1355, ack 388, win 30016, length 536: HTTP
17:11:50.973068 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 1355:1891, ack 388, win 30016, length 536: HTTP
17:11:50.973078 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 1891:2145, ack 388, win 30016, length 254: HTTP
17:11:50.994433 IP 10.1.1.5.49153 > 10.1.1.93.http: Flags [.], ack 283, win 1862, length 0
17:11:51.206788 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 283:819, ack 388, win 30016, length 536: HTTP
17:11:51.213254 IP 10.1.1.5.49153 > 10.1.1.93.http: Flags [.], ack 819, win 2144, length 0
17:11:51.213282 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 2145:2681, ack 388, win 30016, length 536: HTTP
17:11:51.213288 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 2681:2963, ack 388, win 30016, length 282: HTTP
17:11:51.631190 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 819:1355, ack 388, win 30016, length 536: HTTP
17:11:51.637220 IP 10.1.1.5.49153 > 10.1.1.93.http: Flags [.], ack 1355, win 2144, length 0
17:11:51.637272 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 1355:1891, ack 388, win 30016, length 536: HTTP
17:11:51.637279 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 1891:2145, ack 388, win 30016, length 254: HTTP
17:11:51.637284 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 2145:2681, ack 388, win 30016, length 536: HTTP
17:11:51.637285 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 2681:2963, ack 388, win 30016, length 282: HTTP
17:11:51.639927 IP 10.1.1.5.49153 > 10.1.1.93.http: Flags [.], ack 1355, win 2144, length 0
17:11:52.511094 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 1355:1891, ack 388, win 30016, length 536: HTTP
17:11:52.562025 IP 10.1.1.5.49153 > 10.1.1.93.http: Flags [.], ack 1891, win 2144, length 0
17:11:52.562074 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 1891:2145, ack 388, win 30016, length 254: HTTP
17:11:52.562090 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 2145:2681, ack 388, win 30016, length 536: HTTP
17:11:52.562091 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 2681:2963, ack 388, win 30016, length 282: HTTP
17:11:52.562163 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [P.], seq 2963:3217, ack 388, win 30016, length 254: HTTP
17:11:52.564864 IP 10.1.1.5.49153 > 10.1.1.93.http: Flags [.], ack 1891, win 2144, length 0
17:11:52.564915 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 3217:3753, ack 388, win 30016, length 536: HTTP
17:11:52.564923 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 3753:4035, ack 388, win 30016, length 282: HTTP
17:11:54.238768 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 1891:2145, ack 388, win 30016, length 254: HTTP
17:11:54.259023 IP 10.1.1.5.49153 > 10.1.1.93.http: Flags [.], ack 2145, win 1890, length 0
17:11:54.259067 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 2145:2681, ack 388, win 30016, length 536: HTTP
17:11:54.259075 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 2681:2963, ack 388, win 30016, length 282: HTTP
17:11:54.259080 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [P.], seq 2963:3217, ack 388, win 30016, length 254: HTTP
17:11:54.259084 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 3217:3753, ack 388, win 30016, length 536: HTTP
17:11:54.263195 IP 10.1.1.5.49153 > 10.1.1.93.http: Flags [.], ack 2145, win 1890, length 0
17:11:57.630760 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 2145:2681, ack 388, win 30016, length 536: HTTP
17:11:57.635812 IP 10.1.1.5.49153 > 10.1.1.93.http: Flags [.], ack 2681, win 2144, length 0
17:11:57.635855 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [P.], seq 2681:3217, ack 388, win 30016, length 536: HTTP
17:11:57.635865 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 3217:3753, ack 388, win 30016, length 536: HTTP
17:11:57.635868 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 3753:4035, ack 388, win 30016, length 282: HTTP
17:11:57.635923 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [P.], seq 4035:4289, ack 388, win 30016, length 254: HTTP
17:11:57.638644 IP 10.1.1.5.49153 > 10.1.1.93.http: Flags [.], ack 2681, win 2144, length 0
17:11:57.638665 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 4289:4825, ack 388, win 30016, length 536: HTTP
[... cut ...]
17:12:44.543096 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [P.], seq 3753:4289, ack 389, win 30016, length 536: HTTP
17:12:44.544673 IP 10.1.1.5.49153 > 10.1.1.93.http: Flags [R.], seq 389, ack 4289, win 2144, length 0
17:12:47.102767 ARP, Request who-has 10.1.1.5 tell 10.1.1.93, length 28
17:12:47.103102 ARP, Reply 10.1.1.5 is-at 84:f3:eb:0e:ff:90, length 46
17:12:50.228409 IP 10.1.1.5 > 224.0.0.251: igmp v2 report 224.0.0.251
17:12:50.307749 IP 10.1.1.5 > 224.0.0.251: igmp v2 report 224.0.0.251
```

I believe the igmp packets are the mdns advertisements starting after the update has failed.

d-a-v, 2018-08-24:
```
17:12:17.923624 IP 10.1.1.5.49153 > 10.1.1.93.http: Flags [.], ack 3753, win 2144, length 0
----- ^^ the esp is missing 3753+
17:12:17.923646 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [P.], seq 3753:4289, ack 388, win 30016, length 536: HTTP
----- ^^ it is resent
17:12:17.923657 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 4289:4825, ack 388, win 30016, length 536: HTTP
17:12:17.923660 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 4825:5361, ack 388, win 30016, length 536: HTTP
17:12:17.923702 IP 10.1.1.93.http > 10.1.1.5.49153: Flags [.], seq 5361:5897, ack 388, win 30016, length 536: HTTP
----- vv 24 seconds later, esp closes the connection (library timeout?), stating 3723+ is still not received.
17:12:42.026187 IP 10.1.1.5.49153 > 10.1.1.93.http: Flags [F.], seq 388, ack 3753, win 2144, length 0
```

It looks like as if packets are no more received on the esp side. OTA is not working, is the ESP further responsive without rebooting ?

> So, the question is, what is the difference in this scenario between the two different bits of wireless infrastructure?
I won't tell you to read through #2330

CRCinAU, 2018-08-24:
The http server on the ESP responds fine. I can even visit /update on the ESP and upload a new firmware without issue.

dmagyar, 2018-08-26:
I have the same issue, HTTP_UPDATE_FAILED Error (-11): HTTP error: read Timeout or Stream timeout.

On the server side I see that only partial content is received:

```
2018-08-26T20:28:57+02:00 haproxy haproxy[58]: x:16854 [26/Aug/2018:20:28:51.932] http-in~ host_be_1004-iot/srv_1004-iot_0 200 64512 - - cD-- "GET /esp8266/ HTTP/1.0"
2018-08-26T20:31:10+02:00 haproxy haproxy[58]: x:6228 [26/Aug/2018:20:31:04.915] http-in~ host_be_1004-iot/srv_1004-iot_0 200 64512 - - cD-- "GET /esp8266/ HTTP/1.0"
2018-08-26T20:31:29+02:00 haproxy haproxy[58]: x:1327 [26/Aug/2018:20:31:24.229] http-in~ host_be_1004-iot/srv_1004-iot_0 200 64512 - - cD-- "GET /esp8266/ HTTP/1.0"
2018-08-26T20:31:55+02:00 haproxy haproxy[58]: x:31836 [26/Aug/2018:20:31:49.980] http-in~ host_be_1004-iot/srv_1004-iot_0 200 64512 - - cD-- "GET /esp8266/ HTTP/1.0"
```

Every time exactly 64512 bytes, although my firmware is much bigger (~3megs).
There is nothing odd in tcpdump and the network connection / wifi signal is adequate...

Thanks,
Denes

dmagyar, 2018-08-27:
After enabling full debugging I ran into the "SSL cert too big" issue - so I decided to go for self-signed and removed the chain. I no longer get the memory error or any leads in the debug. It simply complains about Stream Read Timeout. The same update against the same server works perfectly over HTTP - so no network/latency/etc issue for sure.

```
Setting time using SNTP
Current time: Thu Jan  1 08:00:03 1970

Checking for updates... [hostByName] request IP for: iot.ur.gd
[hostByName] Host: iot.ur.gd IP: 94.199.52.133
:ref 1
:ref 2
:wr 90 90 0
:wrc 90 90 0
:ack 90
:rn 968
:rd 5, 968, 0
:rdi 968, 5
:rd 80, 968, 5
:rdi 963, 80
:rd 5, 968, 85
:rdi 883, 5
:rd 869, 968, 90
:rdi 878, 869
:rd 5, 968, 959
:rdi 9, 5
:rd 4, 968, 964
:rdi 4, 4
:c0 4, 968
:wr 267 267 0
:wrc 256 267 0
:wrc 11 11 0
:wr 6 6 0
:wrc 6 6 0
:wr 85 85 0
:wrc 85 85 0
:ack 267
:ack 91
:rn 91
:rd 5, 91, 0
:rdi 91, 5
:rd 1, 91, 5
:rdi 86, 1
:rd 5, 91, 6
:rdi 85, 5
:rd 80, 91, 11
:rdi 80, 80
:c0 80, 91
:ref 3
domain name: 'iot.ur.gd'
:ur 3
:wr 469 469 0
:wrc 256 469 0
:wrc 213 213 0
:ack 469
:rn 1460
:rch 1460, 1460
:rch 2920, 1460
:rch 4380, 401
:rd 5, 4781, 0
:rdi 1460, 5
ssl->need_bytes=8400 > 6859
:rd 4776, 4781, 5
:rdi 1455, 1455
:c 1455, 1460, 4781
:rdi 1460, 1460
:c 1460, 1460, 3321
:rdi 1460, 1460
:c 1460, 1460, 1861
:rdi 401, 401
:c0 401, 401
:rn 1460
[... 260 lines cut ...]
:rn 1460
:rd 1460, 1460, 0
:rdi 1460, 1460
:c0 1460, 1460
:rn 1460
:rch 1460, 1460
:rd 25, 2920, 0
:rdi 1460, 25
:wr 69 69 0
:wrc 69 69 0
:wr 69 69 0
:wrc 69 69 0
:rch 2920, 1460
:rch 4380, 1460
:ack 138
:er -14 0x00000000
pm open,type:0 0
ERROR[6]: Stream Read Timeout
[httpUpdate] Update.writeStream failed! (ERROR[6]: Stream Read Timeout)
[httpUpdate] Update failed
:ur 2
:ur 1
:del
HTTP_UPDATE_FAILED Error (6): Update error: ERROR[6]: Stream Read Timeout
```

laercionit, 2018-09-03:
Unfortunately after migrating to 2.4.2 I'm having the same problem, I've tested it on several wifi networks, most of the time...

laercionit, 2018-09-04:
My Debug:

SerialCommand: SystemFirmwareUpdate(http://updates.aquarino.com.br/esp8266/Aquarino-Controller.ino.nodemcu.bin);
sleep disable
[begin] roundedSize:       0x00090000 (589824)
[begin] updateEndAddress:  0x00200000 (2097152)
[begin] currentSketchSize: 0x00091000 (593920)
[begin] _startAddress:     0x00170000 (1507328)
[begin] _currentAddress:   0x00170000 (1507328)
[begin] _size:             0x0008FB90 (588688)
Header: 0xE9 1 2 40
ERROR[6]: Stream Read Timeout
SystemFirmwareUpdate: Update failed. (6): Update error: ERROR[6]: Stream Read Timeout

laercionit, 2018-09-06:
Hello people good night.

I ran several tests and got some information that might be useful.

I am having trouble downloading the file only using NODEMCU V3.
The same code running on NodeMCU V2 works without problems.

Any logic to this?

`
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>

#include <SPI.h>
#include <SD.h>

#define PinSDSS       15
#define DebugSerial Serial

ESP8266WiFiMulti WiFiMulti;

String ConvertFormatBytes(size_t bytes) {
  if (bytes < 1024) {
    return String(bytes) + F("B");
  } else if (bytes < (1024 * 1024)) {
    return String(bytes / 1024.0) + F("KB");
  } else if (bytes < (1024 * 1024 * 1024)) {
    return String(bytes / 1024.0 / 1024.0) + F("MB");
  } else {
    return String(bytes / 1024.0 / 1024.0 / 1024.0) + F("GB");
  }
}

boolean OtaDownloadFile(String URL, String PathToSave) {
  if ((WiFiMulti.run() == WL_CONNECTED)) {
    HTTPClient http;
    http.begin(URL);
    int httpCode = http.GET();
    if ((httpCode > 0)) {
      if (httpCode == HTTP_CODE_OK) {
        int DownloadSize  = http.getSize();
        int RemainingSize = DownloadSize;
        DebugSerial.println(String(F("OTADownloadFile: Get >> ")) + URL);
        DebugSerial.println(String(F("OTADownloadFile: File Size >> ")) + ConvertFormatBytes(DownloadSize));
        DebugSerial.println(String(F("OTADownloadFile: Save File as  >> ")) + PathToSave);
        uint8_t buff[64] = { 0 };
        WiFiClient * stream = http.getStreamPtr();
        unsigned long DebugTimeCycle = millis();

        if (SD.exists(PathToSave)) SD.remove(PathToSave);
        File DownloadFile = SD.open(PathToSave, FILE_WRITE);
        if (!DownloadFile) DownloadFile = SD.open(PathToSave, FILE_WRITE);
        if (DownloadFile) {

          while (http.connected() && (RemainingSize > 0 || RemainingSize == -1)) {
            auto size = stream->available();
            if (size) {

              auto StreamBytes = stream->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));
              DownloadFile.write(buff, StreamBytes);
              DownloadFile.flush();

              if (RemainingSize > 0) {
                RemainingSize -= StreamBytes;
                if ((millis() - DebugTimeCycle >= 1000)) {
                  DebugTimeCycle = millis();
                  float PercentDownloaded = (((DownloadSize - RemainingSize) * 100) / DownloadSize);
                  DebugSerial.print   (String(F("OTADownloadFile: Progress ")) + String(PercentDownloaded) + String(F("%")));
                  DebugSerial.print   (String(F("\tDownloaded ")) + ConvertFormatBytes(DownloadSize - RemainingSize));
                  DebugSerial.println (String(F("\tRemaining ")) + ConvertFormatBytes(RemainingSize));
                }
              }
            }
            yield();
          }
          //-----Verify Download  File ------
          if (DownloadFile.size() == DownloadSize) {
            DebugSerial.println(String(F("OTADownloadFile: Download file DONE")));
            DownloadFile.close(); http.end();
            return (true);
          } else {
            DebugSerial.println(String(F("OTADownloadFile: Download file FAIL")));
            DownloadFile.close(); http.end();
          }
        } else DebugSerial.println(String(F("OTADownloadFile: Open File Failed")));
      } else DebugSerial.printf("OTADownloadFile: Server response, error: % s\n", http.errorToString(httpCode).c_str());
    } else DebugSerial.printf("OTADownloadFile: Server response, error: % s\n", http.errorToString(httpCode).c_str());
    http.end();
    return (false);
  }
}

void setup() {

  DebugSerial.begin(115200);
  DebugSerial.setDebugOutput(true);
  for (uint8_t t = 4; t > 0; t--) {
    DebugSerial.printf("[SETUP] WAIT %d...\n", t);
    DebugSerial.flush();
    delay(1000);
  }
  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("NETWORK", "1a2b3c4d");
  SD.begin(PinSDSS);
}

void loop() {
  if ((WiFiMulti.run() == WL_CONNECTED)) {
    OtaDownloadFile("http://XXXXXXXXXX/Arduino.zip", "/Update/Arduino.zip");
    delay(10000);
  }
}

`

alexwbaule, 2018-09-07:
I Have the same problem, the update dont get completed.

My Debug has (Was tested with the WebUpdate sketch and using Chrome to upload the image)

New client
:ref 2
:ur 2
method: POST url: /update search: 
headerName: Host
headerValue: 192.168.250.191
headerName: Connection
headerValue: keep-alive
headerName: Content-Length
headerValue: 365449
headerName: Cache-Control
headerValue: max-age=0
headerName: Origin
headerValue: http://192.168.250.191
headerName: Upgrade-Insecure-Requests
headerValue: 1
headerName: DNT
headerValue: 1
headerName: Content-Type
headerValue: multipart/form-data; boundary=----WebKitFormBoundaryCLOmedSu7ATGwl5s
headerName: User-Agent
headerValue: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/69.0.3497.81 Safari/537.36
headerName: Accept
headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8
headerName: Referer
headerValue: http://192.168.250.191/
:c0 1, 536
:rn 70
:rch 70, 536
:rch 606, 536
:rch 1142, 536
:rch 1678, 466
headerName: Accept-Encoding
headerValue: gzip, deflate
headerName: Accept-Language
headerValue: en-US,en;q=0.9,pt;q=0.8,mt;q=0.7
:c 1, 70, 2144
if isForm
args: 
Parse Form: Boundary: ----WebKitFormBoundaryCLOmedSu7ATGwl5s Length: 365449
PostArg FileName: update.bin
PostArg Name: update
PostArg Type: application/x-sega-cd-rom
Start File: update.bin Type: application/x-sega-cd-rom
Update: update.bin
sleep disable
[begin] roundedSize:       0x000B7000 (749568)
[begin] updateEndAddress:  0x00100000 (1048576)
[begin] currentSketchSize: 0x00048000 (294912)
[begin] _startAddress:     0x00049000 (299008)
[begin] _currentAddress:   0x00049000 (299008)
[begin] _size:             0x000B7000 (749568)
wifi evt: 7
:c 1, 536, 2074
:c 1, 536, 1538
:c 1, 536, 1002
:c0 1, 466
:rn 70
:rch 70, 536
:rch 606, 536
:rch 1142, 536
:c 1, 70, 1678
:c 1, 536, 1608
:c 1, 536, 1072
:c0 1, 536
:rn 536
:rch 536, 536
wifi evt: 7
:c 1, 536, 1072
:c0 1, 536
:rn 536
:c0 1, 536
:rn 536
:c0 1, 536
:rn 536

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0xad
csum 0xad
vcc9e799f
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

argolomb, 2018-09-07:
Hi, I'm dealing with this problem since yesterday. In my case my web page is host in a nginx server and respponse with a http 1.1

This is what the ESP send to server (Note http 1.0)
```
Update sketch...
[HTTP-Client][begin] host: rcstuffs.com port: 80 uri: /OTA.php
[HTTP-Client] connected to rcstuffs.com:80
[HTTP-Client] sending request header
-----
GET /OTA.php HTTP/1.0
Host: rcstuffs.com
User-Agent: ESP8266-http-Update
Connection: close
x-ESP8266-STA-MAC: 2C:3A:E8:34:07:88
x-ESP8266-AP-MAC: 2E:3A:E8:34:07:88
x-ESP8266-free-space: 2859008
x-ESP8266-sketch-size: 285536
x-ESP8266-sketch-md5: 90113db11796d018dd9d9af48d5d9d4e
x-ESP8266-chip-size: 4194304
x-ESP8266-sdk-version: 2.2.1(cfd48f3)
x-ESP8266-mode: sketch
```

Server response in http 1.1

```
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Fri, 07 Sep 2018 15:15:51 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Server: Apache'
[HTTP-Client][handleHeaderResponse] RX: 'X-Powered-By: PHP/7.2.6'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Disposition: attachment; filename=BlinkWithoutDelay.bin'
[HTTP-Client][handleHeaderResponse] RX: 'x-MD5: 76e9f893364d2497de69a13775fb7333'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 252960'
[HTTP-Client][handleHeaderResponse] RX: 'Vary: Accept-Encoding,User-Agent'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: application/octet-stream'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 252960
```

comment in https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.cpp

// use HTTP/1.0 for update since the update handler not support any transfer Encoding

maybe it's a problem with encoding

alexwbaule, 2018-09-11:
Maybe the error is in _parseForm function. I will do some tests.

I Think its some error from there, because the OTA stops (the write is not called, i put some debugs in Updater.cpp) and the Parsing set the UPLOAD_FILE_WRITE.

d-a-v, 2018-10-02:
Feedback for #5126 is welcome with that particular issue.
(@CRCinAU using your AP with which OTA does not work)

ascillato, 2018-10-11:
@d-a-v 

Hi,

We are also having this issue with OTA.

If using core v2.4.2 we can do an OTA from an URL in a server on Internet without problems.

Now with last stage core (11-oct-2018) we can do an OTA from a local file but we can not do an OTA from an URL in a server on Internet.

I tested using lwIP2 (high bandwidth) and also lwIP2 (high bandwidth - no features).
Also we update the syntax of command ESPhttpUpdate.update because the one that we were using before has the deprecated warning in the STAGE core.

ascillato, 2018-10-12:
@d-a-v 

I updated to the last commit https://github.com/esp8266/Arduino/commit/c8497da6766543c0e0c1dc130ab54c3d722ea017 and stills fails `HTTP_UPDATE` but only when using an URL.

The Hardware is a NodeMCU 4MB ( Flash size selected is 4M (1M) )
Compiled with Arduino IDE
The software is Sonoff-Tasmota (https://github.com/arendst/Sonoff-Tasmota)
The test file is at http://thehackbox.org/tasmota/sonoff-4m.bin
The map file of the test file is at http://thehackbox.org/tasmota/sonoff-4m.map

(If downloading the file manually and then uploading via web menu, OTA works fine)

_Used `DEBUG` with `HTTP_UPDATE `option_

**Serial Debug:**

```
SDK:3.0.0-dev(c0f7b44)/Core:win-2.5.0-dev/lwIP:2.1.0(STABLE-2_1_0_RELEASE/glue:arduino-2.4.2-13-g80224f0)/BearSSL:f55a6ad

bcn 0
del if1
usl
mode : sta(60:01:94:59:17:19)
add if0
sleep enable,type: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)

add 0
aid 15
cnt

connected with NetWireless, channel 1
ip:192.168.1.33,mask:255.255.255.0,gw:192.168.1.1
ip:192.168.1.33,mask:255.255.255.0,gw:192.168.1.1
pm open,type:2 0
[httpUpdate] HTTP error: read Timeout
[httpUpdate] HTTP error: read Timeout
[httpUpdate] HTTP error: read Timeout
[httpUpdate] HTTP error: read Timeout
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
del if0
usl

Panic C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_main.cpp:103 __yield

ctx: sys
sp: 3fffeca0 end: 3fffffb0 offset: 01b0

>>>stack>>>
3fffee50:  3ffee570 00000006 00000000 3fff7954
3fffee60:  00000000 4bc6a7f0 51a9fbe7 00000000
3fffee70:  00000000 00000000 4bc6a7f0 00000000
3fffee80:  3ffea6c4 4023b8ec 40100874 000150b1
3fffee90:  00000000 00000000 000150e2 40242a0c
3fffeea0:  01854d06 4bc6a7f0 1b645a1c 00000000
3fffeeb0:  00000000 00000000 4bc6a7f0 4022c8f5
3fffeec0:  000150b1 ffffffff 3fff7914 40224a47
3fffeed0:  0000012c 3ffea8f4 3fff484c 4023b88d
3fffeee0:  000005e0 00000000 40103ea1 3ffeffb0
3fffeef0:  3fffdc10 3fff7954 3fff2618 40224a89
3fffef00:  0000001e 00000001 3fff48fc 402246e5
3fffef10:  3ffee3b8 3fff7954 3fff2618 402250a0
3fffef20:  3ffee3b8 401015fb 3fff7954 40223b20
3fffef30:  40101636 00000000 00000002 3ffeffb0
[... 218 lines cut ...]
csum 0x2d
v00000000
~ld

SDK:3.0.0-dev(c0f7b44)/Core:win-2.5.0-dev/lwIP:2.1.0(STABLE-2_1_0_RELEASE/glue:arduino-2.4.2-13-g80224f0)/BearSSL:f55a6ad

bcn 0
del if1
usl
mode : sta(60:01:94:59:17:19)
add if0
sleep enable,type: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 15
cnt

connected with NetWireless, channel 1
ip:192.168.1.33,mask:255.255.255.0,gw:192.168.1.1
ip:192.168.1.33,mask:255.255.255.0,gw:192.168.1.1
pm open,type:2 0
```

d-a-v, 2018-10-12:
I can't reproduce yet,
I tried the httpUpdate sketch from core example using git master (and really remote http sketch server) with no issue.
I tried tasmota master but I can't get out from AP mode, core debug debug messages show truncated ssid and password (last char missing for both).

ascillato, 2018-10-12:
> I tried tasmota master but I can't get out from AP mode, core debug debug messages show truncated ssid and password (last char missing for both).

some characters are not supported like *

if you hardcoded them in user_config.h they will work.

Now, I will try again with last core.

ascillato, 2018-10-12:
Hi
we, made several tests, and using the core from  https://github.com/d-a-v/arduino/tree/parserequest and we could not make it work with Alexa
I don't know why it is being ignored the alexa request in the code

If I just change in the last actual core only the line 198 from parsing.cpp to `if(!isEncoded||(0==_currentArgCount)){`
Alexa works
I don't know what is happening with the rest of the parsing you have added that this line stop working.

ascillato, 2018-10-12:
Hi, 

tested the OTA URL with last core and the same result.

ascillato, 2018-10-12:
[debug_OTA_URL.txt](https://github.com/esp8266/Arduino/files/2474593/debug_OTA_URL.txt)

earlephilhower, 2018-10-12:
This bit looks mighty fishy:  a quantum httpClient that is both closed and connected.

````
....
[HTTP-Client][end] tcp is closed
[HTTP-Client][begin] url: http://thehackbox.org/tasmota/sonoff-4m.bin
[HTTP-Client][begin] host: thehackbox.org port: 80 url: /tasmota/sonoff-4m.bin
[HTTP-Client] connect. already connected, try reuse!
[HTTP-Client] sending request header
....
````

ascillato, 2018-10-12:
When using core 2.4.2 it works fine. This behaviour is just with the actual Stage core from the codebase

earlephilhower, 2018-10-12:
@ascillato can you please try with 9bc8ea1b58b2a370e1f5911e345661f0ad8bbff8 ?  There was just recently an update to httpClient, and I want to factor that specific commit out.

ascillato, 2018-10-12:
Ok, will try that

earlephilhower, 2018-10-14:
I tried the example in the 1st post here, using the tasmota url (http://thehackbox.org/tasmota/sonoff-4m.bin).  Everything ran like a champ, no issues seen at all, and it d/l'd and restarted as a tasmota something or other.  Does anyone have a repeatably failing example using a public-facing web server and this code?  
````
[SETUP] WAIT 4...
wifi evt: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt 

connected with NOBABIES, channel 10
dhcp client start...
wifi evt: 0
ip:192.168.1.138,mask:255.255.255.0,gw:192.168.1.1
wifi evt: 3
[SETUP] WAIT 3...
[SETUP] WAIT 2...
[SETUP] WAIT 1...
[WIFI][APlistAdd] add SSID: NOBABIES
[HTTP-Client][end] tcp is closed
[HTTP-Client][begin] url: http://thehackbox.org/tasmota/sonoff-4m.bin
[HTTP-Client][begin] host: thehackbox.org port: 80 url: /tasmota/sonoff-4m.bin
[hostByName] request IP for: thehackbox.org
[hostByName] Host: thehackbox.org IP: 81.4.125.198
:ref 1
[HTTP-Client] connected to thehackbox.org:80
[HTTP-Client] sending request header
-----
GET /tasmota/sonoff-4m.bin HTTP/1.0
Host: thehackbox.org
User-Agent: ESP8266-http-Update
Connection: close
x-ESP8266-STA-MAC: 60:01:94:22:B2:F1
x-ESP8266-AP-MAC: 62:01:94:22:B2:F1
x-ESP8266-free-space: 2842624
x-ESP8266-sketch-size: 299424
x-ESP8266-sketch-md5: 8507e8317e171a443a0d7bd41f440975
x-ESP8266-chip-size: 4194304
x-ESP8266-sdk-version: 3.0.0-dev(c0f7b44)
x-ESP8266-mode: sketch

-----
:wr 403 0
:wrc 403 403 0
:ack 403
:rn 536
:rch 536, 536
:rch 1072, 536
:rch 1608, 536
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Sun, 14 Oct 2018 20:03:12 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Server: Apache'
[HTTP-Client][handleHeaderResponse] RX: 'Last-Modified: Sun, 14 Oct 2018 12:01:34 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Accept-Ranges: bytes'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 496704'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: application/octet-stream'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 496704
[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 496704
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 2842624
[httpUpdate]  - current Sketch Size: 299424
[httpUpdate] runUpdate flash...
:pd 4, 2144, 223
:rpi 313, 4
sleep disable
[begin] roundedSize:       0x0007A000 (499712)
[begin] updateEndAddress:  0x00300000 (3145728)
[begin] currentSketchSize: 0x0004A000 (303104)
[begin] _startAddress:     0x00286000 (2646016)
[begin] _currentAddress:   0x00286000 (2646016)
[begin] _size:             0x00079440 (496704)
:c 1, 536, 2144
:c 1, 536, 1608
:c 1, 536, 1072
:c0 1, 536
:rn 536
:rch 536, 536
:c 1, 536, 1072
:c0 1, 536
:rn 536
:rch 536, 536
:c 1, 536, 1072
:c0 1, 536
:rn 536
Header: 0xE9 1 3 40
Set flash mode from 0x3 to 0x2
:rch 536, 536
:rch 1072, 536
:rch 1608, 536
:c 1, 536, 2144
:c 1, 536, 1608
:c 1, 536, 1072
:c0 1, 536
:rn 536
:rch 536, 536
:c 1, 536, 1072
:c0 1, 536
.... (lots of traffic) ....

:rn 536
:rch 536, 536
:rch 1072, 55
:rcl
:abort
:c 1, 536, 1127
:c 1, 536, 591
:c0 1, 55
Staged: address:0x00286000, size:0x00079440
[httpUpdate] Update ok
[HTTP-Client][end] tcp is closed
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl
wifi evt: 1
STA disconnect: 8

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v2eb5b560
@cp:0
ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_22B2F1, GroupTopic sonoffs) Version 6.2.1.15-2_3_0
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on sonoff-4849 with IP address 192.168.4.1

````

ascillato, 2018-10-15:
@earlephilhower 

Hi, sorry for my late response, was very busy. 
Thanks for testing that.

I tried also the sketch posted at the beginning pointing to the tasmota url (http://thehackbox.org/tasmota/sonoff-4m.bin) and using the core from today (15-oct-2018) (https://github.com/esp8266/Arduino/commit/2eb5b5600a3631cdfb9ddd1150505d82e38f3406) and works fine. No issues.

So, I went to the updated example for HTTP_OTA (because the use of ESPhttpUpdate.update of the first post is deprecated in actual STAGE core) and also works fine.

So, I try to use the OTA URL from inside Tasmota and now with this last version of the core works fine also.

The previous version didn't work.

So, Now I think this issue can be closed now due to the issue is no longer in the core.

ascillato, 2018-10-16:
@earlephilhower 

Hi, sorry for my late response, was very busy. 
Thanks for testing that.

I tried also the sketch posted at the beginning pointing to the tasmota url (http://thehackbox.org/tasmota/sonoff-4m.bin) and using the core from today (15-oct-2018) (https://github.com/esp8266/Arduino/commit/2eb5b5600a3631cdfb9ddd1150505d82e38f3406) and works fine. No issues.

So, I went to the updated example for HTTP_OTA (because the use of ESPhttpUpdate.update of the first post is deprecated in actual STAGE core) and also works fine.

So, to avoid the warning at compilation that says deprecated function, we need to use new syntax for ESPhttpUpdate.update:

Before (deprecated):
`t_httpUpdate_return ret = ESPhttpUpdate.update("http://server/file.bin");`

Now:
```
WiFiClient client;
t_httpUpdate_return ret = ESPhttpUpdate.update(client, "http://thehackbox.org/tasmota/sonoff-4m.bin");
```

So, in Tasmota we were missing the extra `WiFiClient client;` and the update only was working for http updates selecting a local file. OTA URL was working for core 2.4.2. Now with this new line, OTA URL is working fine in Tasmota using actual core STAGE.

So, Now I think this issue can be closed now due to the issue is no longer in the core.

Thanks a lot for the support! :+1:

BattistellaM, 2018-10-19:
Platform
Hardware:	ESP-12F
Core Version:	2.4.2 (2.5.0 second test)
Development Env:	Arduino IDE 1.8.7
Operating System:	Windows 10

Web server : Apache  with PHP or Fenix, does not matter...

Problem Description
same HTTP_UPDATE_FAILD Error (6) has reported here above.  
Did the upgrade to commit [https://github.com/esp8266/Arduino/commit/e5493552215f279b0ca3c069d5ebcdfe06773d62](url)  
but lead again  stop at (more or less) same stage.

Here below the debug log captured from Serial1 and a screen shot from WireShark.
Any hint?

Update<CR><LF>
[HTTP-Client][end] tcp is closed<LF>
[hostByName] Host: 192.168.0.54 is a IP!<LF>
:ref 1<CR><LF>
[HTTP-Client] connected to 192.168.0.54:80<LF>
[HTTP-Client] sending request header<LF>
-----<LF>
GET /fw0.16.7.bin HTTP/1.0<CR><LF>
Host: 192.168.0.54<CR><LF>
User-Agent: ESP8266-http-Update<CR><LF>
Connection: close<CR><LF>
x-ESP8266-STA-MAC: 5C:CF:7F:75:82:30<CR><LF>
x-ESP8266-AP-MAC: 5E:CF:7F:75:82:30<CR><LF>
x-ESP8266-free-space: 729088<CR><LF>
x-ESP8266-sketch-size: 316256<CR><LF>
x-ESP8266-sketch-md5: d03cabe40129f13b27168427203eb68b<CR><LF>
x-ESP8266-chip-size: 4194304<CR><LF>
x-ESP8266-sdk-version: 3.0.0-dev(c0f7b44)<CR><LF>
x-ESP8266-mode: sketch<CR><LF>
x-ESP8266-version: 1.16.7<CR><LF>
<CR><LF>
-----<LF>
:wr 418 0<CR><LF>
:wrc 418 418 0<CR><LF>
:ack 418<CR><LF>
:rn 271<CR><LF>
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK'<LF>
[HTTP-Client][handleHeaderResponse] RX: 'Accept-Ranges: bytes'<LF>
[HTTP-Client][handleHeaderResponse] RX: 'ETag: "290240-1537433172525"'<LF>
[HTTP-Client][handleHeaderResponse] RX: 'Date: Fri, 19 Oct 2018 07:49:55 GMT'<LF>
[HTTP-Client][handleHeaderResponse] RX: 'Cache-Control: public, max-age=0'<LF>
[HTTP-Client][handleHeaderResponse] RX: 'Last-Modified: Thu, 20 Sep 2018 08:46:12 GMT'<LF>
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: application/octet-stream'<LF>
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 290240'<LF>
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'<LF>
:c0 1, 271<CR><LF>
[HTTP-Client][handleHeaderResponse] RX: ''<LF>
[HTTP-Client][handleHeaderResponse] code: 200<LF>
[HTTP-Client][handleHeaderResponse] size: 290240<LF>
[httpUpdate] Header read fin.<LF>
[httpUpdate] Server header:<LF>
[httpUpdate]  - code: 200<LF>
[httpUpdate]  - len: 290240<LF>
[httpUpdate] ESP8266 info:<LF>
[httpUpdate]  - free Space: 729088<LF>
[httpUpdate]  - current Sketch Size: 316256<LF>
[httpUpdate]  - current version: 1.16.7<LF>
:rn 1460<CR><LF>
:rch 1460, 1460<CR><LF>
:rch 2920, 1460<CR><LF>
19/10/2018 09:50:00 [SG8] - :rch 4380, 1189<CR><LF>
:rch 5569, 1<CR><LF>
:rn 19<CR><LF>
19/10/2018 09:50:01 [SG8] - :rch 5570, 1<CR><LF>
19/10/2018 09:50:03 [SG8] - :rch 5571, 1<CR><LF>
19/10/2018 09:50:07 [SG8] - :rch 5572, 1<CR><LF>
19/10/2018 09:50:14 [SG8] - :rch 5573, 1<CR><LF>
19/10/2018 09:50:29 [SG8] - :rch 5574, 1<CR><LF>
19/10/2018 09:50:31 [SG8] - :rcl<CR><LF>
:abort<CR><LF>

[image]

ascillato, 2018-10-19:
@BattistellaM 

Hi,

Have you tried using last core version from master branch and the example provided for that?

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/examples/httpUpdate/httpUpdate.ino

It works fine for me now.

CRCinAU, 2018-10-23:
I switched to the staging version of esp8266 framework on PIO using the following in platformio.ini:
```
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
```

ESP version shows:
```
SDK:3.0.0-dev(c0f7b44)/Core:unspecified/lwIP:2.1.0(STABLE-2_1_0_RELEASE/glue:arduino-2.4.2-13-g80224f0)/BearSSL:f55a6ad
```

This now successfully does an OTA update (via URL /update) on my Cisco access points.

I'd like to thank @ivankravets for his help on setting me in the right direction for testing this with PlatformIO.

CRCinAU, 2018-10-24:
I've now started testing with OTA pull from a web server. The code is simple:
```
void checkForUpdate() {
	update_status = "Checking for update...\n";

	WiFiClient client;
	t_httpUpdate_return result = ESPhttpUpdate.update(client, "http://10.1.1.93/arduino/update/");

	update_status += "Returned: ";
	switch(result) {
		case HTTP_UPDATE_FAILED:
			update_status += "Update failed:\nLastError: ";
			update_status += ESPhttpUpdate.getLastError();
			update_status += "\nError: ";
			update_status += ESPhttpUpdate.getLastErrorString().c_str();
			update_status += "\n";
			break;
		case HTTP_UPDATE_NO_UPDATES:
			update_status += "No Update Available.\n";
			break;
		case HTTP_UPDATE_OK:
			update_status += "Updated OK.\n";
			break;
	}
}
```

When this runs, I get the following result:
```
Checking for update...
Returned: Update failed:
LastError: 6
Error: Update error: ERROR[6]: Stream Read Timeout
```

Are there any known issues with running in this method? The hardware is an ESP01_1m - so I don't easily have a serial port to debug with on this particular program. (I might be able to attempt the same on a different device later)

ascillato, 2018-10-24:
you didn't put which file to download

ascillato, 2018-10-24:
The example https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/examples/httpUpdate/httpUpdate.ino works fine for me

CRCinAU, 2018-10-24:
The script on the web side will check the HTTP header posted and return the correct binary file.

The output from the web server is:
```
Wed Oct 24 11:57:43 2018 - W8-Dh-Q8O4Nz3zdhau6UvAAAAEE: Session started. Remote IP: 10.1.1.5, MAC Address: 84:F3:EB:0E:FF:90, SDK Version: 3.0.0-dev(c0f7b44)
Wed Oct 24 11:57:43 2018 - W8-Dh-Q8O4Nz3zdhau6UvAAAAEE: [update] Known device - proceeding...
Wed Oct 24 11:57:43 2018 - W8-Dh-Q8O4Nz3zdhau6UvAAAAEE: [update] Firmware differs from loaded. Sending File. Header: 44a4b64465381a9bdbb9d78f34b99e88, File: 10b91091e175727456d0fded3c350358
Wed Oct 24 11:58:43 2018 - W8-Dh-Q8O4Nz3zdhau6UvAAAAEE: [update] Error sending file. Transfer incomplete!
```

This is via the Cisco AP that didn't update at all using http://$ip/update until this patchset was merged, so not sure if there are still issues.

BattistellaM, 2018-10-25:
> Have you tried using last core version from master branch and the example provided for that?
> 
> https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/examples/httpUpdate/httpUpdate.ino
> 
> It works fine for me now.

Updated to last staging and works for me.  Right now on local network all seem fine, will try soon with external server and let you know the outcomes.
Thanks.

apicquot, 2018-11-12:
I am having the same issue with 2.4.2 version. Is there any tutorial or explanation about how to use a stage version of this code in Arduino IDE or Visual Code ?

Pablo2048, 2018-11-12:
Yes, it is - right in the readme.md -> Using git version ...

bogd, 2018-11-28:
I came here because I seem to have hit this exact same bug while trying to do an OTA HTTP upgrade - I get Stream Read Timeout (with both a remote HTTP server, and a local one, on the same network). I also tried both nginx and apache2, just to exclude any server-specific issue - same result with both.

@Pablo2048 - I'm sorry, but even after going through the readme I cannot figure out how to use the staging version with the Arduino IDE under Windows.

I can see the instructions for cloning the git repo in a certain folder on Mac and Linux - but there is absolutely no mention of Windows in there.

So maybe someone can help me, @apicquot , and others in the same situation - how do we upgrade to the latest staging version? On Windows (Win10, x64), using Arduino IDE?

Thank you!

Pablo2048, 2018-11-28:
@bogd So if you are unable to locate Arduino installation directory try to install Arduino IDE as an portable one (STFW for instructions on how to do it). Then you can put git version inside portable/sketchbook/hardware directory.

apicquot, 2018-11-28:
For Arduino IDE: i simply downloaded the git and replaced the 2.4.2 folder with the new git (I made sure that path to new library was unchanged) I then ran the get.py as mentioned in the README file and it worked. This is not for sure the cleaned way to install it but it worked (I struggle with the README install as well)

bogd, 2018-11-28:
@apicquot - thank you! My problem was that I couldn't find that folder (2.4.2) under Windows. I finally managed to locate it - it is NOT under the Arduino installation directory, but instead it is under /Users/(username)/Library/Arduino15/packages (you can find the exact location in Arduino IDE, under File/Preferences/"more preferences can be edited..." . 

Hopefully this will help someone else struggling with the same issue.

@Pablo2048 - you seem to have a wonderful personality... Nevertheless, thank you for the reply :)

devyte, 2018-12-02:
The original issue here seems fixed (thread has wandered off-topic). Closing.

closed 2018-12-02 by devyte

apicquot, 2018-12-02:
@bogd, did you manage to do the install ? https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version is more detailed: I realised that the stage version has to go in the sketchbook folder - not in the board manager folder. hope it helps

rohitjust24idpl, 2019-02-16:
@CRCinAU @d-a-v @josep112 
Facing this issue on particular WiFi networks(not all). Works fine on few networks. I am sure the network speed is good, download speed is more than 10mbps.

**Debug Output**
```
SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
:ref 1
:wr 414 414 0
:wrc 256 414 0
:wrc 158 158 0
:ack 414
:rn 1380
:rch 1380, 1380
:rch 2760, 1380
:rch 4140, 1380
[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 463216
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 573440
[httpUpdate]  - current Sketch Size: 474320
stopAll 3fff1f94 3fff1f94
:ur 1
[httpUpdate] runUpdate flash...
:pd 4, 5520, 299
:rpi 1081, 4
sleep disable
[begin] roundedSize:       0x00072000 (466944)
[begin] updateEndAddress:  0x00100000 (1048576)
[begin] currentSketchSize: 0x00074000 (475136)
[begin] _startAddress:     0x0008E000 (581632)
[begin] _currentAddress:   0x0008E000 (581632)
[begin] _size:             0x00071170 (463216)
Header: 0xE9 1 2 40
:c 1, 1380, 5520
:c 1, 1380, 4140
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
:rn 1380
:rch 1380, 1380
:c 1, 1380, 2760
:c0 1, 1380
ERROR[6]: Stream Read Timeout
[httpUpdate] Update.writeStream failed! (ERROR[6]: Stream Read Timeout)
[httpUpdate] Update failed
:close
:ur 1
:del
6SPIFFS_close: fd=5
SPIFFS_close: fd=5
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl
wifi evt: 1
STA disconnect: 8

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
```

toribola, 2019-11-12:
Hi,

I have the same issue with SDK 2.6.0
I am unable to do HTTP OTA updates.

Regards,

CRCinAU, 2019-11-12:
> I am unable to do HTTP OTA updates.

Its probably your code. Try looking at these examples - both with on http and https.

https://git.crc.id.au/netwiz/ESP8266_Code

toribola, 2019-11-12:
> > I am unable to do HTTP OTA updates.
> 
> Its probably your code. Try looking at these examples - both with on http and https.
> 
> https://git.crc.id.au/netwiz/ESP8266_Code

I used the Arduino example sketch.

d-a-v, 2019-11-12:
@toribola This is a closed issue.
Please open a new one with a properly filled issue-template, especially with the core version you used and an MCVE (or the name of the included example if applicable).
It is preferred that you try with latest release core-2.6.0 or git version before reporting.