esp8266/Arduino #5025: How do I get rid of all Serial debugs during the boot?

Fabi12345, 2018-08-10:
Hi 

In my project I often reboot the ESP8266 and then read from it over Serial with another MCU.
The ESP always begins with some garbage over the Serial like:
### Debug Messages
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
⸮]pJ⸮!	%⸮⸮⸮bz⸮⸮

And thats really annoying when you wanna parse that message, how can I disable all that previous debug messages, I already opened (almost all I guess) .cpp files in the libraries and commented all debugs, but that one above still remains.

#### Platform

- Hardware:			[ESP-12]
- Core Version:			[2.4.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>

void setup() {
  Serial.begin(9600);
  WiFi.mode(WIFI_STA);
......

```

devyte, 2018-08-10:
You can't. Those chars come from Espressif code, and it can't be shut up. The stream is at 74480bps for most boards, and happens during boot, before the core even starts. 
I understand that the bps can change if you swap the crystal, but I've never seen it done.
Espressif mentions something about rerouting those chars to another pin, but I've also never seen it done.

I suggest implementing your code on the other side of the serial in a way that is tolerant to that garbage. Usually that's done by listening for a startup message with a tiny state machine, e.g.: have your sketch print out "Starting skech" or something, then your other-side code listens for that string by checking every char received.

If that is somehow not possible, then I suggest implementing a hardware transparent gate that boots closed (i.e.: serial line is shut) and you enable it on startup.

Closing.

closed 2018-08-10 by devyte

lrmoreno007, 2018-08-12:
You can send with Serial1 (after Serial1.begin) (in GPIO2 - TX) and receive with Serial (after Serial.begin)(in GPIO3 - RX), but you will can have problems mixing two serials.

Or you can swap Serial after Serial.begin with Serial.swap(), working then with GPIO15 (TX) and GPIO13 (RX).

Garbage go always to GPIO1 (TX).