esp8266/Arduino #5027: ESP SW and HW watchdog resets even when using blink example

enwi, 2018-08-11:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			        ESP-12 ([D-Duino V3](https://www.tindie.com/products/lspoplove/d-duino-v3arduino-and-nodemcu-and-esp8266-and-096oled-2/))
- Core Version:			Commit 559a286f
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio
- Flash Size: 		        4MB (2M SPIFFS)
- lwip Variant:                  v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	        80Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:             115200

### Problem Description

Running any program leads to the ESP8266 crashing, even the simple blink example or blink without delay. There the led will light once, twice or never and the ESP either gets a hardware watchdog reset or a software watchdog reset

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

LED_BUILTIN is in this case 2.

```cpp
int ledState = LOW;

unsigned long previousMillis = 0;
const long interval = 1000;

void setup() {
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  Serial.println(millis());
  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= interval)
  {
    previousMillis = currentMillis;
    if (ledState == LOW)
    {
      ledState = HIGH;  // Note that this switches the LED *off*
    }
    else
    {
      ledState = LOW;  // Note that this switches the LED *on*
    }
    digitalWrite(LED_BUILTIN, ledState);
  }
}
```
### Output of program
Note that I have cut out the output of the print function, so this issue does not get too long. Also this outpu shows the outpu until the ESP stopped working at all.
```
{l $⸮⸮| ⸮l⸮|⸮⸮l⸮b|⸮⸮⸮⸮r⸮b⸮b⸮⸮on⸮lon⸮⸮cbp⸮⸮$s$s$p⸮n⸮⸮l⸮⸮Bo⸮⸮ol⸮⸮c⸮⸮on⸮ l⸮⸮ l ⸮no$`or⸮ےn"⸮lrےnB⸮l⸮⸮l⸮|B⸮⸮⸮l⸮⸮N⸮41
[... 973 lines cut ...]
0x40202745: loop() at /var/folders/14/n8849v691b30wl2lr5d47phm0000gn/T/arduino_modified_sketch_333169/BlinkWithoutDelay.ino line 14
0x40202be8: loop_wrapper() at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 125
```

4:
```
0x401064ae: millis at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.c line 180
0x4020a405: sys_check_timeouts at core/timeouts.c line 325
0x40202b17: loop_task(ETSEvent*) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 133
0x40201a0f: uart_do_write_char at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/uart.c line 303
0x40201e1b: uart_write at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/uart.c line 340
0x40201a11: uart_do_write_char at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/uart.c line 303
0x40201e1b: uart_write at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/uart.c line 340
0x4010649f: millis at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.c line 180
0x4010649c: millis at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.c line 180
0x402027f0: HardwareSerial::write(unsigned char const*, unsigned int) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/HardwareSerial.h line 159
0x40202939: Print::write(char const*) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Print.h line 60
0x402029c5: Print::printNumber(unsigned long, unsigned char) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 260
0x40201eb9: uart_flush at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/uart.c line 376
0x40202016: uart_init at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/uart.c line 438
0x402027f0: HardwareSerial::write(unsigned char const*, unsigned int) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/HardwareSerial.h line 159
0x401064ca: millis at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring.c line 183
0x40202a18: Print::println(unsigned long, int) at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 226
0x40202b7d: esp_schedule() at /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 9
```

5chufti, 2018-08-11:
a) standard flash speed is 40MHz AFAIK
b) did you try with a different board/module to rule out a hw fault?

earlephilhower, 2018-08-12:
Same here, no problem at all.  Looks like a HW problem.  Check another board and your power supply, the ESP can be finicky.

closed 2018-08-12 by earlephilhower

enwi, 2018-08-12:
Unfortunately I don't have another board to test, but I will try to change the flash speed to 40MHz and see if that makes any difference. I mean I can use one of my other ESP boards to test, but those are not the same ESP-12F, those are other ones.

RudyFiero, 2018-08-12:
This issue system is for software issues for the Arduino-ESP8266 framework in this repository. Your problem is not that. Changing code here is not going to fix your problem. If your esp can't run the blink sketch then you most likely have power or other hardware related problems.

enwi, 2018-08-12:
That's what I am thinking too now, but there cannot be any power problem. I've checked that. So probably the ESP is defective and I will exchange it for a new one.

enwi, 2018-08-12:
I've inspected the board a little more and I could make out a solder link between the 3.3V line and the RX connection of the ESP. After removing that the same behaviour as before persisted, but if I don't use any serial communication the ESP does not reset anymore, so it seems that the UART of the ESP might be destroyed causing it to reset, when it is being used. Weirdly enough I can still upload code.

JoseeMallah, 2019-11-03:
Similar issue, please everyone I need help:

`// Modified by Becky Stern 2017
// based on the Adafruit IO Subscription Example

// Adafruit invests time and resources providing this open source code.
// Please support Adafruit and open source hardware by purchasing
// products from Adafruit!
//
// Written by Todd Treece for Adafruit Industries
// Copyright (c) 2016 Adafruit Industries
// Licensed under the MIT license.
//
// All text above must be included in any redistribution.

int currentTemp;
int lastTemp;
int vary = 0;
int decrease = 0;
boolean constant = true;
int current = 0;
int day = 0;

String TC [100];
String C [100];
String AC [100];
String AH [100];
String H [100];
String TH [100];

int nbtc = 0;
int nc = 0;
int nbac = 0;
int nbah = 0;
int nh = 0;
int nbth = 0;

//int nbtc = -1;
//int nc = -1;
//int nbac = -1;
//int nbah = -1;
//int nh = -1;
//int nbth = -1;

int n = 0;

int i = 0;

String w = " ";
int t;

/************************ Adafruit IO Configuration *******************************/

// visit io.adafruit.com if you need to create an account,
// or if you need your Adafruit IO key.
#define IO_USERNAME    "JoseeMallah"
#define IO_KEY         "912c4add0d5f491399cc94406dc0092f"

/******************************* WIFI Configuration **************************************/

#define WIFI_SSID       "Nada"
#define WIFI_PASS       "N1J7H49*aoo"

#include "AdafruitIO_WiFi.h"
AdafruitIO_WiFi io(IO_USERNAME, IO_KEY, WIFI_SSID, WIFI_PASS);

/************************ Main Program Starts Here *******************************/
#include <ESP8266WiFi.h>
#include <AdafruitIO.h>
#include <Adafruit_MQTT.h>
[... 432 lines cut ...]
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v855b91ea
~ld

Esp decoder gives this (thanks to @kavindyasinthasilva):

Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x4000bf64
EXCVADDR: 0x10f5ea77

Decoding stack results
0x4010023c: millis() at C:\Users\User\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_wiring.cpp line 188
0x401006d9: _umm_free(void*) at C:\Users\User\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 1380
0x40202bc8: Adafruit_MQTT::processPackets(short) at C:\Users\User\Documents\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT.cpp line 418
0x402097d8: Adafruit_MQTT_Client::connected() at C:\Users\User\Documents\Arduino\libraries\Adafruit_MQTT_Library-master\Adafruit_MQTT_Client.cpp line 48
0x40201f84: AdafruitIO::run(unsigned short) at C:\Users\User\Documents\Arduino\libraries\Adafruit_IO_Arduino-master\src\AdafruitIO.cpp line 179
0x402017df: loop() at C:\Users\User\Documents\Arduino\Project_NodeMCU/Project_NodeMCU.ino line 142
0x4010015c: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\User\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 156
0x40207c6d: esp_schedule() at C:\Users\User\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 109
0x40207d90: loop_wrapper() at C:\Users\User\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 176