esp8266/Arduino #5033: Using pinMode, attachInterrupts() and Wifi Access Point breaks

WolfBSail, 2018-08-13:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12E]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[no idea]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I have a similar issue as #1020, using ESP12E board. When I am declaring certain pins, using attachInterrupts() and setting up Access Point WiFi in the same void setup() the wifi stops working. I have tested out that if I only declare pins D0, D1, D2, D3, D4, A0 and set up WiFi without the interrupts it does work. Please seen my setup code below!
I have tried detachInterrupt(), but it did not work.

Does anyone have the same problem? Is there a solution to this problem?

Thank you in advance.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <WebSocketsServer.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <ESP8266WiFi.h>
#include <Wire.h>       //I2C library
#include <RtcDS3231.h>  //RTC library

WebSocketsServer webSocket = WebSocketsServer(81); // websocket instance

void setup() {
  pinMode(4, INPUT);
  pinMode(3, INPUT);
  pinMode(2, INPUT);  
  pinMode(1, OUTPUT);  
  pinMode(0, OUTPUT);
  
  Serial.begin(9600);

  rtcObject.Begin();  
  attachInterrupt(digitalPinToInterrupt(secPin), secondReceive, FALLING);
  attachInterrupt(digitalPinToInterrupt(voltagePin), voltageDrop, FALLING);

  // Init EEPROM
  EEPROM.begin(512);

  // look at whether power-cut time correction is neccessary
  if(EEPROM.read(addr[3])==1){
        
    clockEnabled=1;
    startTime=millis();
    Serial.println();
    Serial.println("Blackout has happened, time correction...");
    Serial.println();
    getRtcTime();
    EEPROM.write(addr[6], rtcHour);    
    EEPROM.write(addr[7], rtcMinute);    
    EEPROM.write(addr[8], rtcSecond);    
    powerCutTimeCorrection();    
    EEPROM.write(addr[3], 0);   
    EEPROM.commit();
    loopTime=millis();
  }
  else{
    configRTC(ntpYear, ntpMonth, ntpDay,ntpHour, ntpMinute, ntpSecond);     
    RtcDateTime currentTime = rtcObject.GetDateTime();
    rtcObject.SetDateTime(currentTime);
  }

  //init rtc functions
  rtcObject.SetSquareWavePin(DS3231SquareWavePin_ModeClock);
  rtcObject.SetSquareWavePinClockFrequency(DS3231SquareWaveClock_1Hz); //SQW set to 1Hz

  detachInterrupt(digitalPinToInterrupt(secPin));
  detachInterrupt(digitalPinToInterrupt(voltagePin));

  //creating the ESP's own network
  
  WiFi.disconnect();
  WiFi.mode(WIFI_AP);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
  WiFi.softAP(ssidESP, passwordESP);
  Serial.print("Connected to ");
  Serial.println(ssidESP);
  Serial.print("Access Point IP address = ");
  Serial.println(WiFi.softAPIP());
  Serial.println("");
 

  handleWebsite();   // crating html string;
  dnsServer.start(DNS_PORT, "*", apIP);
  server.on("/", handleWebsite);
  server.on("/generate_204", handleWebsite); // for android captive
  server.on("/fwlink", handleWebsite);       // windows captive
  server.begin();

  server.onNotFound([]() {
    server.send(200, "text/html", webSite); 
  });

  webSocket.begin();
  webSocket.onEvent(webSocketEvent);
  
}

void loop() {  
  dnsServer.processNextRequest(); //maintains webserver status  
  server.handleClient();   
  webSocket.loop();
  jsonReq();//JSON Request function  
  clockWorkOrange();              //Main clock function                      
}

devyte, 2018-08-13:
You mention D numbering, but your code uses GPIO numbering, and they're not the same.
E.g.: on a nodemcu board, D1 is GPIO5 and GPIO1 is TX.
Your code is using GPIO0-4, and some of those are not allowed for user applications. I suggest you study the board pinout and how it connects to the ESP12 pinout.
Closing as user error.

closed 2018-08-13 by devyte

WolfBSail, 2018-08-13:
Even tough, it does not change the main issue... please see the whole
question!