esp8266/Arduino #5036: Compile/Upload failure when using some debug features in 2.4.2 ONLY

All-Round-Tinkerer, 2018-08-13:
### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-8266]
- Core Version:			[2.4.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Adafruit Feather Huzzah]
- Flash Mode:			[?]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[?]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [921600] 

### Problem Description

When using some debug systems (like Core or OOM) the sketch refuses to compile/upload, most likely due to a conflict with ESP8266mDNS in Arduino OTA.

Is there any other way debugging could be done? I included the minimal code that still reproduces the problem.

Is it just my configuration or an actual bug?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ArduinoOTA.h>

void setup() {
}
void loop() {

}
```

### Debug Messages

```
Arduino: 1.8.5 (Linux), Board: "Adafruit Feather HUZZAH ESP8266, 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Serial, SSL+TLS_MEM+HTTP_CLIENT+HTTP_SERVER+CORE+WIFI+HTTP_UPDATE+UPDATER+OTA+OOM, Only Sketch, 921600"

/home/simon/Downloads/arduino-1.8.5-linux64/arduino-1.8.5/arduino-builder -dump-prefs -logger=machine -hardware /home/simon/Downloads/arduino-1.8.5-linux64/arduino-1.8.5/hardware -hardware /home/simon/.arduino15/packages -tools /home/simon/Downloads/arduino-1.8.5-linux64/arduino-1.8.5/tools-builder -tools /home/simon/Downloads/arduino-1.8.5-linux64/arduino-1.8.5/hardware/tools/avr -tools /home/simon/.arduino15/packages -built-in-libraries /home/simon/Downloads/arduino-1.8.5-linux64/arduino-1.8.5/libraries -libraries /home/simon/Arduino/libraries -fqbn=esp8266:esp8266:huzzah:CpuFrequency=80,VTable=flash,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Serial,DebugLevel=SSLTLS_MEMHTTP_CLIENTHTTP_SERVERCOREWIFIHTTP_UPDATEUPDATEROTAOOM,FlashErase=none,UploadSpeed=921600 -ide-version=10805 -build-path /tmp/arduino_build_171356 -warnings=none -build-cache /tmp/arduino_cache_228890 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/home/simon/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=/home/simon/.arduino15/packages/esp8266/tools/mkspiffs/0.2.0 -prefs=runtime.tools.esptool.path=/home/simon/.arduino15/packages/esp8266/tools/esptool/0.4.13 -verbose /home/simon/Arduino/bug_report/bug_report.ino
/home/simon/Downloads/arduino-1.8.5-linux64/arduino-1.8.5/arduino-builder -compile -logger=machine -hardware /home/simon/Downloads/arduino-1.8.5-linux64/arduino-1.8.5/hardware -hardware /home/simon/.arduino15/packages -tools /home/simon/Downloads/arduino-1.8.5-linux64/arduino-1.8.5/tools-builder -tools /home/simon/Downloads/arduino-1.8.5-linux64/arduino-1.8.5/hardware/tools/avr -tools /home/simon/.arduino15/packages -built-in-libraries /home/simon/Downloads/arduino-1.8.5-linux64/arduino-1.8.5/libraries -libraries /home/simon/Arduino/libraries -fqbn=esp8266:esp8266:huzzah:CpuFrequency=80,VTable=flash,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Serial,DebugLevel=SSLTLS_MEMHTTP_CLIENTHTTP_SERVERCOREWIFIHTTP_UPDATEUPDATEROTAOOM,FlashErase=none,UploadSpeed=921600 -ide-version=10805 -build-path /tmp/arduino_build_171356 -warnings=none -build-cache /tmp/arduino_cache_228890 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/home/simon/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=/home/simon/.arduino15/packages/esp8266/tools/mkspiffs/0.2.0 -prefs=runtime.tools.esptool.path=/home/simon/.arduino15/packages/esp8266/tools/esptool/0.4.13 -verbose /home/simon/Arduino/bug_report/bug_report.ino
Using board 'huzzah' from platform in folder: /home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2
Using core 'esp8266' from platform in folder: /home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2
Detecting libraries used...
"/home/simon/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/include" "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/lwip2/include" "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_171356/core" -c -w  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_SSL -DDEBUG_ESP_TLS_MEM -DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_HTTP_SERVER -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_UPDATE -DDEBUG_ESP_UPDATER -DDEBUG_ESP_OTA -DDEBUG_ESP_OOM -DARDUINO=10805 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP12"   -DESP8266 "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266" "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/variants/adafruit" "/tmp/arduino_build_171356/sketch/bug_report.ino.cpp" -o "/dev/null"
"/home/simon/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/include" "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/lwip2/include" "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_171356/core" -c -w  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_SSL -DDEBUG_ESP_TLS_MEM -DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_HTTP_SERVER -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_UPDATE -DDEBUG_ESP_UPDATER -DDEBUG_ESP_OTA -DDEBUG_ESP_OOM -DARDUINO=10805 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP12"   -DESP8266 "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266" "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/variants/adafruit" "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ArduinoOTA" "/tmp/arduino_build_171356/sketch/bug_report.ino.cpp" -o "/dev/null"
"/home/simon/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/include" "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/lwip2/include" "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_171356/core" -c -w  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_SSL -DDEBUG_ESP_TLS_MEM -DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_HTTP_SERVER -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_UPDATE -DDEBUG_ESP_UPDATER -DDEBUG_ESP_OTA -DDEBUG_ESP_OOM -DARDUINO=10805 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP12"   -DESP8266 "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266" "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/variants/adafruit" "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ArduinoOTA" "-I/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WiFi/src" "/tmp/arduino_build_171356/sketch/bug_report.ino.cpp" -o "/dev/null"
[... cut ...]
/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/libc/xtensa-lx106-elf/include/assert.h:18:29: note: in expansion of macro 'PSTR'
              __ASSERT_FUNC, PSTR(#__e)))
                             ^**
/home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WiFi/src/include/UdpContext.h:154:9: note: in expansion of macro 'assert'
         assert(isValidOffset(pos));
         ^P8266mDNS in folder: /home/simon/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266mDNS (legacy)
exit status 1
Error compiling for board Adafruit Feather HUZZAH ESP8266.
```

earlephilhower, 2018-08-13:
This is a segment problem related to `ICACHE_RODATA_ATTR` which I don't yet understand. 

Removing that decorator from the static variable in the debug `#define malloc()` macro in `umm_malloc_cfg.h` lets things build fine (but places lots of strings in RAM which is not a good thing).

d-a-v, 2018-08-13:
related: #4650

and worth reading: #2078

I think we should introduce a second `assert` function that emits no error message, that still could be traced back thanks to the stack dump and stack decoder, then we should use it there:
[libraries/ESP8266WiFi/src/include/UdpContext.h:154:9](https://github.com/esp8266/Arduino/blob/a0e6d22ac1adc52b3fb9e3b37e8fd1c036ff227d/libraries/ESP8266WiFi/src/include/UdpContext.h#L154)
(because this particular one beeing in a `.h` file is in the way)

and / or

Move this assert in a separate `.cpp` file.

earlephilhower, 2018-08-13:
If you manually change the segment you can get this specific problem to go away:  `__attribute__((section(".irom.text.mallocdebug")))` in place of icache_ro_data and store_attr in the umm_malloc_cfg.h file builds working executables.

It's not a general fix because, per GCC manuals, `section` should only be used on _global_ variables:
https://gcc.gnu.org/onlinedocs/gcc-3.2/gcc/Variable-Attributes.html
````
Use the section attribute with an initialized definition of a global variable, as shown in the example. GCC issues a warning and otherwise ignores the section attribute in uninitialized variable declarations.

You may only use the section attribute with a fully initialized global definition because of the way linkers work. The linker requires each object be defined once, with the exception that uninitialized variables tentatively go in the common (or bss) section and can be multiply "defined". You can force a variable to be initialized with the -fno-common flag or the nocommon attribute. 
````

All-Round-Tinkerer, 2018-08-14:
@earlephilhower Thanks for this, your suggestion solved my problem! Any idea if this will be fixed in the next releases?

PR #5048 (2018-08-15): Move all PROGMEM to their own section
According to the GCC man page, __section__ attributes should only be used
for global variables.  However, the PROGMEM and ICACHE_RODATA macros use
this variable decorator even for local variables.  Most of the time it works,
but when a static or inlined function tries to use a PROGMEM/PSTR/etc.
variable the compiler can throw an error like:
  error: XXX causes a section type conflict with YYY

Change the PROGMEM macro to emit a section name that is unique (a combo
of the file, line, and counter variables to ensure uniqueness).  The
standard linker script w [... cut]
commit 56b98fd4 (2018-08-15): Move all PROGMEM to their own section (#5048)

According to the GCC man page, __section__ attributes should only be used
for global variables.  However, the PROGMEM and ICACHE_RODATA macros use
this variable decorator even for local variables.  Most of the time it works,
but when a static or inlined function tries to use a PROGMEM/PSTR/etc.
variable the compiler can throw an error like:
  error: XXX causes a section type conflict with YYY

Change the PROGMEM macro to emit a section name that is unique (a combo
of the file, line, and counter variables to ensure uniqueness).  The
standard linker [... cut]
closed 2018-08-15 by devyte