esp8266/Arduino #5039: Exception Type 3 while using WiFiClientSecure

thomasDOTwtf, 2018-08-14:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					ESP8266
- Core Version:			 a0e6d22
- Development Env:		Platformio
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			default
- Flash Size: 		4MB
- lwip Variant:   v1.4
- Reset Method:		nodemcu
- Flash Frequency:	default
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

After creating a WiFiClientSecure connection and sending some data I get a type 3 exception.

I'm able to reproduce that issue with the current master and with both SSL implementations.

I tried to switch between BearSSL::WiFiClientSecure and axTLS::WiFiClientSecure both are failing.

It doesn't matter if the readStringUntil and read code is included or not the exception occurs anyways.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Adafruit_NeoPixel.h>
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <helper.h>
#include <sfbucwa.h>

#define WIFI_SSID        "mywifi"
#define WIFI_PASSWORD    "supersecure"
#define ARDUINO_HOSTNAME "status-cat"
#define OTApwd "veryhighsecure"
#define PIXEL_PIN    4
#define PIXEL_COUNT 1

SkypeUCWA skype = SkypeUCWA("example.de");
Adafruit_NeoPixel led = Adafruit_NeoPixel(PIXEL_COUNT, PIXEL_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
    led.begin();
    led.show();
    led.setPixelColor(0, 255, 255, 0);
    led.show();
    Serial.begin(9600);
    Serial.println("Booting");
    WiFi.mode(WIFI_STA);
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    while (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.println("Connection Failed! Rebooting...");
[... 132 lines cut ...]

```
Booting
Ready
IP address: 10.42.13.56
Heap Before: 40560
Heap After: 14512
HTTP/1.1 200 OK
Cache-Control: no-store
Pragma: no-cache
Content-Type: application/json
X-MS-Server-Fqdn: xxxxxxxxxxxxxxxxxxxxxx
X-MS-Correlation-Id: xxxxxxxxxxxxxxxxxxxxx
client-request-id: xxxxxxxxxxxxxxxxxxxxxxxxxxxx
Strict-Transport-Security: max-age=31536000; includeSubDomains
X-Content-Type-Options: nosniff
Date: Tue, 14 Aug 2018 00:52:26 GMT
Connection: keep-alive
Content-Length: 253
Strict-Transport-Security: max-age=31536000; includeSubDomains; preload
Strict-Transport-Security: max-age=31536000; includeSubDomains

headers received
{"access_token":"xxxxxxxxxxxxxxxxxxxxxxx","expires_in":27874,"ms_rtc_identityscope":"local","token_type":"Bearer"}Fatal exception 3(LoadStoreErrorCause):
```

earlephilhower, 2018-08-14:
You've run out of memory most likely.  Add in some `ESP.getFreeHeap()` printouts throughout the code and you'll see it shrink until the `String`, which doesn't have exceptions since they are not supported, end up getting a NULL to a malloc somewhere.

thomasDOTwtf, 2018-08-14:
Hey that's very unlikely as I think because I already sorted that out as I thought.

There is already a getFreeHeap() before and after the request is performed. 

```
Heap Before: 40560
Heap After: 14512
```

It doesn't matter if the readStringUntil and read below the last freeHeap command is included or not the exception occurs anyways.

thomasDOTwtf, 2018-08-14:
It has been the passed led argument in the Wheel method.
Changed definition to Adafruit_NeoPixel* and passed it with &led.
Using led->Color inside the method. Thanks anyways and sorry for bothering.

closed 2018-08-14 by thomasDOTwtf