esp8266/Arduino #5042: Error in use WiFiClientSecure.

c0otlass, 2018-08-14:
### Basic Info
i used WiFiClientSecure and create an function to get command http client, but it crashed
#### Platform

- Hardware:					[|ESP8285 device]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[|Wemos D1 mini r2]
- Flash Size: 		[4MB]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

my code crashed and stack decoder could not decode it.

### Sketch

```cpp
/*
 *  HTTP over TLS (HTTPS) example sketch
 *
 *  This example demonstrates how to use
 *  WiFiClientSecure class to access HTTPS API.
 *  We fetch and display the status of
 *  esp8266/Arduino project continuous integration
 *  build.
 *
 *  Created by Ivan Grokhotkov, 2015.
 *  This example is in public domain.
 */

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char* ssid = "....";
const char* password = ".....";
String GetCommand(String url,char* host,int httpsPort){
    WiFiClientSecure client;
  Serial.print("connecting to ");
  Serial.println(host);
  if (!client.connect(host, httpsPort)) {
    Serial.println("connection failed");
    return "";
  }

    Serial.print("requesting URL: ");
  Serial.println(url);

  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "User-Agent: ESP8266\r\n" +
               "Connection: close\r\n\r\n");

  Serial.println("request sent");
  while (client.connected()) {
    String line = client.readStringUntil('\n');
    if (line == "\r") {
      Serial.println("headers received");
      break;
    }
  }
  String line = client.readStringUntil('z');
  Serial.println("reply was:");
  Serial.println("==========");
  Serial.println(line);
  Serial.println("==========");
  Serial.println("closing connection");
}
void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
    String resualt = GetCommand("/get","httpbin.org",443);
  Serial.println(resualt);
  // Use WiFiClientSecure class to create TLS connection

}

void loop() {

      delay(1000);

}
```

### Debug Messages

```
WiFi connected
IP address: 
192.168.137.233
connecting to httpbin.org
requesting URL: /get
request sent
headers received
reply was:
==========
{
  "args": {}, 
  "headers": {
    "Connection": "close", 
    "Host": "httpbin.org", 
    "User-Agent": "ESP8266"
  }, 
  "origin": "5.117.192.70", 
  "url": "https://httpbin.org/get"
}

==========
closing connection
~
```

thomasDOTwtf, 2018-08-14:
you can decode your stack in the following scheme:
java -jar EspStackTraceDecoder.jar <pathto>xtensa-lx106-elf-addr2line.exe <pathto>\firmware.elf <pathto>\stack.txt

earlephilhower, 2018-08-18:
Coding error.

The compiler warns that you are not returning any value from GetCommand(), so the serial.printlin(resultat) prints garbage which causes a WDT since it's not 0-terminated.

The stack dump decodes fine for my WDT crash, so I think you also have some issues with your installation of the toolset.

closed 2018-08-18 by earlephilhower