esp8266/Arduino #5044: Error in expansion of macro malloc, realloc

luffykesh, 2018-08-14:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ x] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[2.4.2]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Nodemcu] (not tested in other)
- Flash Mode:			[qio]
- Flash Size: 		[4MB(3MSPIFFS)]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)
- **Debug Level: OOM**

### Problem Description

The Code fails to compile when Debug Level is set to OOM. The errors are embedded below.
**Error 1**
Note: Error in expansion of ```realloc``` and ```malloc``` macros
```cpp
In file included from /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/Arduino.h:307:0,
                 from /var/folders/f8/2_2p327x6sz15httt766r7c00000gn/T/arduino_build_602235/sketch/sketch_oct27a.ino.cpp:1:
/Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/umm_malloc/umm_malloc_cfg.h:183:43: error: mem_debug_file causes a section type conflict with mem_debug_file
 #define realloc(p,s) ({ static const char mem_debug_file[] ICACHE_RODATA_ATTR STORE_ATTR = __FILE__; realloc_loc(p, s, mem_debug_file, __LINE__); })
                                           ^
/Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/include/mem.h:45:20: note: in expansion of macro 'realloc'
 #define os_realloc realloc
                    ^
/Users/rushikesh/Documents/Arduino/sketch_oct27a/sketch_oct27a.ino:15:16: note: in expansion of macro 'os_realloc'
  temp = (char*)os_realloc(buffer,2000);
                ^
In file included from /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/Arduino.h:307:0,
                 from /var/folders/f8/2_2p327x6sz15httt766r7c00000gn/T/arduino_build_602235/sketch/sketch_oct27a.ino.cpp:1:
/Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/umm_malloc/umm_malloc_cfg.h:181:40: note: 'mem_debug_file' was declared here
 #define malloc(s) ({ static const char mem_debug_file[] ICACHE_RODATA_ATTR STORE_ATTR = __FILE__; malloc_loc(s, mem_debug_file, __LINE__); })
                                        ^
/Users/rushikesh/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/DynamicJsonBuffer.hpp:26:12: note: in expansion of macro 'malloc'
     return malloc(size);
            ^
```
**Error 2**
**If both ```os_realloc``` and ```os_malloc``` calls are removed, the below error is thrown.**
Note: Error in``` FPSTR``` and ```PSTR``` also.
```cpp
/Users/rushikesh/Documents/Arduino/sketch_oct27a/sketch_oct27a.ino: In function 'void foo()':
/Users/rushikesh/Documents/Arduino/sketch_oct27a/sketch_oct27a.ino:16:2: warning: 'temp' is used uninitialized in this function [-Wuninitialized]
  if(temp==NULL) os_free(buffer);
  ^
/Users/rushikesh/Documents/Arduino/sketch_oct27a/sketch_oct27a.ino:16:32: warning: 'buffer' may be used uninitialized in this function [-Wmaybe-uninitialized]
  if(temp==NULL) os_free(buffer);
                                ^
In file included from /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/Arduino.h:259:0,
                 from /var/folders/f8/2_2p327x6sz15httt766r7c00000gn/T/arduino_build_602235/sketch/sketch_oct27a.ino.cpp:1:
/Users/rushikesh/Documents/Arduino/sketch_oct27a/sketch_oct27a.ino: At global scope:
/Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/pgmspace.h:16:51: error: __c causes a section type conflict with mem_debug_file
 #define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
                                                   ^
/Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.h:38:76: note: in definition of macro 'FPSTR'
 #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
                                                                            ^
/Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.h:39:34: note: in expansion of macro 'PSTR'
 #define F(string_literal) (FPSTR(PSTR(string_literal)))
                                  ^
/Users/rushikesh/Documents/Arduino/sketch_oct27a/sketch_oct27a.ino:6:17: note: in expansion of macro 'F'
  Serial.println(F("Flash string"));
                 ^
In file included from /Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/Arduino.h:307:0,
                 from /var/folders/f8/2_2p327x6sz15httt766r7c00000gn/T/arduino_build_602235/sketch/sketch_oct27a.ino.cpp:1:
/Users/rushikesh/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/umm_malloc/umm_malloc_cfg.h:181:40: note: 'mem_debug_file' was declared here
 #define malloc(s) ({ static const char mem_debug_file[] ICACHE_RODATA_ATTR STORE_ATTR = __FILE__; malloc_loc(s, mem_debug_file, __LINE__); })
                                        ^
/Users/rushikesh/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/DynamicJsonBuffer.hpp:26:12: note: in expansion of macro 'malloc'
     return malloc(size);
            ^
```

**Note: Compilation will be successful if DyanamicJsonBuffer's object is not created**

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <ArduinoJson.h>
extern "C" {
#include "mem.h"
}
void setup(){
	Serial.println(F("Flash string"));
}
void loop(){}

void foo()
{
	char *buffer,*temp;
	// error in FPSTR & PSTR if os_malloc and os_realloc calls are removed
	buffer = (char*)os_malloc(1000);
	temp = (char*)os_realloc(buffer,2000);
	if(temp==NULL) os_free(buffer);
	else os_free(temp);
	// code will compile successfully if, jsonBuffer is not created
	DynamicJsonBuffer jsonBuffer(1000);
}
```

earlephilhower, 2018-08-14:
Thanks, @luffykesh.  It's been isolated in #5036 with a 1-line MCVE, believe it or not.  There's a temporary workaround there, if you're stuck right now.

devyte, 2018-08-14:
Looks like this and #5036 are the same underlying issue. Closing this in favor of #5036.

closed 2018-08-14 by devyte