esp8266/Arduino #5051: SPIFFS is not retaining data after power off

neerajneo, 2018-08-16:
Hi,
I am using SPIFF in esp8266. But when I am not able to get the data once my device has been powered off. Is my assumption right that SPIFF data can be retained after power off. Below is my code:-
#include <ESP8266WiFi.h>
#include <FS.h>

const char* filename = "/samplefile.txt";

void setup()
{
  int i;
  delay(1000);
  Serial.begin(115200);
  Serial.println();
  File f;

  if (SPIFFS.begin())
  {
    Serial.println("SPIFFS Initialize....ok");
  }
  else
  {
    Serial.println("SPIFFS Initialization...failed");
  }

  if (SPIFFS.format())
  {
    Serial.println("File System Formated");
  }
  else
  {
    Serial.println("File System Formatting Error");
  }

  if (SPIFFS.exists(filename))
  {
    Serial.println("File exist");
    f = SPIFFS.open(filename, "a+");
    if (!f)
    {
      Serial.println("file open failed");
    }
    else
    {
      Serial.println("Writing Data to File2");
      f.print("This is  written in file2\n");
    }
  }

  else if (!(SPIFFS.exists(filename)))
  {
    Serial.println("File does not exist");
    File f = SPIFFS.open(filename, "w");
    if (!f)
    {
      Serial.println("file open failed");
    }
    else
    {
      //Write data to file
      Serial.println("Writing Data to File2");
      f.print("This is written in file1\r\n");
    }

  }
  f.close();

  if (SPIFFS.exists(filename))
    Serial.println("File exist2");

  f = SPIFFS.open(filename, "r");

  Serial.println("---------------------------------------");
  for (i = 0; i < f.size(); i++) //Read upto complete file size
  {
    Serial.print((char)f.read());
  }
  Serial.println("---------------------------------------");
  f.close();
  Serial.println("File Closed");
}

void loop()
{
  delay(500);

}

Please help with this. All the help is highly appreciated.

Pablo2048, 2018-08-16:
@neerajneo in your code there is:
`if (SPIFFS.format())`
so the filesystem is formatted (aka all data erased) at every setup()...

5chufti, 2018-08-16:
IIRC spiffs will automatically be formated if SPIFFS.begin() does not find the "magic bytes" in memory.