esp8266/Arduino #5053: WiFiClientSecure  loadCACert  always returns true

romelemperado, 2018-08-16:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

I can connect to the mqtt broker using self signed certs but when I try to disabled my wifi/router I can still see that this line  `bool ret = espClient.loadCACert(ca);` returns true even the absense of internet? If that is the case then it is not really verifying? I need your thoughts on this issue..
```
boolean checkCertificates(){

  // Get certs from file system and load into WiFiSecure client

  SPIFFS.begin();
  File ca = SPIFFS.open(CA_CERT_FILE, "r");
  if(!ca) {
    Serial.println("Couldn't load CA cert");
  } else {
    bool ret = espClient.loadCACert(ca);
    Serial.print("Loading CA cert returned ");
    Serial.println((ret)? "true" : "false");
    ca.close();

  }
```
}

Eszartek, 2018-08-16:
Your program is reading  the CA_CERT_FILE from flash storage, not the internet. You don't need internet access to read from a  flash file system.

romelemperado, 2018-08-16:
Closing this issue.. Thank you Eszartek... :)

closed 2018-08-16 by romelemperado