esp8266/Arduino #5059: spi in slave mode :: to recieve data from master side.

prateek027, 2018-08-18:
Hi All,
i am using nodeMCU ESP8266 (amica & lolin) and ble module. 
from ble module i used to send some data to server using wifi.
so i need spi slave program for esp8266 which receive data in bytes (not 32 bytes receive at single time).

JiriBilek, 2018-08-18:
AFAIK, it is not possible. The SPI slave mode in 8266 is constrained to 4 or 32 byte messages.

prateek027, 2018-08-18:
Hi Jiri,
So, you want say that ESP8266 receive only 4 bytes or 32 bytes, fine.
but when send 4 bytes through master then it not show any data whenever it is not completed 32 Bytes.

it also not show me the exact data what i am sending from master.
so if you have any source code then pass to me.

Thanks

JiriBilek, 2018-08-18:
Hi,
the source code is here:
https://github.com/esp8266/Arduino/tree/master/libraries/SPISlave/src

prateek027, 2018-08-18:
Hi,
i am already using this code. the problem is with this code is :: we cant receive correct byte order. can you told me in which order it receive data from master like MSB first or lsb first. 
or it have any specific order to receive and display to uart terminal. 

im using like this:;

SPISlave.onData([](uint8_t * data, size_t len) {
for(i=0; i<len; i++) {
    Serial.printf(data[i],HEX);
}

it is ok or not.

prateek027, 2018-08-20:
Hi, just tell me one thing.

what is the clock rate of spi and which mode you are using in spi slave