esp8266/Arduino #5060: SHA256 Signature Problem with 2.4.2

kherNI, 2018-08-18:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12
- Core Version:			Boards Manager - 2.4.2
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			NodeMCU 1.0 (ESP-12E Modules
- Flash Mode:			?
- Flash Size: 		        4MB
- lwip Variant:                  v2 Lower Memory
- Reset Method:		?
- Flash Frequency:	        ?
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:              256000

### Problem Description

We have a sketch that involves making a POST to Azure IoTHub.  To do this we have to create a signature using SHA256.  All versions prior to 2.4.2 work; however, with 2.4.2 the server responds with an error 401 : Unauthorized, which is typically the result of an ill prepared signature.  The SHA256 library has external dependencies to:
     - <string.h>
     - <inttypes.h>
     - "Print.h"

I wish I could provide more details but I'm at a loss on how to provide an MVCE for others that could replicate this error but I've included some conceptual code below.  In looking at the changelog for 2.4.2 there appears to be changes to string.h.

```
loop{
      initializeIotHub();
      tlsClient.print(buildHttpPOSTRequest(sensorData));
}

void initializeIotHub() {
  String url = urlEncode(cloud.host) + urlEncode(TARGET_URL) + (String)cloud.id;
  cloud.fullSas =  createIotHubSas(cloud.key, url);
  yield();
}

String buildHttpPOSTRequest(String data) {
  String Request;
  Request = F("POST /devices/");
  Request += ID;
  Request += POST_END_POINT;
  Request += F(" HTTP/1.1\r\nHost: ");
  Request += HOST;
  Request += F("\r\nAuthorization: SharedAccessSignature ");
  Request += SAS;
  Request += F("\r\nContent-Type: application/atom+xml;type=entry;charset=utf-8\r\nContent-Length: ");
  Request += data.length();
  Request += F("\r\n\r\n");
  Request += data;
  return Request;
}

String createIotHubSas(char *key, String url) {
  String stringToSign = url + "\n" + EXPIRYDATE;   /EXPIRYDATE = 1924991999 - Expires Tuesday, December 31, 2030 11:59:59 PM GMT
  int keyLength = strlen(key);

  int decodedKeyLength = base64_dec_len(key, keyLength);
  char decodedKey[decodedKeyLength];

  base64_decode(decodedKey, key, keyLength); 

  Sha256.initHmac((const uint8_t*)decodedKey, decodedKeyLength);
  Sha256.print(stringToSign);
  char* sign = (char*) Sha256.resultHmac();
  // END: Create signature

  // START: Get base64 of signature
  int encodedSignLen = base64_enc_len(HASH_LENGTH);
  char encodedSign[encodedSignLen];
  base64_encode(encodedSign, sign, HASH_LENGTH);

  // SharedAccessSignature
  return "sr=" + url + "&sig=" + urlEncode(encodedSign) + "&se=" + cloud.sasExpiryDate;
}
````

NdK73, 2018-08-19:
Without knowing which SHA256 lib you're using I can't be sure, but this code
```char* sign = (char*) Sha256.resultHmac();```
smells wrong.
You're forcibly casting Sha256.resultHmac() result to a char *, then you assume that that char * points to a buffer. That's true only if resultHmac allocates a buffer (that you'll have to free) or uses a static one (making it not-reentrant). If it encodes the HMAC in a long int, you're accessing random memory...

yoursunny, 2018-08-26:
`Sha256.resultHmac()` appears to be [Cryptosuite](https://github.com/Cathedrow/Cryptosuite). It does return a static buffer of `uint8_t*` type.

kherNI, 2018-08-26:
Sorry for my delay.  I'm not sure where the library came from, as the code originated from a Microsoft example.  Below are the SHA256 details:
```
#include <string.h>
#include "sha256.h"

uint32_t sha256K[] = {
  0x428a2f98,0x71374491,0xb5c0fbcf,0xe9b5dba5,0x3956c25b,0x59f111f1,0x923f82a4,0xab1c5ed5,
  0xd807aa98,0x12835b01,0x243185be,0x550c7dc3,0x72be5d74,0x80deb1fe,0x9bdc06a7,0xc19bf174,
  0xe49b69c1,0xefbe4786,0x0fc19dc6,0x240ca1cc,0x2de92c6f,0x4a7484aa,0x5cb0a9dc,0x76f988da,
  0x983e5152,0xa831c66d,0xb00327c8,0xbf597fc7,0xc6e00bf3,0xd5a79147,0x06ca6351,0x14292967,
  0x27b70a85,0x2e1b2138,0x4d2c6dfc,0x53380d13,0x650a7354,0x766a0abb,0x81c2c92e,0x92722c85,
  0xa2bfe8a1,0xa81a664b,0xc24b8b70,0xc76c51a3,0xd192e819,0xd6990624,0xf40e3585,0x106aa070,
  0x19a4c116,0x1e376c08,0x2748774c,0x34b0bcb5,0x391c0cb3,0x4ed8aa4a,0x5b9cca4f,0x682e6ff3,
  0x748f82ee,0x78a5636f,0x84c87814,0x8cc70208,0x90befffa,0xa4506ceb,0xbef9a3f7,0xc67178f2
};

#define BUFFER_SIZE 64

uint8_t sha256InitState[] = {
  0x67,0xe6,0x09,0x6a, // H0
  0x85,0xae,0x67,0xbb, // H1
  0x72,0xf3,0x6e,0x3c, // H2
  0x3a,0xf5,0x4f,0xa5, // H3
  0x7f,0x52,0x0e,0x51, // H4
  0x8c,0x68,0x05,0x9b, // H5
  0xab,0xd9,0x83,0x1f, // H6
  0x19,0xcd,0xe0,0x5b  // H7
};

void Sha256Class::init(void) {
  memcpy_P(state.b,sha256InitState,32);
  byteCount = 0;
  bufferOffset = 0;
}

uint32_t Sha256Class::ror32(uint32_t number, uint8_t bits) {
  return ((number << (32-bits)) | (number >> bits));
}

void Sha256Class::hashBlock() {
  uint8_t i;
  uint32_t a,b,c,d,e,f,g,h,t1,t2;

  a=state.w[0];
  b=state.w[1];
  c=state.w[2];
  d=state.w[3];
  e=state.w[4];
  f=state.w[5];
  g=state.w[6];
  h=state.w[7];
  
  for (i=0; i<64; i++) {
    if (i>=16) {
      t1 = buffer.w[i&15] + buffer.w[(i-7)&15];
      t2 = buffer.w[(i-2)&15];
      t1 += ror32(t2,17) ^ ror32(t2,19) ^ (t2>>10);
      t2 = buffer.w[(i-15)&15];
      t1 += ror32(t2,7) ^ ror32(t2,18) ^ (t2>>3);
      buffer.w[i&15] = t1;
    }
    t1 = h;
    t1 += ror32(e,6) ^ ror32(e,11) ^ ror32(e,25); // ∑1(e)
    t1 += g ^ (e & (g ^ f)); // Ch(e,f,g)
    t1 += pgm_read_dword(sha256K+i); // Ki
    t1 += buffer.w[i&15]; // Wi
    t2 = ror32(a,2) ^ ror32(a,13) ^ ror32(a,22); // ∑0(a)
    t2 += ((b & c) | (a & (b | c))); // Maj(a,b,c)
    h=g; g=f; f=e; e=d+t1; d=c; c=b; b=a; a=t1+t2;
  }
  state.w[0] += a;
  state.w[1] += b;
  state.w[2] += c;
  state.w[3] += d;
  state.w[4] += e;
  state.w[5] += f;
  state.w[6] += g;
  state.w[7] += h;
}

void Sha256Class::addUncounted(uint8_t data) {
  buffer.b[bufferOffset ^ 3] = data;
  bufferOffset++;
  if (bufferOffset == BUFFER_SIZE) {
    hashBlock();
    bufferOffset = 0;
  }
}

size_t Sha256Class::write(uint8_t data) {
  ++byteCount;
  addUncounted(data);
}

void Sha256Class::pad() {
  // Implement SHA-256 padding (fips180-2 §5.1.1)

  // Pad with 0x80 followed by 0x00 until the end of the block
  addUncounted(0x80);
  while (bufferOffset != 56) addUncounted(0x00);

  // Append length in the last 8 bytes
  addUncounted(0); // We're only using 32 bit lengths
  addUncounted(0); // But SHA-1 supports 64 bit lengths
  addUncounted(0); // So zero pad the top bits
  addUncounted(byteCount >> 29); // Shifting to multiply by 8
  addUncounted(byteCount >> 21); // as SHA-1 supports bitstreams as well as
  addUncounted(byteCount >> 13); // byte.
  addUncounted(byteCount >> 5);
  addUncounted(byteCount << 3);
}

uint8_t* Sha256Class::result(void) {
  // Pad to complete the last block
  pad();
  
  // Swap byte order back
  for (int i=0; i<8; i++) {
    uint32_t a,b;
    a=state.w[i];
    b=a<<24;
    b|=(a<<8) & 0x00ff0000;
    b|=(a>>8) & 0x0000ff00;
    b|=a>>24;
    state.w[i]=b;
  }
  
  // Return pointer to hash (20 characters)
  return state.b;
}

#define HMAC_IPAD 0x36
#define HMAC_OPAD 0x5c

uint8_t keyBuffer[BLOCK_LENGTH]; // K0 in FIPS-198a
uint8_t innerHash[HASH_LENGTH];

void Sha256Class::initHmac(const uint8_t* key, int keyLength) {
  uint8_t i;
  memset(keyBuffer,0,BLOCK_LENGTH);
  if (keyLength > BLOCK_LENGTH) {
    // Hash long keys
    init();
    for (;keyLength--;) write(*key++);
    memcpy(keyBuffer,result(),HASH_LENGTH);
  } else {
    // Block length keys are used as is
    memcpy(keyBuffer,key,keyLength);
  }
  // Start inner hash
  init();
  for (i=0; i<BLOCK_LENGTH; i++) {
    write(keyBuffer[i] ^ HMAC_IPAD);
  }
}

uint8_t* Sha256Class::resultHmac(void) {
  uint8_t i;
  // Complete inner hash
  memcpy(innerHash,result(),HASH_LENGTH);
  // Calculate outer hash
  init();
  for (i=0; i<BLOCK_LENGTH; i++) write(keyBuffer[i] ^ HMAC_OPAD);
  for (i=0; i<HASH_LENGTH; i++) write(innerHash[i]);
  return result();
}
Sha256Class Sha256;
```

earlephilhower, 2018-09-10:
Closing since it appears unrelated to anything in the core.  The axtls and bearssl SHA256 functions are required for pretty much any SSL connection, and are tested as part of their respective test suites.

This is better addressed at https://www.esp8266.com where general programming questions can be discussed.

closed 2018-09-10 by earlephilhower

adrionics, 2018-11-14:
@kherNI did you ever solve this? I am having the same problem.

kherNI, 2018-11-14:
No I did not.  I ended up having to revert back to 2.4.1.

adrionics, 2018-11-15:
@kherNI  Here is the answer!!!
In the following file: 
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Print.cpp
Line 51, comment out "break".

Previous versions of that function were not doing the check for "Write of last byte didn't complete, abort additional processing"

At least that fixed it for me.

devyte, 2018-11-15:
@adrionics can you provide an MCVE sketch that shows the original issue, and where your fix works?

adrionics, 2018-11-15:
@devyte the problem occurs with the Microsoft Azure example as posted in the OP here.

These lines:
  Sha256.print(stringToSign);
  char* sign = (char*) Sha256.resultHmac();

With the Print.cpp line 51 including the break statement to interrupt processing, outputting the "sign" array to serial shows some cryptic chars. With the break removed, it outputs no chars.

The problem with providing a complete sketch is that it needs an Azure account to work properly.
We may be able to provide a complete example in time.

The question would be, why did Print.cpp get the change that includes "Write of last byte didn't complete, abort additional processing" and the line 51 beak statement?

kherNI, 2018-11-20:
@adrionics you're brilliant, thanks!!!  Commenting out line 51 worked.

devyte, 2018-11-20:
The break change checks that the previous write operation actually succeeded. If it didn't, it errors out. Removing it equates to removing an error check that is telling you something is wrong, and then being glad that code execution continues.
I suspect an off-by-one error somewhere (e.g. strlen() instead of strlen()+1, in other words account for a nullterm in a buffer declaration), or a missing nullterm in a char string (print out of mysterious chars is indicative of a missing nullterm). 
I still think this is user error.

adrionics, 2019-02-23:
Update:
I found a library that is a direct replacement.
https://github.com/daknuett/cryptosuite2

The two things I needed to do:
1) Add "#define SHA1_DISABLED" to the config.h file in the library.
2) Define the hash length in your sketch: "#define HASH_LENGTH 32"
Then, the same code in the Azure example works.

I need to see if there are any other problems or memory leaks, but it does work so far without the modification to the "Print.cpp" file.

I hope this is useful!