esp8266/Arduino #5062: enhancement request:  would it be possible to integrate  ESP8266WebServer and WiFiServer into each other?

dsyleixa, 2018-08-19:
general enhancement request:

would it be possible to integrate  ESP8266WebServer and WiFiServer into each other?
so that one had  ESP8266WebServer.arg("uname")  for WiFiServer and WiFiServer.available() for ESP8266WebServer and so on?
Currently I am sort of 'overwhelmed' and 'overburdened' by having to use both libs side to side in my programs to communicate both with a website and with remote clients:

WiFiServer   wifiserver(80);
ESP8266WebServer lanserver(8081);

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12E]
- Core Version:			[Board Manager 2.4.2]
- Development Env:		[Arduino IDE|]
- Operating System:		[Windows|

### Settings in IDE

- Module:  			[Nodemcu 1.0]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mh
- Upload Using:		SERIAL]
- Upload Speed:   [115200|] (serial upload only)

### Problem Description

dsyleixa, 2018-08-19:
re: "WiFiWebServer is built on top of it to implement the specific hypertext protocol."
is   WiFiWebServer  an optional  substitute for ESP8266WebServer ?

devyte, 2018-08-19:
I meant ESP8266WebServer.

dsyleixa, 2018-08-20:
ah, ok.
But when ESP8266WebServer is built on top of WiFiServer, why is there no ESP8266WebServer.available()?

d-a-v, 2018-08-20:
`WiFiClient` instance is in `ESP8266WebServer.client()`: `ESP8266WebServer.client().available()`

dsyleixa, 2018-08-20:
I see, thanks for your input. I'll try to drop WiFiServer then, if possible.