esp8266/Arduino #5063: 2.4.2 - getCoreVersion

laercionit, 2018-08-20:
Unfortunately I do not know when this function started to give a problem.
I am using the latest core available 2.4.2

Serial.println(ESP.getCoreVersion());

result:
00000000

5chufti, 2018-08-20:
you are sure you are on a release version?
check in packages\esp8266\hardware\esp8266_242\2.4.2\cores\core_version.h for valid entries
and check ESP.getFullVersion() also

laercionit, 2018-08-21:
On core_version.h
#define ARDUINO_ESP8266_GIT_VER 0x00000000
#define ARDUINO_ESP8266_GIT_DESC unspecified

// ARDUINO_ESP8266_RELEASE is defined for released versions as a string containing the version name, i.e. "2_3_0_RC1"
// ARDUINO_ESP8266_RELEASE is used in the core internally. Please use ESP.getCoreVersion() function instead.
// ARDUINO_ESP8266_RELEASE_<version number> are defined for releases, for use in #ifdef... constructs

laercionit, 2018-08-21:
Using this version and zip file

[image]

5chufti, 2018-08-22:
AFAIK the vesion only gets set in the release package, that is the one two links above yours and to be installed via the board manager in arduino ide. For other installations using the source archive one has to care for the core_version.h by oneself (e.g. copy it from said board manager archiive) as this is only a snapshot of the git which does not hold any version info.

laercionit, 2018-08-22:
Got it. So I just need to copy the core_version file from the first link
and put it in its hardware directory?

5chufti, 2018-08-22:
should work

laercionit, 2018-08-22:
Even replacing the entire CORE directory did not work.

In my IDE arduino I maintain at least the last two stable versions and the master, so I can map and perform impact tests of the evolution of the core in my program.

Unfortunately I can not put multiple versions of CORE using the Arduino IDE's automatic installation.

For this reason I use source and I organize the Core.

[image]

laercionit, 2018-08-22:
Would you have any guidance for this kind of use, how could I keep GetCore running and have multiple versions of Core for testing and compilation?

5chufti, 2018-08-23:
just tried modifying core_version.h in my setup, worked as expected.
Seems your setup is corrupted, sure you modified in right directory?
so no esp8266 arduino core issue ... please close!

laercionit, 2018-08-23:
Hello .. Good evening

Unfortunately I did not have the same result.

I am using Source Code from Version 2.4.2
I made the replacement of the file Core_Version.h, I always receive 000000 when requesting the Core Version

I have done all the installation of my IDE, the project compiles perfectly, everything working, but I did not get the same result as you did when I changed the file core_version.h

https://github.com/esp8266/Arduino/archive/2.4.2.zip

laercionit, 2018-08-23:
I am using the hardware tree to install the colors, I think this is the only difference of our configuration. Is that it?

[image]

laercionit, 2018-08-23:
It manages to install by portable mode and ran GetCore. Thank you.

Unfortunately I can not keep two versions of Core in Esp8266 like this in your directory.

I tried putting 2.4.1 and 2.4.2 and the boards are not listed in boars to select.

Can you guide me on how to keep two colors in the same IDE?

[image]

closed 2018-08-24 by laercionit