esp8266/Arduino #5067: kind request for a webserver example with credential

dsyleixa, 2018-08-21:
### Problem Description

hi,
according to this thread and d-a-v 's comment
https://github.com/esp8266/Arduino/issues/5021#issuecomment-414652494
I kindly ask  for a webserver example with credential:
The user is requested to enter a  username and his password and press [Enter] or a [submit button]
and then is allowed to proceed, e.g. by being automatically forwarded and by loading the "authorized" website.
A simple sketch (which does not work flawlessly yet though) is referenced here:
https://github.com/esp8266/Arduino/issues/5021#issuecomment-411997320
https://github.com/esp8266/Arduino/issues/5021#issuecomment-414604258

any help is highly appreciated!

Thanks in advance!

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12E]
- Core Version:			[Board Manager 2.4.2]
- Development Env:		[Arduino IDE|]
- Operating System:		[Windows|

### Settings in IDE

- Module:  			[Nodemcu 1.0]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mh
- Upload Using:		SERIAL]
- Upload Speed:   [115200|] (serial upload only)

dsyleixa, 2018-10-24:
withernow be some examples for login/password (perhaps pull method, not get)?

dsyleixa, 2018-12-19:
will there now be some examples available for login/password (perhaps pull method, not get)?

d-a-v, 2018-12-19:
Try to use 2.5.0-beta2 release with this example made for you:

`File > examples > ESP8266WebServer > HttpAdvancedAuth`

dsyleixa, 2018-12-20:
yes, I'll install and test ASAP! thank you very much!

ctvarlan, 2018-12-20:
Take a look here, an example at the bottom of the page:
https://tttapa.github.io/ESP8266/Chap10%20-%20Simple%20Web%20Server.html

[image: CT Varlan on about.me]

CT Varlan
about.me/c.t.varlan
  <http://about.me/c.t.varlan>

dsyleixa, 2018-12-21:
@ctvarlan:
do you mean this part?
```
void handleLogin() {                         // If a POST request is made to URI /login
  if( ! server.hasArg("username") || ! server.hasArg("password") 
      || server.arg("username") == NULL || server.arg("password") == NULL) { // If the POST request doesn't have username and password data
    server.send(400, "text/plain", "400: Invalid Request");         // The request is invalid, so send HTTP status 400
    return;
  }
  if(server.arg("username") == "John Doe" && server.arg("password") == "password123") { // If both the username and the password are correct
    server.send(200, "text/html", "<h1>Welcome, " + server.arg("username") + "!</h1><p>Login successful</p>");
  } else {                                                                              // Username and password don't match
    server.send(401, "text/plain", "401: Unauthorized");
  }
}
```

tbh, I am a bit confused how to integrate that into my own code design:

```
void loop() {
  // ...
  if (!authorized) {
     handleNotAuthorized();
     delay(100);    
   }
   if (authorized) {
     handleWebsite();
     delay(10);
   }
  // ...
}
```

```
void handleNotAuthorized() {
  String readString = "";
  char   strinput[MAXLEN], strupwd[TOKLEN], struname[TOKLEN] ;
  
  WiFiClient client = wifiserver.available();

  //---------------------------------------

  strcpy(strinput, "");
  strcpy(strupwd, "");
  strcpy(struname, "");

  while ( client.connected() ) {
    if (authorized) return;

    if ( client.available() ) {
      char c = client.read();

      //read char by request
      readString = "";
      while ( (readString.length() < TOKLEN) && (c != '\n') ) {
          readString += c;
          c = client.read();
      }

    readString.toCharArray(strinput, MAXLEN);
    // cstringarg( char* haystack, char* vname, char* sarg )
    // haystack pattern: &varname=1234abc,  delimiters &, \n, \0, SPACE 
    cstringarg(strinput, "uname", struname);  // uname
    cstringarg(strinput, "upwd", strupwd);   // upwd

    // debug
    Serial.print("strupwd     >>>"); Serial.print(strupwd); Serial.println("<<<");
    Serial.print("website_upwd>>>"); Serial.print(website_upwd); Serial.println("<<<");
    Serial.print("readString>>>");Serial.println(readString);

    if ( (strlen(strupwd) == strlen(website_upwd)) && (strcmp(website_upwd, strupwd) == 0) )
    {
      authorized = true;
      //debug
      //Serial.print("check: authorized="); Serial.println(authorized);
      readString = "";
      return;
    }   
      
      //if HTTP request has ended
      if (c == '\n') {
         client.flush();  // <-- useless here, this empties the output buffer, not the input 

        //now output html data header
        String script = "";

        script += ("HTTP/1.1 401 Log-In Required");
        script += ("Content-Type: text/html \n");
        script += ("\n");  //  do not forget this one //????
        script += ("<!DOCTYPE html> \n");
        script += ("<html> \n");
        script += ("<head> \n");

        // utf-8 für "°" Zeichen
        script +=  "<meta http-equiv=\"Content-Type\" content=\"text/html; charset=utf-8\"> \n" ;
        script +=  "<title>" ;
        script +=  website_title ;
        script +=  "</title> \n" ;
        script +=  "</head> \n" ;
        script +=  "<body> \n" ;

        script += "<h1><p style=\"color:rgb(255,0,191);\"> " + (String)website_url ;
        script += (String)": &nbsp; <wbr> <wbr> " + "Not authorized !</p> </h1> \n" ;
        script += ("<h2><p style=\"color:rgb(255,0,191);\"> log in to proceed: </p> </h2> \n");

        script += ("<FORM ACTION='/' method=GET > \n");
        script += ("<h2>user name:  <INPUT TYPE=text NAME='uname' VALUE=''  MAXLENGTH='50'> </h2> \n");
        script += ("<h2>password :  <INPUT TYPE=PASSWORD NAME='upwd' VALUE='' MAXLENGTH='50'> </h2> \n");

        script += ("<h2><INPUT TYPE=SUBMIT></h2> \n");

        script += ("</FORM> \n");
        script += ("<BR> \n");
        script += ("</body> \n");
        script += ("</html> \n");

        client.print(script);

        //stopping client
        client.stop();
        delay(1);
      }
    }
    delay(1);
  }
}

```
  lanserver.handleClient();

dsyleixa, 2018-12-21:
> 
> Try to use 2.5.0-beta2 release with this example made for you: `File > examples > ESP8266WebServer > HttpAdvancedAuth`

@d-a-v :
hello,
thanks again, I tested your example, but I'm afraid it does not work as expected:
I wanted to design my own website to log in with username and password, not just a user dialog
(please see post above)
kind regards!

earlephilhower, 2019-01-31:
Closing this as a "How do I...?" question.  Your best bet for something like this is a forum such as https://www.esp8266.com where people can help you with your app's design.  Nothing for the core to do as we already support HTTP authentication (the linked examples).

closed 2019-01-31 by earlephilhower

dsyleixa, 2019-02-01:
> 
> 
> Closing this as a "How do I...?" question. Your best bet for something like this is a forum such as https://www.esp8266.com where people can help you with your app's design. Nothing for the core to do as we already support HTTP authentication (the linked examples).

it was the proposal of @d-a-v   to post a request here about a html example code for the core library  to be used inside the Arduino IDE!
So why do you close it now, it is not resolved!

https://github.com/esp8266/Arduino/issues/5067#issuecomment-448711922
>  
d-a-v commented on 19 Dec 2018

Please post a new request issue asking for a webserver example with credential, maybe someone (not excluding me) will take time to do it well for the core's examples.
> --
> 
>