esp8266/Arduino #5068: Problems to compile

nodaro, 2018-08-21:
Hello.

I´m trying to develop a program to send messages by telegram, I read this web http://wiki.intrusos.info/doku.php/electronica:esp8266:telegram because I´m spanish.

I´m using online editor, and I have imported this library in the IDE, but when I try to compile the exaple of the web I recibed this error:

`In file included from /tmp/525398441/custom/ESP8266WiFi/src/ESP8266WiFiType.h:26:0,

from /tmp/525398441/custom/ESP8266WiFi/src/ESP8266WiFi.h:33,

from /tmp/525398441/telegram_humedad/telegram_humedad.ino:5:

/home/ubuntu/opt/libraries/latest/freertos_10_0_0_11/src/queue.h:34:3: error: #error "include Arduino_FreeRTOS.h" must appear in source files before "include queue.h"

#error "include Arduino_FreeRTOS.h" must appear in source files before "include queue.h"

^

In file included from /tmp/525398441/custom/ESP8266WiFi/src/ESP8266WiFiSTA.h:28:0,

from /tmp/525398441/custom/ESP8266WiFi/src/ESP8266WiFi.h:34,

from /tmp/525398441/telegram_humedad/telegram_humedad.ino:5:

/tmp/525398441/custom/ESP8266WiFi/src/ESP8266WiFiGeneric.h:27:22: fatal error: functional: No such file or directory

#include <functional>

^

compilation terminated.

exit status 1`

Can someone help me?

Thanks.

per1234, 2018-08-22:
The Arduino Web Editor doesn't even support 3rd party boards like the ESP8266 so this is absolutely the wrong place to post this.

Please do this:
1. Close this invalid issue.
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post all the information we'll need to help you.

nodaro, 2018-08-23:
Sorry for the inconveniences