esp8266/Arduino #5070: exit status 1 error compiling for board nodeMCU 1.0 (ESP 12E Module)

ashaha, 2018-08-22:
----------------------------- Delete below -----------------------------

Above is the error I've been receiving since a long time. I've tried all the solutions mentioned on previous github conversations but I'm yet to resolve the issue. 
============
- Hardware:					[ESP8266 device]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		
- Flash Frequency:	
- CPU Frequency:		[80Mhz]
- Upload Using:		
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I've been trying to execute the below code that I got from the Internet. The error shown is consistent for even a simple program written by me. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

/*
Geekstips.com
IoT project - Communication between two ESP8266 - Talk with Each Other
ESP8266 Arduino code example
*/
#include <DHT.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#define DHTPIN 2

// AP Wi-Fi credentials
const char* ssid = "shah's";
const char* password = "mitesh@123";

// Local ESP web-server address
String serverHost = "http://192.168.4.1/feed";
String data;
// DEEP_SLEEP Timeout interval
int sleepInterval = 5;
// DEEP_SLEEP Timeout interval when connecting to AP fails
int failConnectRetryInterval = 2;
int counter = 0;

float h;
float t;
// Static network configuration
IPAddress ip(192, 168, 4, 4);
IPAddress gateway(192, 168, 4, 1);
IPAddress subnet(255, 255, 255, 0);

DHT dht(DHTPIN, DHT22);
WiFiClient client;

void setup() {
  ESP.eraseConfig();
  WiFi.persistent(false);
  Serial.begin(74880);
  Serial.println();
  Serial.println("**************************");
  Serial.println("**************************");
  Serial.println("******** BEGIN ***********");
[... 75 lines cut ...]
Using board 'nodemcuv2' from platform in folder: C:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2
Using core 'esp8266' from platform in folder: C:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2
Detecting libraries used...
"C:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/include" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Virali\AppData\Local\Temp\arduino_build_229843/core" -c -w  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"   -DESP8266 "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\variants\nodemcu" "C:\Users\Virali\AppData\Local\Temp\arduino_build_229843\sketch\sketch_aug22a.ino.cpp" -o "nul"
"C:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/include" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Virali\AppData\Local\Temp\arduino_build_229843/core" -c -w  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"   -DESP8266 "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\variants\nodemcu" "-IC:\Users\Virali\Documents\Arduino\libraries\DHT_sensor_library-1.3.0" "C:\Users\Virali\AppData\Local\Temp\arduino_build_229843\sketch\sketch_aug22a.ino.cpp" -o "nul"
"C:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/include" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Virali\AppData\Local\Temp\arduino_build_229843/core" -c -w  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"   -DESP8266 "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\variants\nodemcu" "-IC:\Users\Virali\Documents\Arduino\libraries\DHT_sensor_library-1.3.0" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src" "C:\Users\Virali\AppData\Local\Temp\arduino_build_229843\sketch\sketch_aug22a.ino.cpp" -o "nul"
"C:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/include" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Virali\AppData\Local\Temp\arduino_build_229843/core" -c -w  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"   -DESP8266 "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\variants\nodemcu" "-IC:\Users\Virali\Documents\Arduino\libraries\DHT_sensor_library-1.3.0" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src" "C:\Users\Virali\AppData\Local\Temp\arduino_build_229843\sketch\sketch_aug22a.ino.cpp" -o "nul"
"C:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/include" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include" "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Virali\AppData\Local\Temp\arduino_build_229843/core" -c -w  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"   -DESP8266 "-IC:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\e [... cut ...]
Using library DHT_sensor_library-1.3.0 at version 1.3.0 in folder: C:\Users\Virali\Documents\Arduino\libraries\DHT_sensor_library-1.3.0 
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi 
Using library ESP8266HTTPClient at version 1.1 in folder: C:\Users\Virali\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

5chufti, 2018-08-22:
1) this is most likely not a core issue but a user fault
2) have you installed the Adafruit Unified Sensor Library (as requested in the [docs](https://github.com/adafruit/DHT-sensor-library))???

devyte, 2018-08-22:
This looks like an installation issue, I.e. missing lib.
Closing due to user error.
Also, I'm reading a whole bunch of DHT22s without issues, although I must admit I'm not using adafruit's libs.

closed 2018-08-22 by devyte

karanjeet1, 2019-01-17:
this error basically arise when the libraries you are using in your code are not compatible.
Basically, when you update for library from previous versions to new version then this error comes.
so its suggested to edit your code according to updated library or downgrade you library.
hope you got my point.

Mugesh-Rajaram, 2019-05-12:
Getting error  while Verify / Compile in the Arduino IDE 
Can anyone help me on solving this issue.

[image]

[image]

[image]

5chufti, 2019-05-12:
screenshots - especially red on black - are a PITA.
but most likely this isn't even a problem of the arduino esp8266 core, so please look for help in a forum like www.esp8266.com

Mugesh-Rajaram, 2019-05-12:
@5chufti  what do you mean by **PITA**

earlephilhower, 2019-05-12:
@Mugesh-Rajaram PITA is an acronym for pain in the butt (which means something is really hard to do).

The red on black text is impossible to read.  You need to paste the text of the error, not an image.

In any case, your problem is stated right at the bottom:  You have the `Arduino shield WiFi` in your libraries, and it's using that instead of the 8266 version.  They're completely different.  Remove your libraries/WiFi directory and then rebuild.

smtechchauhan, 2019-10-02:
Above is the error I've been receiving since a long time. I've tried all the solutions mentioned on previous github conversations but I'm yet to resolve the issue.

my sketch

// The Code is modified and written by Alok Shaw, the creator of the Youtube channel : www.youtube.com/roboshala and runs an educational blog : www.roboshala.com

#include <ESP8266WiFi.h>
#include<FirebaseArduino.h>
#define FIREBASE_HOST "test-fcb91.firebaseio.com"                     //Your Firebase Project URL goes here without "http:" , "\" and "/"
#define FIREBASE_AUTH "yoezZbKI1mZmcIlAAsMRC2Mdf4tinEQ5k02uQl"       //Your Firebase Database Secret goes here
#define WIFI_SSID "Roboshala"                                               //your WiFi SSID for which yout NodeMCU connects
#define WIFI_PASSWORD "roboindia@alok"                                      //Password of your wifi network 

#define Relay1 12 //D6
int val1;

#define Relay2 4 //D2
int val2;

#define Relay3 5  //D1
int val3;

#define Relay4 14 //D5
int val4;

void setup() 
{
  Serial.begin(115200);                                                   // Select the same baud rate if you want to see the datas on Serial Monitor
  pinMode(Relay1,OUTPUT);
  pinMode(Relay2,OUTPUT);
  pinMode(Relay3,OUTPUT);
  pinMode(Relay4,OUTPUT);

  digitalWrite(Relay1,HIGH);
  digitalWrite(Relay2,HIGH);
  digitalWrite(Relay3,HIGH);
  digitalWrite(Relay4,HIGH);

  
  WiFi.begin(WIFI_SSID,WIFI_PASSWORD);
  Serial.print("connecting");
  while (WiFi.status()!=WL_CONNECTED){
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("connected:");
  Serial.println(WiFi.localIP());

  Firebase.begin(FIREBASE_HOST,FIREBASE_AUTH);
  Firebase.setInt("S1",0);                     //Here the varialbe"S1","S2","S3" and "S4" needs to be the one which is used in our Firebase and MIT App Inventor
  Firebase.setInt("S2",0); 
  Firebase.setInt("S3",0); 
  Firebase.setInt("S4",0); 
  
}
void firebasereconnect()
{
  Serial.println("Trying to reconnect");
    Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  }

void loop() 
{
  if (Firebase.failed())
      {
      Serial.print("setting number failed:");
      Serial.println(Firebase.error());
      firebasereconnect();
      return;
[... 70 lines cut ...]

In file included from C:\Users\Ersumit\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseArduino.h:22:0,

                 from C:\Users\Ersumit\AppData\Local\Temp\Rar$DI02.047\Firebase_HomeAutomation\Firebase_HomeAutomation.ino:4:

C:\Users\Ersumit\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:86:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

   std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

           ^

Multiple libraries were found for "FirebaseArduino.h"
 Used: C:\Users\Ersumit\Documents\Arduino\libraries\firebase-arduino-master
Multiple libraries were found for "ArduinoJson.h"
 Used: C:\Users\Ersumit\Documents\Arduino\libraries\ArduinoJson-6.x
Multiple libraries were found for "ESP8266HTTPClient.h"
 Used: C:\Users\Ersumit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266HTTPClient
Multiple libraries were found for "ESP8266WiFi.h"
 Used: C:\Users\Ersumit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Davithronix, 2021-04-13:
I'm also experiencing the same compilling error on my code for NODEMCU-12E. Pls any help from anyone?

Davithronix, 2021-04-13:
I'm just getting started with NODEMCU Wifi module and have been having issue on compiling the code. "Exit status 1 error compiling for oard nodeMCU 1.0" is the error message given.
The below is my code

#define ledPin1 2 /* LED connected to GPIO 2 */
#define ledPin2 16 /* LED connected to GPIO 16 */
 
void setup() {
  // put your setup code here, to run once:
  pinMode(ledPin1, OUTPUT);
  pinMode(ledPin2, OUTPUT);

}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(ledPin1, LOW);
  digitalWrite(ledPin2, HIGH);
  delay(1000);
  digitalWrite(ledPin1, HIGH);
  digitalWrite(ledPin2, LOW);
  delay(1000);

}

5chufti, 2021-04-13:
Hi,
a) pls complete the comment indicators at the end of the #define lines. 
as it is now, the 2nd define is commented out and not active (everything between a starting /* and closing */  is treated as comment (even over mutiple lines). to just have a comment regarding the line you can use it like
#define LED D3  // my used gpio

b) activate the "show verbose output" under files/preferences for compile and upload to get helpful indication on where reported errors appear