esp8266/Arduino #5071: calling ESP8266WiFiMesh::deactivateAP() doesn't fully disable wifi and corrupts SSID

rorosaurus, 2018-08-22:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ x] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

Calling `ESP8266WiFiMesh::deactivateAP()` in the example HelloMesh program produces a Wifi network with this result, after refreshing wifi on my phone:

[image]

I would have expected the wifi network to be shut off completely, not remove password, SSID, etc.

rorosaurus, 2018-08-22:
Maybe this is expected behavior?  But I didn't expect to need to call `WiFi.mode(WIFI_OFF);` after calling `ESP8266WiFiMesh::deactivateAP()`

devyte, 2018-08-22:
CC @aerlon

aerlon, 2018-08-24:
Interesting. Neither my phone, nor my laptop can detect the WiFi once `ESP8266WiFiMesh::deactivateAP()` is called, but other ESP8266 nodes are able to do it. It seems as though an AP with an empty SSID is created. The intended result is to completely shut down the AP, as you guessed, so this is a bug.

This problem is caused by the use of `WiFi.mode(WIFI_AP_STA);` as the default WiFi mode in the library. It seems that the node will automatically set up an AP in this WiFi mode, even though `WiFi.softAP()` is never called. You can try this yourself by adding `delay(30000);` after `meshNode.begin();` in HelloMesh, an AP will be created even though we have never called `activateAP()`. 

@devyte Is it expected behaviour that just having the ESP8266 enter `WiFi.mode(WIFI_AP_STA);` will cause an AP to be created (even if we never call `WiFi.softAP()`), or is this a bug in the core WiFi code?

If this is not a bug in the core code, the default WiFi mode of the mesh library will have to change. This should be possible, but I will need to do some testing.

@rorosaurus `ESP8266WiFiMesh::deactivateAP()` requires a slight change to fix the bug you found. Changing `WiFi.softAPdisconnect();` to `WiFi.softAPdisconnect(true);` in the method seems to fix it, at least till the node re-enters the `WIFI_AP_STA` mode. We cannot use `WiFi.mode(WIFI_OFF);` in `ESP8266WiFiMesh::deactivateAP()` because deactivating the node AP would then also disconnect any active WiFi station the node may have. (you can of course use `WiFi.mode(WIFI_OFF);` as a work around for now, I'll include the fix for this bug in my intermediate release which should be ready within a week)

devyte, 2018-08-25:
My guess is that of you change mode to ap_sta, then an ap will be brought up with whatever parameters possible. I would expect some default if nothing has been set up, but I'm just guessing.
My own policy is to always specify every parameter myself, and not rely on sdk defaults or whatever. That means that I do something along the lines of:
If newmode is sta or sta_ap
  Set up sta with my parameters
If newmode is ap or sta_ap
  Set up ap with my parameters
Change mode to newmode

(Sorry for lousy pseudo code, I'm travelling with just my phone).

aerlon, 2018-08-25:
Yes, after reviewing the core source code it seems as though `WiFi.mode()` is always in charge of actually enabling and disabling the AP. All other WiFi methods dealing with this functionality just wrap the `WiFi.mode()` method in different ways. I'll modify the mesh library to take this into account.

aerlon, 2018-09-25:
I should probably add that `uint64ToString` and `stringToUint64` are now stand-alone functions, so if your code used the old class methods you will probably need to `#include <TypeConversionFunctions.h>` and change the function call signature a bit. The updated HelloMesh example can be used as a reference for this.

devyte, 2019-11-10:
Reference: #6280

devyte, 2020-07-07:
Still not done with this, pushing back to 3.

d-a-v, 2021-03-31:
Does merged #6280 address this issue ?

aerlon, 2021-04-10:
Mesh version 2.2 should address everything mentioned here. The issue can be closed.

closed 2021-04-10 by d-a-v