esp8266/Arduino #5072: Illegal instruction using Wire library

erpalma, 2018-08-22:
#### Platform

- Hardware:			ESP-12
- Core Version:			git master 
- Development Env:		Arduino IDE
- Operating System:		Linux

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio
- Flash Size: 		        4MB
- lwip Variant:                   v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	        40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

After some minutes I get illegal instruction exception and wdt reset while reading the values out of a BME280 sensor connected in I2C using Wire library. 

### Debug Messages

```
Exception 0: Illegal instruction
Decoding 11 results
0x4020a2e2: esp_yield at /opt/arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 89
0x4020a2e2: esp_yield at /opt/arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 89
0x4020a3e5: optimistic_yield at /opt/arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 115
0x40207625: TwoWire::available() at /opt/arduino/hardware/esp8266com/esp8266/libraries/Wire/Wire.cpp line 259
0x40206eb9: BME280::readRegisterRegion(unsigned char*, unsigned char, unsigned char) at /home/user/Arduino/libraries/SparkFun_BME280/src/SparkFunBME280.cpp line 551
0x4020894d: Print::write(char const*) at /opt/arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 122
0x4020a305: esp_schedule at /opt/arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 95
0x4020a39f: __yield at /opt/arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 100
0x40204837: loop at /home/user/Arduino/esptest2/esptest2.ino line 1230
0x4020a370: loop_wrapper at /opt/arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 125
0x40100a49: cont_wrapper at /opt/arduino/hardware/esp8266com/esp8266/cores/esp8266/cont.S line 81
```

RudyFiero, 2018-08-23:
Obviously the problem is with loop. It disappears.

5chufti, 2018-08-23:
where is your MCVE sketch?
how are we supposed to find an error in your soft w/o seeing it?

erpalma, 2018-08-30:
I just want to add that in the end the issue was certainly caused by a faulty (fake?) batch of ESP-12F that I was using. I moved the code to an older ESP-12E and it worked flawlessly.