esp8266/Arduino #5073: Sketch Fails to Load via USB using Linux

keypunch416, 2018-08-23:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git) with 2.4.2  (2.4.1, 2.3.0 have failed in past)
- [X] I have searched the issue tracker for a similar issue and the web at at large
- [X] If there is a stack dump, I have decoded it. Does not apply as code will not load.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					ESP8285
- Core Version:			[20180803] 2.4.2
- Development Env:		[Arduino IDE 1.8.5]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio|dio|other] Not Sure what this is or how to find, using USB
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Crystal Frequency:	[26Mhz]
- Flash Frequency:	[unknown]
- CPU Frequency:		[80Mhz]
- Upload Using:		[USB]
- Upload Speed:   [115200)

### Problem Description

Using USB to compile/upload Examples>01.Basics>Blink sketch to an ESP8285 using "Board: Generic ESP8285 Module".  Sketch upload via USB fails on what I believe is the "serialport_receive_C0: 00 instead of C0" part of upload detailed messages.  Places upload detailed messages in the Debug Messages section.  This issue has existed with ESP8266 core for Arduino V2.3.0, V2.4.0, V2.4.1 and now with V2.4.2.

Nano, Uno, Leonardo, Due, STM32duino (Maple Mini STM32F103), ESP8266 all work fine via USB using Arduino IDE 1.8.5 (and in past 1.8.2, 1.8.3, 1,8,4).

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/Blink
*/

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
[... 688 lines cut ...]
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/MD5Builder.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/Print.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/Schedule.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/ScheduledFunctions.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/Stream.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/StreamString.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/Tone.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/Updater.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/WMath.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/WString.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/abi.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/base64.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/cbuf.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/core_esp8266_main.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/debug.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/pgmspace.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/spiffs_api.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/spiffs_hal.cpp.o"
Archiving built core (caching) in: /tmp/arduino_cache_174675/core/core_esp8266_esp8266_esp8285_CpuFrequency_80,VTable_flash,ResetMethod_ck,CrystalFreq_26,FlashSize_1M0,led_2,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_115200_1d463750e0aa13611cccc7c1d4c959f4.a
Linking everything together...
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -CC -E -P -DVTABLES_IN_FLASH "/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/ld/eagle.app.v6.common.ld.h" -o "/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/ld/eagle.app.v6.common.ld"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -Wl,-Map "-Wl,/tmp/arduino_build_282627/Blink.ino.map" -g -Wall -Wextra -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static "-L/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/lib" "-L/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/ld" "-L/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/libc/xtensa-lx106-elf/lib" "-Teagle.flash.1m0.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read  -o "/tmp/arduino_build_282627/Blink.ino.elf" -Wl,--start-group "/tmp/arduino_build_282627/sketch/Blink.ino.cpp.o" "/tmp/arduino_build_282627/arduino.ar" -lhal -lphy -lpp -lnet80211 -llwip2 -lwpa -lcrypto -lmain -lwps -lbearssl -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group  "-L/tmp/arduino_build_282627"
"/home/lubuntu/.arduino15/packages/esp8266/tools/esptool/0.4.13/esptool" -eo "/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/bootloaders/eboot/eboot.elf" -bo "/tmp/arduino_build_282627/Blink.ino.bin" -bm dout -bf 40 -bz 1M -bs .text -bp 4096 -ec -eo "/tmp/arduino_build_282627/Blink.ino.elf" -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -ec
Sketch uses 247856 bytes (24%) of program storage space. Maximum is 1023984 bytes.
Global variables use 27984 bytes (34%) of dynamic memory, leaving 53936 bytes for local variables. Maximum is 81920 bytes.

keypunch416, 2018-08-23:
Atempt using Module: [Generic ESP8285 Module] and set "Flash Size: 1M (No SPIFFS)" and same still open Examples>01.Basics>Blink sketch with the ESP8285 board and ESP8266 core for Arduino V2.4.2.

[image]

Detailed Messages:

/tmp/arduino-1.8.5/arduino-builder -dump-prefs -logger=machine -hardware /tmp/arduino-1.8.5/hardware -hardware /home/lubuntu/.arduino15/packages -tools /tmp/arduino-1.8.5/tools-builder -tools /tmp/arduino-1.8.5/hardware/tools/avr -tools /home/lubuntu/.arduino15/packages -built-in-libraries /tmp/arduino-1.8.5/libraries -libraries /tmp/arduino-1.8.5/Source/libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,VTable=flash,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=qio,FlashSize=1M0,led=2,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200 -ide-version=10805 -build-path /tmp/arduino_build_282627 -warnings=all -build-cache /tmp/arduino_cache_174675 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/home/lubuntu/.arduino15/packages/esp8266/tools/mkspiffs/0.2.0 -prefs=runtime.tools.esptool.path=/home/lubuntu/.arduino15/packages/esp8266/tools/esptool/0.4.13 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -verbose /tmp/arduino-1.8.5/examples/01.Basics/Blink/Blink.ino
/tmp/arduino-1.8.5/arduino-builder -compile -logger=machine -hardware /tmp/arduino-1.8.5/hardware -hardware /home/lubuntu/.arduino15/packages -tools /tmp/arduino-1.8.5/tools-builder -tools /tmp/arduino-1.8.5/hardware/tools/avr -tools /home/lubuntu/.arduino15/packages -built-in-libraries /tmp/arduino-1.8.5/libraries -libraries /tmp/arduino-1.8.5/Source/libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,VTable=flash,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=qio,FlashSize=1M0,led=2,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200 -ide-version=10805 -build-path /tmp/arduino_build_282627 -warnings=all -build-cache /tmp/arduino_cache_174675 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/home/lubuntu/.arduino15/packages/esp8266/tools/mkspiffs/0.2.0 -prefs=runtime.tools.esptool.path=/home/lubuntu/.arduino15/packages/esp8266/tools/esptool/0.4.13 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -verbose /tmp/arduino-1.8.5/examples/01.Basics/Blink/Blink.ino
Using board 'generic' from platform in folder: /home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2
Using core 'esp8266' from platform in folder: /home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2
Build options changed, rebuilding all
Detecting libraries used...
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/include" "-I/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/lwip2/include" "-I/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_282627/core" -c -w  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-I/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266" "-I/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/variants/generic" "/tmp/arduino_build_282627/sketch/Blink.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/include" "-I/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/lwip2/include" "-I/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_282627/core" -c -w  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-I/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266" "-I/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/variants/generic" "/tmp/arduino_build_282627/sketch/Blink.ino.cpp" -o "/tmp/arduino_build_282627/preproc/ctags_target_for_gcc_minus_e.cpp"
"/tmp/arduino-1.8.5/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino_build_282627/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/include" "-I/home/lubuntu/.arduino15/packages/esp [... cut ...]
	espcomm_send_command: sending command payload
	serialport_receive_C0: 00 instead of C0
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 00 instead of C0
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 00 instead of C0
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

keypunch416, 2018-08-23:
#### Platform

- Hardware:					ESP8285
- Core Version:			[20180803] 2.4.2
- Development Env:		[Arduino IDE 1.8.5]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[1MB (no SPIFFS)]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Crystal Frequency:	[26Mhz]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[USB]
- Upload Speed:   [9600]

### Problem Description

Attempt using Module: [Generic ESP8285 Module] and set "Flash Size: 1M (No SPIFFS)" and same still open Examples>01.Basics>Blink sketch with the ESP8285 board and ESP8266 core for Arduino V2.4.2. Sketch upload via USB using 9600 based on further searching and reading.Sketch upload via USB fails on what I believe is the "serialport_receive_C0: 00 instead of C0" part of upload detailed messages.  Placed upload detailed messages in the Debug Messages section.

[image]

### Debug Messages

```

Sketch uses 247856 bytes (24%) of program storage space. Maximum is 1023984 bytes.
Global variables use 27984 bytes (34%) of dynamic memory, leaving 53936 bytes for local variables. Maximum is 81920 bytes.
/home/lubuntu/.arduino15/packages/esp8266/tools/esptool/0.4.13/esptool -vv -cd ck -cb 9600 -cp /dev/ttyUSB0 -ca 0x00000 -cf /tmp/arduino_build_282627/Blink.ino.bin 
esptool v0.4.13 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 9600
	setting port from /dev/ttyUSB0 to /dev/ttyUSB0
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/ttyUSB0 at 9600
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 00 instead of C0
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 00 instead of C0
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 00 instead of C0
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 00 instead of C0
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 00 instead of C0
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
[... 135 lines cut ...]
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/Schedule.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/ScheduledFunctions.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/Stream.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/StreamString.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/Tone.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/Updater.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/WMath.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/WString.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/abi.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/base64.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/cbuf.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/core_esp8266_main.cpp.o"
"/home/lubuntu/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "/tmp/arduino_build_282627/arduino.ar" "/tmp/arduino_build_282627/core/debug.cpp.o"
[... cut ...]
"/home/lubuntu/.arduino15/packages/esp8266/tools/esptool/0.4.13/esptool" -eo "/home/lubuntu/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/bootloaders/eboot/eboot.elf" -bo "/tmp/arduino_build_282627/Blink.ino.bin" -bm qio -bf 40 -bz 1M -bs .text -bp 4096 -ec -eo "/tmp/arduino_build_282627/Blink.ino.elf" -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -ec
Sketch uses 247856 bytes (24%) of program storage space. Maximum is 1023984 bytes.
Global variables use 27984 bytes (34%) of dynamic memory, leaving 53936 bytes for local variables. Maximum is 81920 bytes.

```

5chufti, 2018-08-23:
repeating the same entry over and over again will not speed up anything.
It is most likely driving away people willing to answer.
As a lot of people are using linux with out any problem, it is probably a fault on your side but most likely not in the arduino esp8266 core - thus no "issue".
Did you ever considder the module or cable you are using to be faulty?
The 
> serialport_receive_C0: 00 instead of C0

is an indication of a communication problem, the module boot loader does not answer correctly on the status request of the esp flasher.

For this type of "requests" please use an appropriate user forum like www.esp8266.com and close this issue ...

keypunch416, 2018-08-23:
Too late now here.

Additional testing post your comments using at least three different USB cables indicates the USB cable is not issue.

Tested (Micro) USB cable normally use and two other (Micro) USB cables with Arduino DUE programming port and all work perfect with Due.

Issues of past and current with ESP8266 and same issues with ESP8285 persist.  Additional creative testing indicates interesting results with the ESP8266 Core (V2.4.2) that I do not have time to post the details.

I made the effort to post the details because I cannot assume I know what may, or may not, be a factor in the issue(s).  I do know from my many years of software problem analysis that some problems like this one that I have been trying to figure out for a year on the assumption I was doing something incorrect are not always as the problem initially seems to be caused by.  Key word here "not".  It means I may be doing something incorrect, or not.  That said, the creative tests I have done quickly that will take time to post the results details suggests there may be more to this issue than I was assuming over the past year.  The past year I was assuming I was at fault in some manner.  This means I did not report this issue over the time I noted previously of issue(s) existing in V2.3.0, V2.4.0, V2.4.1, and V2.4.2 to same net effect.

This is far from the first time I have suspected cables and have tried many different USB cables over the past year that all work with STM32 devices, ESP32, and the Due that have a Micro USB as opposed to Mini USB never having a glitch in loading sketches.

Linux is noted as it is the OS these issues were experienced with.  I do not have Windows, nor have had Windows on any machine of mine.  Therefore I cannot do any comparative tests with Windows.

d-a-v, 2018-08-23:
esp8285 are DIO and 1MB (only, when you select generic esp8266 board)
Have you tried another reset method ?

keypunch416, 2018-08-23:
@d-a-v,

This is why as I noted in my entries with the ESP8266 tests I selected "Flash Size: [1MB (no SPIFFS)]".

Did the test using DIO and "Flash Size: [1MB (no SPIFFS)]". Failed, but different messages for failure.  I need to go to sleep soon.  Saved screen shot from Tools and Arduino detailed text output.  Key appears to be different failed results from load via flash of sketch, but still fails.   My creative tests hinted in my last post I think will be interesting.  I will not be able to redo those tests and save those results until later tomorrow or the following day.

@5chufti I would suggest to you that if three USB cables, including one I used with the ESP8285 all work with the Arduino Due then the communications failure is "probability" not the USB cable.  Further based on test I did for d-a-v and the creative tests I did since your first comment it suggests the issue is not likely the USB cables.

5chufti, 2018-08-23:
@keypunch416 I didn't mention cables except for the first reply, seems you need a good nights sleep ;)
still you could keep your posts much more readable by pointing out and only posting the difference that obviously only you can see (apart from different baudrate)

devyte, 2018-08-23:
Comm errors like these are documented for the most part, and their occurrence are usually bad hw (faulty, wrong design, bad power source, wrong reset circuit,  etc), bad cable, bad reset (wrong type selected, done incorrectly, etc), user error (wrong IDE settings,etc), or similar. I don't remember seeing an issue where a comm error turned out to be a bug in the core or its tools.
Closing. Feel free to continue discussing, there's always the chance it really is a bug, but from what I've seen so far I don't think it is.

closed 2018-08-23 by devyte

keypunch416, 2018-08-23:
@devyte

Sorry, but I believe this is a bug based on the quick additional testing I did last night I did not have time to save the results at time with intent of redoing those interesting reveling tests than in combination with testing the thre USB cables with Zero and Due all worked just fine.  Given the ongoing assumptions this is a hardware or USB cable issue that has been already proved not to be case I feel there is no point in my making the effort to prove there is a good possibly of a bug based on the interesting quick test results that can be duplicated 100%.  It appears there is little to no interest in this being considered a actual issue/bug.  Therefore I will not spend my time to demonstrate given assumptions despite the "for most part", "usually" or " turned out to be a bug".

I have worked with software and software engineers as a caree for 45 years.  I know first hand how many times the engineering teams I worked with dismissed a customer bug I spent some time in lab testing to validate the bug the engineers denied as being a bug.  A bug report with automated scripts and/or custom "C"/scripts delivered to engineering with the repeatable test case.  Only after alot of time, at times a few months, engineering had to admit as a bug after where I spent much time and effort explaining as the bug reports were the simple and summarized issue in a package sent to engineering with all the technical details and background to arrive at comfirming the customer issue being a bug in the bug log.  I had the same extended duty cycle with bugs of similar nature where developers claimed the bug did not exist.  In one a few of these cases these bugs actualy brought own the national systems for several hours each day that management claimed was not occurring, but the first line support staff were dealing with the hours od system down each day as result not knowing the why and denying the bug report opened for months that was actually getting much worse with each code drop.  A bug in the end the developers had me assigned me to help them them to fix what actually several bugs all piling up to the one major overall bug.  Be clear I am not patting myself on the back as I had other cuastomer or QA bugs to address rather than do these types of battles.  For sure not all bugs/issues were this challenging technically or to have engineering address, but usually at least a handful a year were this way.  Also many bugs that are reported to or found by the ESP8266 Library many users likely have not encountered and therefore did not know about.  That means just because software engineer has not seen or believes there is a bug that no bug exists.  Given how I know how much time and effort it takes to prove an issue engineers I will not spend time trying to convince there just may be an issue given the additional interesting testing I did last night quickly that clearly needs the supporting documentation and observations to be logged and time I was willing to take to do so that was not considered to be helpful.

Please do not ask me to spend more time on this issue.  Only one person seemed to ask important and useful information that followed up with related test that clearly had very different results as I noted that again there clearly is no interest in learning about by yourself.

keypunch416, 2018-08-24:
Additional interesting tests now confirms issue is bug and not USB cables at all.  Therefor bug is not a communications error despite error messages of issue indicating such.  Will not document this nor the other interesting tests that now clearly prove this is a bug in the library.  It means there are issue(s) in the library and/or tools.

d-a-v, 2018-08-24:
@keypunch416
You need to understand that we have *lots* of issues about communication error through USB. Most (like 99%) of them were not a core bug (missing capacitor, bad usb driver, bad pin configuration, bad software configuration, etc), hence @devyte's answer after this long thread. I'm sure we can reopen the issue now that you are stating this is not a hardware bug and have more information to provide.

Can you tell us exactly which esp8285 board is it ?
You can also try to flash manually by using espressif's official flash tool (esptool.py) and report here.

keypunch416, 2018-08-24:
@d-a-v 

This issue has been closed when I had indicated there was interesting information to still be documented in this issue.  That means there was now interest in the interesing infoormation that clearly indicated the issue was not a communications related matter or the 99% of time isses.  This notwithstanding I had clearly proven the USB cables were not the issue.  I suspect those with the issues I reported in this issue and never stating what had been the issues over the past year that varied in ways that collectively and just on each issue of its own all clearly related to this same issue I opened would never be believed based on just rerporting current state as I did in this issue.  I really do mean what was occuring in prior attempts one could not imagine in their widest dreams.  This of course skips the point I made that I had been working on this for a year to resolve never having reported any of the prior cannot imagine issues with the prior versions of the libraries.

My position remains the same.  I will not put any more effort nor time to documenting this issue/issues as it it is clear it will take alot of my time and effort with the likely outcome of one or more times the issue will be closed again or that the issue is my error or some other cause than the ESP8266 Arduino library/tools.

I have enough information from the additional and interesting testing I have done that has not been documented (was to be until issue was closed) and the follow on testing once the issue was closed prior to my providing the additional information when I have more time to look at the code and see if I can fix the issues as I see the issues.  Should I get that far with the learning curve of the code for the issues I will make changes, validate and test the changes.  If those changes resolve and/or address other secondary issues as I see them I will effect on my system, not on any code respository or online posting simply because even if I do experience has demonstrated, like this issue, the changes I make will be challenged and/or dismissed as not correct or meet the design in some manner.  Far worse has occurred when I create patches for bugs or feature enhancements in past I had contributed multiple times multiple projects, so I no longer do so.

In conclusion this issue is to remain closed.  That is what many wish or believe.  I know from my years of on the job experience what the progression and path this issue would likley take and why.  That is in a situation where that was my job to analyze or predict bugs and I have no interest in the challenges in that progression or path for a role known to the developers, technical staff, and devops from mainframe to embedded on any OS or hardware local to world wide systems internally or by customers that most of time I would never had known or used prior to the issue reported or I would predict bugs for.  That means my career was based mostly on dealing with software issues on systems and software I had never seen or known about prior to the issues reported or being asked to predict for.  So I know what is involved and the time and challenges to make the case of an issue all too well.  Ergo this issue remains closed as I have no interest in the challenges to justify reopening the issue, and I have lost interest in reporting other ESP8266 Library issues I also know about that still exist or new with the 2.4.x releases I have added code when I can to work about the issues new with 2.4.x.

@d-a-v thans for your interest and concerns.

devyte, 2018-08-25:
@keypunch416 I don't have 45 years experience working as a software engineer, but I do have 25 or so. Given that you're a fellow engineer I'm going to take the time to clarify a few things here.
First, the code in this repo is a community effort, and this issue tracker is meant to coordinate efforts by the community to fix reported issues. That is fundamentally a very different system vs. a business driven environment where users report issues and the engineers have to fix them.
You, as the OP of this issue, have the onus of proving that the issue you're having is really in the code in this repo, and is not a usage error or app coding mistake or hw issue. The reason for this is quite simple: we maintain the code in this repo, period. We don't provide support for your particular project, nor do we provide help with usage, or any other type of support. Any other users who think they're up against the same problem that you reported are welcome to step in and help out with figuring things out. 
In other words, it is your choice whether you choose to report something, but if you do, you're expected to do your homework and stick with it. Any developers who find your issue interesting, or who can provide insight, can then do so. If you don't do your homework, don't expect anyone here to spend significant effort on looking at yourthe issue, because in the end, it is YOUR issue.
This is NOT just a reporting tool, I.e. you're not doing anyone a favor by just reporting a problem and then going away expecting some developer to pick it up and solve it for you. There are currently 458 open issues, and only 3 active maintainers (maybe 4 depending on how you count), and we're all volunteers who work on this in our free time. As such, we're stretched beyond thin to get releases out the door with meaningful enhancements and bugfixes, so we have to prioritize. As said above, in pretty much all cases seen with flash serial comm errors it has turned out to be a layer 8 error in one way or another. I've seen hardware mistakes, connection mistakes, wiring mistakes, settings mistakes, driver problems, permission problems, etc etc etc. , so again, from what I have seen reported so far, nothing points to a bug in the core.
This issue affects you, and you're therefore the most interested in getting it fixed. How much time you want to spend on it is up to you. As I said above, you're free to continue discussing with anyone else who is interested. You're also free to post your findings, or not. If you do, and if they point to a bug in this repo, I'll reopen, that has happened several times before. But you have not even seen fit to explain what hardware you're using, as @d-a-v asked, so if you decide to continue here, I suggest starting with that, which should have been part of your original report.
Finally, I'm very aware of the troubles behind incomplete logs, I've had to request complete logs myself, but please use common sense here. This is obviously a flashing issue, so the details of how gcc is called to compile each file of your project is irrelevant and just adds noise to your report. That is what @5chufti was aiming at. The interesting parts are maybe the beginning, plus the end of the build part, plus obviously the flashing part.

d-a-v, 2018-08-27:
FWIW, I have some of them and I am regularly using esptool-ck under linux without any issue.