esp8266/Arduino #5078: HTTP Post throws Exception(3)

rezariantono, 2018-08-25:
#### Platform

- Hardware:					[ESP-8266]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- CPU Frequency:		[80Mhz|
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

ESP throws exception when making HTTP Post request to remote server

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

#include <ESP8266HTTPClient.h>

HTTPClient http;

const String apiEndpoint = "http://192.168.0.12:20050/alarm-system/1105/system-api/alarm/triggered";

void setup() {

    http.begin(apiEndpoint);
    http.addHeader("Content-Type", "application/json");
    http.POST(body);
    http.end();
}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x40100579: realloc at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\umm_malloc\umm_malloc.c line 1532
EXCVADDR: 0x400700d9

Decoding stack results
0x40211ee3: String::reserve(unsigned int) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\WString.cpp line 141
0x40211e97: String::changeBuffer(unsigned int) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\WString.cpp line 151
0x40211ee3: String::reserve(unsigned int) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\WString.cpp line 141
0x40211f15: String::copy(char const*, unsigned int) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\WString.cpp line 170
0x4021201c: String::operator=(String const&) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\WString.cpp line 215
0x4020bc06: HTTPClient::beginInternal(String, char const*) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/WString.h line 190
0x40211e97: String::changeBuffer(unsigned int) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\WString.cpp line 151
0x40211ee3: String::reserve(unsigned int) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\WString.cpp line 141
0x4021201c: String::operator=(String const&) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\WString.cpp line 215
0x4020c65a: HTTPClient::begin(String) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 127
0x40212044: String::String(String const&) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\WString.cpp line 41
0x40208a96: triggerAlarm(Sensor) at D:\Dev\NodeMCU\alarm_system/alarm_system.ino line 249
0x40211f15: String::copy(char const*, unsigned int) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\WString.cpp line 170
0x4021201c: String::operator=(String const&) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\WString.cpp line 215
0x40212044: String::String(String const&) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\WString.cpp line 41
0x40208c45: loopSensorsAndCheckStatus() at D:\Dev\NodeMCU\alarm_system/alarm_system.ino line 207
0x40208f51: loop() at D:\Dev\NodeMCU\alarm_system/alarm_system.ino line 86
0x40212b20: loop_wrapper() at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\core_esp8266_main.cpp line 121

devyte, 2018-08-25:
Your logs show you're using core 2.4.1. A lot of issues have been fixed since thenbeen. Is this issue still valid with core 2.4.2?

rezariantono, 2018-08-25:
Updated to 2.4.2 and the same error still showing

### Debug Messages

```

Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x401005b1: realloc at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc\umm_malloc.c line 1536
EXCVADDR: 0x400700a1

Decoding stack results
0x4010020c: _umm_free at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc\umm_malloc.c line 1295
0x401006a8: realloc at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc\umm_malloc.c line 1645
0x402127f7: String::changeBuffer(unsigned int) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\WString.cpp line 156
0x40212843: String::reserve(unsigned int) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\WString.cpp line 146
0x402127f7: String::changeBuffer(unsigned int) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\WString.cpp line 156
0x40212843: String::reserve(unsigned int) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\WString.cpp line 146
0x40212875: String::copy(char const*, unsigned int) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\WString.cpp line 175
0x4021297c: String::operator=(String const&) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\WString.cpp line 220
0x4020c3ca: HTTPClient::beginInternal(String, char const*) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/WString.h line 190
0x402127f7: String::changeBuffer(unsigned int) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\WString.cpp line 156
0x40212843: String::reserve(unsigned int) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\WString.cpp line 146
0x4021297c: String::operator=(String const&) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\WString.cpp line 220
0x4020cd5e: HTTPClient::begin(String) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 173
0x402129a4: String::String(String const&) at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\WString.cpp line 41
0x40209199: triggerAlarm(Sensor) at D:\Dev\NodeMCU\alarm_system/alarm_system.ino line 248
0x402092dd: loopSensorsAndCheckStatus() at D:\Dev\NodeMCU\alarm_system/alarm_system.ino line 209
0x402095e5: loop() at D:\Dev\NodeMCU\alarm_system/alarm_system.ino line 87
0x40213588: loop_wrapper() at C:\Users\reza1\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_main.cpp line 125

```

For your information, this is my whole code
Im just beginning to learn ESP8266, so im sorry for any mistake 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <string>

#include<ESP8266WiFi.h>
#include<ESP8266WiFiMulti.h>
#include<WiFiClientSecure.h>

#include<TimeLib.h>
#include<NtpClientLib.h>

#include <ESP8266HTTPClient.h>

#include<ESP8266WebServer.h>
#include<ArduinoJson.h>

#include<FS.h>

#include <ArduinoOTA.h>

class  Sensor {

  public:
    int gpio_number;
    String sensor_name;
    String user;
    String default_state;
    int activate_on;
    int alarm_delay;
    int alarm_on;
    bool pull_up;
    bool triggered;
};
[... 338 lines cut ...]
  sensorRes.set("triggered", sensor.triggered);

  if (sensor.user.length() > 1) {
    sensorRes.set("user", sensor.user);
  }

  return sensorRes;
}

void performDdnsUpdateIfRequired() {

  if (lastDdnsUpdate < (now() - (5 * 60) )) {

    lastDdnsUpdate = now();

    http.begin(ddnsUpdateLink);
    http.addHeader("Content-Type", "application/json");
    http.GET();
    http.end();

    Serial.println("DDNS update performed " + String(now()));
  }
}

```

devyte, 2018-08-30:
You have StaticJsonBuffer<2000> jsonBuffer; as a global declaration, and are trying to allocate from it on every loop. JsonBuffer is not meant to be reused, because it can't free memory until destruction.
You need to study the JsonBuffer docs and examples. My own suggestion: use a local Dynamic buffer instead.
Closing due to user error.

closed 2018-08-30 by devyte