esp8266/Arduino #5080: Esp8266 not Detected by R-PI(Master)

farhansarwar206, 2018-08-26:
Hi everyone,
                   I am working on I2C comm. b/w Raspberry PI (Master) and Esp8266 (Slave).But I was unable to detect ESP8266  on RPI.While RPI is detecting other I2C sensors as in below snap.
[image]

Code for ESP8266 Wemos D1:
#include <Wire.h>
#define SLAVE_ADDRESS 0x04
#define LED  13
int number = 0;
void setup() {
  pinMode(LED, OUTPUT);
  Serial.begin(9600);
  Wire.begin(SLAVE_ADDRESS);
  Wire.onReceive(receiveData);
  Wire.onRequest(sendData);
  Serial.println("Ready!");
}
void loop() {
  delay(100);
}
void receiveData(int byteCount) {
  Serial.print("receiveData");
  while (Wire.available()) {
    number = Wire.read();
    Serial.print("data received: ");
    Serial.println(number);
    if (number == 1) {
      Serial.println(" LED ON");
      digitalWrite(LED, HIGH);
    } else {
      Serial.println(" LED OFF");
      digitalWrite(LED, LOW);
    }
  }
}
void sendData() {
  Wire.write(number);
}
_______________________________________________________
Code for R-PI is Below:
import RPi.GPIO as gpio
import smbus
import time
import sys
bus = smbus.SMBus(1)
address = 0x04
def main():
    gpio.setmode(gpio.BCM)
    gpio.setup(17, gpio.OUT)
    status = False
    while 1:
        gpio.output(17, status)
        status = not status
        bus.write_byte(address, 1 if status else 0)
        print "Arduino answer to RPI:", bus.read_byte(address)
        time.sleep(1)
if __name__ == '__main__':
    try:
        main()
    except KeyboardInterrupt:
        print 'Interrupted'
        gpio.cleanup()
        sys.exit(0)
___________________________________________________
I have used the same code for Arduino UNO and its working fine.But ESP8266 is just not responding as I2C device.Do I need some drivers for esp8266 ???
Any idea is welcome !

5chufti, 2018-08-26:
where did you read that i2c on esp8266 supports slave mode?

devyte, 2018-08-27:
@5chufti is correct, it currently isn't supported. However, there is #3237 which is experimental, and feedback would be appreciated. There are other PRs as well about slave mode, but I'll leave such homework for the OP.

5chufti, 2018-08-27:
I know, there is also a [thread about over here](https://www.esp8266.com/viewtopic.php?f=33&t=13546)

gotfredsen, 2019-02-15:
I left a related comment on #3046: ESP8266(master) to ESP8266(slave) _works_ over I2C, but the slave _don't work_ on RPi, Micro:Bit and others.