esp8266/Arduino #5083: Esp8266 Wifi connection often disconnects (bcn_timout)

stefanodelbosco, 2018-08-27:
#### Platform

- Hardware:				NodeMCU 1.0 (ESP12E-Module)
- Core Version:			latest git 2.4.2 && 2.2.1
- Development Env:		Arduino IDE (last git) && Platformio 2.2.1
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			Nodemcu
- Flash Size: 		        4MB
- CPU Frequency:		80Mhz
- Upload Using:		OTA && SERIAL
- Upload Speed:   [115200 (Serial)]

### Problem Description

I looked for a solution between the various issues but none worked right now.
I have updated sdk to the latest version with git
The connection to the WiFi remains active for a time ranging from 30 seconds to 6 hours.
The module disconnects and reconnects by the connectDeviceToWiFi () function.

I tried to lower the time beecon in the router. Now it's every 30 ms (default 100ms).

I've been trying for three days to find a solution but the situation is not changing.
I tried with two different modules near the router and both give problems.

The RSSI signal is always between -70 and -78. I did a -68 signal tests but the problem persists.

I hope someone can help me.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266CustomTools.h>
#include "ota.h"
#include <ArduinoOTA.h>

ESP8266WebServer server(80);
ESP8266CustomTools device("SensorTest", D7, "sensor");

String IP_SERVER = "192.168.1.5";

void setup() {
  Serial.begin(9600);
  pinMode(D7, INPUT);
  device.connectDeviceToWiFi();
  device.syncCurrentTimestamp();
  device.generateOTPSecret();
  server.on("/info", handleInfo);
  server.begin();
  Serial.println(ESP.getCoreVersion());
  Serial.println(ESP.getSdkVersion());

  ota_setup("Esp8266", "PASSWORD");
}

void loop() {
    server.handleClient();
    //Check if connection is alive
    device.connectDeviceToWiFi();
    //Update IP/MAC
    device.checkAndUpdateConnectionInfo();
    if(digitalRead(D7) == 0){
      Serial.println("Change state");
    }
    device.checkAndUpdateTimestampOTP();
    //OTA Check
    ArduinoOTA.handle();
    delay(10);
}

```
#### ESP8266CustomTools::connectDeviceToWiFi()

```
void ESP8266CustomTools::connectDeviceToWiFi(){
  while(WiFi.status() != WL_CONNECTED){
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(200);
    WiFi.begin(ssid, password);
    long timewait = 0;
    long timeout = 10000;
    while(WiFi.status() != WL_CONNECTED && timewait < timeout){
       delay(1000);
       timewait = timewait + 1000;
    }
    if(WiFi.status() == WL_CONNECTED){
      String IP = WiFi.localIP().toString();
      String MACADDRESS = WiFi.macAddress();
      MACADDRESS.toLowerCase();
      Serial.println(IP);
      HTTPClient http;
      http.begin("http://" + IP_SERVER + "/api/device/connected_wifi_device?ip=" + IP + "&mac=" + MACADDRESS + "&type=" + TYPE);
      int httpCode = http.GET();
      String payload = http.getString();
      if(httpCode == 200){
        Serial.println("Message sent");
        last_server_update = millis();
      }
      http.end();
    }
  }
}
```

### Debug Messages

```
SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
scandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with dlink-XXXX, channel 10
dhcp client start...
ip:192.168.1.4,mask:255.255.255.0,gw:192.168.1.1
192.168.1.4
    SyncIpServer httpCode: 200
    SyncTimestamp httpCode: 200
    SyncOTP httpCode: 200
    COREVersion: 2_4_2
    SDKVerison: 2.2.1(cfd48f3)

bcn_timout,ap_probe_send_start
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
scandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with dlink-XXXX, channel 10
dhcp client start...
ip:192.168.1.4,mask:255.255.255.0,gw:192.168.1.1
192.168.1.4
....
```

stefanodelbosco, 2018-08-29:
After two days of testing, after applying that command, I had no more disconnections.
I think I found the problem and I would say that I can close this issue.
In case I have problems again, I will reopen the issues.
Thanks for the help @d-a-v

closed 2018-08-29 by stefanodelbosco

PrateekGoyal18, 2019-07-04:
> After two days of testing, after applying that command, I had no more disconnections.
> I think I found the problem and I would say that I can close this issue.
> In case I have problems again, I will reopen the issues.
> Thanks for the help @d-a-v

Did you mention this line "WiFi.setSleepMode(WIFI_NONE_SLEEP)" in setup ?

stefanodelbosco, 2019-07-04:
I think so.
I will check this evening and I will let you know.

stefanodelbosco, 2019-07-04:
My code:

void setup() {
  Serial.begin(9600);
  WiFi.setSleepMode(WIFI_NONE_SLEEP);
  [.....]
}

PrateekGoyal18, 2019-07-05:
Thanks @stefanodelbosco

tablatronix, 2019-08-22:
I am having this issue suddenly , will not connect at all , if it does it disconnects quickly after
What is the deal with this ? sleep mode none made no difference

```
bcn_timout,ap_probe_send_start
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
```

tablatronix, 2019-08-24:
It seems that if I call analogread too often my wifi drops out, I am not sure what mechanism is going on , I am trying to see how the adc works and how long the reads take and see if it is a wifi servicing issue but not enough for wdt.. does anyone know how to ensure wifi has time to service? Is a yeild() enough?

devyte, 2019-10-17:
@e346615 you need to test with latest git before asking that.

e346615, 2019-10-19:
Yes, I did, but I read somewhere that only version 2.5.1 worked. I tried both, but neither are working for me.

pryadav333, 2019-11-10:
it seems that the NodeMCU/ESP 8266 is not reliable for IoT projects where we need continuous wifi connectivity. It goes most of time in sleep mode and doesn't respond to the commands. Ideal all time wifi active should be its state.

d-a-v, 2019-11-10:
@pryadav333  This thread is closed. Please open a new issue if you still have this problem with version 2.6.0 of this core.

Tejas-MD, 2021-05-24:
I have this issue on latest release of core: 3.0 . I don't understand why the issue is closed. Please help out with a direction to proceed? I tried `WiFi.setSleepMode(WIFI_NONE_SLEEP)` , not working.

Pablo2048, 2021-05-25:
There is one rule of thumb in such situation - do a full chip erase before uploading your code. I guess that the problem is by switching SDK version using new core 3.0.0 vs the old one so rfcalib/config area is corrupted...

jeroenst, 2021-07-11:
> it seems that the NodeMCU/ESP 8266 is not reliable for IoT projects where we need continuous wifi connectivity. It goes most of time in sleep mode and doesn't respond to the commands. Ideal all time wifi active should be its state. All wakeup commands fails most of time to wake up the nodemcu/esp8266

FYI: I have esp8266/nodemcu devices with uptimes over 60 days and continuous wifi without any problem.

Tejas-MD, 2021-07-11:
For me, it works fine after `WiFi.setSleepMode(WIFI_NONE_SLEEP)` --> Chip erase ---> Flash with new firmware. 

Very grateful to @Pablo2048