esp8266/Arduino #5084: wificlientsecure failed to verify letsencrypt ca cert

chegewara, 2018-08-27:
I am using this example. I changed ca cert to letsencrypt certificate https://letsencrypt.org/certs/lets-encrypt-x3-cross-signed.pem.txt converted with https://certyfikatyssl.pl/ssl-tools/convert-certificate.html#certificateConverter to der and then to hex with `xxd -i -a` command. Here is log:
```
Setting time using SNTP.
Current time: Tue Aug 28 06:43:55 2018
=== CERTIFICATE ISSUED TO ===
Common Name (CN):		Let's Encrypt Authority X3
Organization (O):		Let's Encrypt
Country (C):			US
Basic Constraints:		critical, CA:TRUE, pathlen:0
Key Usage:			critical, Digital Signature, Key Cert Sign, CRL Sign
=== CERTIFICATE ISSUED BY ===
Common Name (CN):		DST Root CA X3
Organization (O):		Digital Signature Trust Co.
Not Before:			Thu Mar 17 16:40:46 2016
Not After:			Wed Mar 17 16:40:46 2021
RSA bitsize:			2048
Sig Type:			SHA256
connecting to website.com
State:	sending Client Hello (1)
State:	receiving Server Hello (2)
State:	receiving Certificate (11)
=== CERTIFICATE ISSUED TO ===
Common Name (CN):		website.com
Organization (O):		<Not Part Of Certificate>
Basic Constraints:		critical, CA:FALSE, pathlen:10000
Key Usage:			critical, Digital Signature, Key Encipherment
Subject Alt Name:		website.com 
=== CERTIFICATE ISSUED BY ===
Common Name (CN):		Let's Encrypt Authority X3
Organization (O):		Let's Encrypt
Country (C):			US
Not Before:			Wed Aug 15 11:22:41 2018
Not After:			Tue Nov 13 11:22:41 2018
RSA bitsize:			4096
Sig Type:			SHA256
=== CERTIFICATE ISSUED TO ===
Common Name (CN):		Let's Encrypt Authority X3
Organization (O):		Let's Encrypt
Country (C):			US
Basic Constraints:		critical, CA:TRUE, pathlen:0
Key Usage:			critical, Digital Signature, Key Cert Sign, CRL Sign
=== CERTIFICATE ISSUED BY ===
Common Name (CN):		DST Root CA X3
Organization (O):		Digital Signature Trust Co.
Not Before:			Thu Mar 17 16:40:46 2016
Not After:			Wed Mar 17 16:40:46 2021
RSA bitsize:			2048
Sig Type:			SHA256
State:	receiving Server Hello Done (14)
State:	sending Client Key Exchange (16)
State:	sending Finished (16)
State:	receiving Finished (16)
=== CERTIFICATE ISSUED TO ===
Common Name (CN):		website.com
Organization (O):		<Not Part Of Certificate>
Basic Constraints:		critical, CA:FALSE, pathlen:10000
Key Usage:			critical, Digital Signature, Key Encipherment
Subject Alt Name:		website.com 
=== CERTIFICATE ISSUED BY ===
Common Name (CN):		Let's Encrypt Authority X3
Organization (O):		Let's Encrypt
Country (C):			US
Not Before:			Wed Aug 15 11:22:41 2018
Not After:			Tue Nov 13 11:22:41 2018
RSA bitsize:			4096
Sig Type:			SHA256
Verify:				No trusted cert is available
=== CERTIFICATE ISSUED TO ===
Common Name (CN):		Let's Encrypt Authority X3
Organization (O):		Let's Encrypt
Country (C):			US
Basic Constraints:		critical, CA:TRUE, pathlen:0
Key Usage:			critical, Digital Signature, Key Cert Sign, CRL Sign
=== CERTIFICATE ISSUED BY ===
Common Name (CN):		DST Root CA X3
Organization (O):		Digital Signature Trust Co.
Not Before:			Thu Mar 17 16:40:46 2016
Not After:			Wed Mar 17 16:40:46 2021
RSA bitsize:			2048
Sig Type:			SHA256
Verify:				No trusted cert is available
Error: No trusted cert is available
ERROR: certificate verification failed!
connecting to website.com
Alert: close notify
```
Certificate seems to be recognized and all seems to be properly setup, but cant pass with `verifyCertChain` and without verification connection is failed.

igrr, 2018-08-27:
In this case your root CA is "DST Root CA X3" (see the last section of output) and you need to load that into WiFiClientSecure. If you pass it an intermediate certificate (such as Let's Encrypt Authority in this case) it will not recognize it as the root one.

chegewara, 2018-08-27:
thanks a lot. certificate is verified now

chegewara, 2018-08-27:
hi @igrr 
One more question, what it means when response from this part of code is only `72`:
```
  String line = client.readStringUntil('\n');
  if (line.startsWith("{\"state\":\"success\"")) {
    Serial.println("esp8266/Arduino CI successfull!");
  } else {
    Serial.println("esp8266/Arduino CI has failed");
  }
  Serial.println("reply was:");
  Serial.println("==========");
  Serial.println(line);   <---- 72
  Serial.println("==========");
  Serial.println();
```
response should be hash in form:
`C1036D1855A04662856CE6707854016`

chegewara, 2018-08-28:
hi @igrr thanks for help, but now i have problem with sending post data with wificlientsecure.
here is code i am using to send request:
```
client.print(String("POST ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "Content-Type: application/x-www-form-urlencoded\r\n" +
               "Authorization:  hash\r\n\r\n" +
               "amount=1\r\n"
               "Connection : close\r\n\r\n");
```
Im getting response that amount is missing.

igrr, 2018-08-28:
`amount=1` line does not look like an HTTP header... If you intended it to be sent in the request body, then it has to go after the \r\n\r\n part.

If you aren't very familiar with HTTP protocol, my advice would be to use ESP8266HTTPClient library, at least initially.

chegewara, 2018-08-28:
Yes, it is body and as you can see i have `\r\n\r\n` in this line:
`"Authorization:  hash\r\n\r\n" +`

devyte, 2018-08-28:
@chegewara you ignored the issue template, so it's difficult to follow what you're doing. Given @igrr 's answer, I won't close this for now.
However, it seems to me that you're trying to use the "default" WiFiClientSecure. That default uses axtls as the ssl lib under the hood. You should know that axtls is no longer supported by the original authors, and we therefore plan to deprecate it in favor of our bearssl-based WiFiClientSecure, and eventually retire it. In the meantime, our own WiFiClientSecure implementation will be supported for issues within our code, but not for issues tracked to axtls.
The bearssl lib is currently very active, and we already offer experimental support for it. The api is 99% the same as the current axtls one (which is why we haven't switched over yet), and there are usage examples available.
I strongly suggest switching to the bearssl::WiFiClientSecure implementation.

chegewara, 2018-08-28:
@devyte 
Sorry if my issue is trivial or im working with wrong library, but im pretty new and noob with esp8266 (i prefer esp32). This is example i am using because its only i found that is using ca cert instead of fingerprint.
https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi/examples/HTTPSRequestCACert

I appreciate any help and im not a paracite, i am trying to help other programmers too when i can.

devyte, 2018-08-28:
CC @earlephilhower

igrr, 2018-08-28:
> Yes, it is body and as you can see i have \r\n\r\n in this line:
"Authorization: hash\r\n\r\n" +

Why is `Connection : close part of the body then? It is normally a header. The server is trying to decode the body with `application/x-www-form-urlencoded` content-type and is probably failing.

For testing, if you don't have access to server logs, you can use `httpbin.org` or a similar service.

chegewara, 2018-08-28:
@igrr i have code with and without `Connection: close`, but i will follow @devyte suggestion and give a tru BearSSL library. Thanks all for help.

closed 2018-08-28 by chegewara