esp8266/Arduino #5085: server.serveStatic ON SD CARD?

laercionit, 2018-08-28:
I am studying a little of ESP8266, I had contact with server.serveStatic
I wonder if there is a possibility of using the SD CARD instead of SPIFFS.

Would anyone have any ideas?

duncan-a, 2018-08-28:
> Likely possible, but not currently implemented. Out of curiosity, what
> would you serve off an SD?
> The reason it's not implemented is very complex (the reason itself, not
> the code) so it will be a while before we get to it.
> Closing, and I'd close twice if I could because you ignored the issue
> template and the instructions it contains.

" I'd close twice if I could because you ignored the issue template and the
instructions it contains"

I doubt your sarcasm is appreciated by anyone, least of all the recipients.

Maybe it's time to moderate the moderators???

d-a-v, 2018-08-28:
(The question has been answered though)

You are welcome to join and contribute the community effort for improving the esp8266/arduino core, answering and maintaining the opened issues.

When this happen, you will thank the issue template and silently blame people who don't use it.

Thank you @devyte for the current version of the issue template, meant, for those who wish and can, help users.