esp8266/Arduino #5087: Missing file SPISlave.a reported

paultanner, 2018-08-28:
- [Y ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ Y] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ Y] I have tested that the issue is present in current master branch (aka latest git).
- [ Y] I have searched the issue tracker for a similar issue.
- [ N/A] If there is a stack dump, I have decoded it.
- [ Y] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[community version 2.3.0]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[512K]
- lwip Variant:   [N/A]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Build on Arduino 1.6.12 throws error

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

see SPISlave_Test in the repo

### Debug Messages

```
xtensa-lx106-elf-gcc: error: libraries/SPISlave/SPISlave.a: No such file or directory
exit status 1
Error compiling for board Generic ESP8266 Module.
```

paultanner, 2018-08-28:
I fixed this by updating the ESP8266 board to 2.4.2

5chufti, 2018-08-28:
ok, then please close the issue