esp8266/Arduino #5094: Serial.flush should change to Serial.flush(uint32_t data_width, uint32_t parity, uint32_t stop)

yldzmuhammed, 2018-08-30:
#### Platform

- Hardware:				[LoLin new NodeMCU V3 board]
- Core Version:			[Updated today.So should be 2.4.2?]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu 0.9]
- Flash Size: 		[4MBB]
- lwip Variant:   [v2 Lower Memory|Higher Bandwidth]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [921600]

### Problem Description

I am building a Rs485 project. So esp has to control RE and DE pins. I have done some test and finaly find a easy way (for me at least). I added second flush function like:
```

void HardwareSerial::flush(uint32_t data_width, uint32_t parity, uint32_t stop)
{
    if(!_uart || !uart_tx_enabled(_uart)) {
        return;
    }

    uart_wait_tx_empty(_uart);
    //Workaround for a bug in serial not actually being finished yet
    //Wait for 8 data bits, 1 parity and 2 stop bits, just in case
    delayMicroseconds(((data_width + parity + stop) * 1000000) / uart_get_baudrate(_uart) + 1);
}

```

I belive in the HardwareSerial.h file the prototype should change:

`void flush(uint32_t data = 8_width, uint32_t parity = 0, uint32_t stop = 1);`

I user using serial port with default settings (8N1) then user dont have to anything. Otherwise, function must call with serial config parameters.

With this function, esp can handle tx enable, rx enable nicely.

What do you think? Is there any better way to handle RS485 comm direction problem?

NOTE: I am not a cpp expert.

d-a-v, 2018-08-30:
Those parameters should not come to this function that way.
They are already internally stored and their values should be extracted from there.

What would be nice is to move this workaround into `uart.c:uart_wait_tx_empty()`, in there you can read `USC0(uart->uart_nr)` which is the serial mode (8N1 or whatever).
With a couple of defines, you can extract from it the width, the parity and the stop bit.
```
#define MODE2WIDTH(mode) (((mode) >> 2) + 5) // see UART_NB_BIT_* 
```
then `MODE2WIDTH(USC0(uart->uart_nr))` would give you the needed `data_width`.

We would ~love~be thrilled if you could make a pull request !
(note: this is C so you are an expert :)

devyte, 2019-10-29:
This is already done in #5293 . Closing.

closed 2019-10-29 by devyte